betaflight/betaflight #11155: Bi-directionnal DShot not working with Mamba H743 (BMI270)

LordGG, 2021-12-24:
### Describe the bug
When I enable bi-directionnal DShot on Mamba H743 (BMI270), motors 1 and 3 stop working and have 100% errors rate. Motors work fine without bi-dir dshot.

### To Reproduce
Just enable bi-directionnal DShot (DShot600), and test motors in the Motors' tab.

### Expected behavior
0% error and all 4 motors working.

### Flight controller configuration

```
# version
# Betaflight / STM32H743 (SH74) 4.3.0 Dec 23 2021 / 19:25:24 (5933d96bc) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAH743, version: e4b51680, date: 2021-10-04T00:52:01Z

# start the command batch
batch start

board_name MAMBAH743
manufacturer_id DIAT

# feature
feature -RX_PARALLEL_PWM

# serial
serial 4 1 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 13 3 1300 1700 0 0
aux 3 26 3 1800 2100 0 0
aux 4 35 2 1800 2100 0 0
aux 5 36 1 1800 2100 0 0

# master
set acc_calibration = -30,59,51,1
set mag_hardware = NONE
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_output_reordering = 3,2,1,0,4,5,6,7
set align_board_yaw = 180
set vbat_scale = 108
set small_angle = 180
set deadband = 2
set yaw_deadband = 3
set osd_vbat_pos = 2497
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2081
set osd_throttle_pos = 2433
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 2465
set osd_disarmed_pos = 2283
set debug_mode = GYRO_SCALED

profile 0

# profile 0
set iterm_relax_cutoff = 17

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 219
set pitch_rc_rate = 219
set yaw_rc_rate = 213
set roll_expo = 82
set pitch_expo = 82
set yaw_expo = 73
set roll_srate = 1
set pitch_srate = 1
set yaw_srate = 0

# end the command batch
batch end
```
```
# 
# Building AutoComplete Cache ... Done!
# 
# resource
resource BEEPER 1 E03
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 C08
resource MOTOR 8 C09
resource PPM 1 B09
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 E08
resource SERIAL_TX 8 E01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 D00
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource SERIAL_RX 8 E00
resource I2C_SCL 1 B06
resource I2C_SCL 2 B10
resource I2C_SDA 1 B07
resource I2C_SDA 2 B11
resource LED 1 E05
resource LED 2 E04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 E12
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 E13
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B02
resource SPI_MOSI 4 E14
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C02
resource ADC_CURR 1 C03
resource ADC_EXT 1 C00
resource PINIO 1 C05
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 E15
resource GYRO_CS 1 A04
resource GYRO_CS 2 E11
resource USB_DETECT 1 E06
```

### Setup / Versions
 
 - Flight controller: Mamba H743 (BMI270)
 
 - Other components: ESCs Mamba F55_128K BL32, Caddx Vista, RX Beta FPV 2.4 Ghz, T-Motor 2505, 1000µF cap + Spike absorber
 
 - How are the different components wired up:
Vista on UART5, RX on UART1

SteveCEvans, 2021-12-24:
Tested using 4.3.0RC1 using a [Tekko32 F4 Metal 4in1 ESC](http://www.holybro.com/product/tekko32-f4-metal-4in1-esc-65a/) with BLHeli 32 Revision 32.8 and the following config with no issues.

[image]

```
# diff all

# version
# Betaflight / STM32H743 (SH74) 4.3.0 Dec 24 2021 / 18:48:24 (norevision) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAH743, version: e4b51680, date: 2021-10-04T00:52:01Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAH743
manufacturer_id DIAT
mcu_id 003700333230510a33343930
signature 

# feature
feature -RX_PARALLEL_PWM

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 48,30,-4073,1
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

LordGG, 2021-12-24:
> Tested using 4.3.0RC1 using a [Tekko32 F4 Metal 4in1 ESC](http://www.holybro.com/product/tekko32-f4-metal-4in1-esc-65a/) with BLHeli 32 Revision 32.8 and the following config with no issues.

I have tried the nightly from yesterday, RC was not yet available. I'll give a try, thanks ! Do you have the version with BMI270 ?

Diatone confirmed the issue and told me they sent a PR last month. Could have it been merged to this first RC ?

LordGG, 2021-12-26:
I have just tested with 4.3 RC1, still the same issue : H743 with BMI270 have only 2 motors working when bi-directionnal DShot is enabled.

Diatone replied me that :
[image]

haslinghuis, 2021-12-27:
@LordGG could you please be specific about the PR?

DusKing1, 2021-12-27:
I have received the same report month ago directly from DIATONE but I can't reproduce it successfully, but your rig reminds me, that if you @LordGG  can take a look at the resistor on the ESC's dshot signal lines measure the value for that and tell us if that's a 1k 10k or 100k resistor? I don't have the DIATONE ESC same as yours in my hand.

LordGG, 2021-12-27:
> @LordGG could you please be specific about the PR?

I tried... but couldn't get a statisfy answer :/ 
[image]

> if you @LordGG can take a look at the resistor on the ESC's dshot signal lines measure the value for that and tell us if that's a 1k 10k or 100k resistor?
I removed the FC, but there is an EMI shield glued on top of ESCs, so I can't see a lot. I gave a look at the bottom but I don't see resistors directly wired to  signal lines.
[image]

> Also, please check if dshot 300 works for you?
I tried, it doesn't work better : still 2 motors not working with 100% errors. I know their previous ECSs (F50 BL32) don't support DShot600 with Bi-dir dshot. I also did this test : I replaced the H743 with a Mamba F722 APP, and 4 motors works perfectly with DShot600 and 0% error.

DusKing1, 2021-12-27:
Ok I saw four resistors right between the base board and the chip board, next to the JST connector, can you try to measure it?

LordGG, 2021-12-27:
> Ok I saw four resistors right between the base board and the chip board, next to the JST connector, can you try to measure it?

They are so small I didn't see them ^^ They are 220 Ohms resistors.

LordGG, 2021-12-27:
I am not sure it can help, but here is what could get from Diatone :
[image]

This bi-directionnal DShot issue seems only related to 4.3 on this target. I didn't get what they meant about the gyros.

SteveCEvans, 2021-12-28:
@LordGG I've just tried configuring the mixer as `OCTOX8` and connecting a [Holybro Tekko32 F3 Metal 65A ESC](http://www.holybro.com/product/tekko32-f3-metal-esc-65a/). I was only able to test four motors at a time, but all eight worked.

[image]

[image]

I can confirm this FC has the BMI gyro.

```
# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.33V, Core temp=42degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 4168, max available: 131072
Devices detected: SPI:2, I2C:1
Gyros detected: gyro 1, gyro 2 locked dma
GYRO=BMI270, ACC=BMI270, BARO=DPS310
OSD: MAX7456
System Uptime: 12 seconds, Current Time: 2021-12-28T15:06:10.471+00:00
CPU:12%, cycle time: 314, GYRO rate: 3184, RX rate: 15, System rate: 9
Voltage: 1259 * 0.01V (3S battery - OK)
I2C Errors: 2
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP RPMFILTER
```

LordGG, 2021-12-28:
@SteveCEvans Thanks for your feedback and for doing the test.  

I am lost : Diatone confirmed a known issue with 4.3 and bidir dshot, other users had to fall back to 4.2.11, and for me, switching the FC to a F722 solved the issue. The fact it works for you with different ESCs is confusing but good to know.

If ESCs were involved in the issue, I would have expected to meet the same issue with another FC. Maybe be something specific between this FC and those Mamba F55_128K BL32 ?!

If I am brave enough, I'll try to switch the ESCs.

snow520lee, 2021-12-31:
I have the same exact issue with a JHEMCU H745. with 4.3.0 only motor 1 and 2 works... 3 and 4 no spinning and 100% error.
Tried BLHeli 16.71 and 16.9, same result. turn off bi-directional, all 4 motor works.

AlexRice13, 2022-01-25:
Here I am using MAMBA F65 128k esc with MAMBA H7 FC BMI270 version, still got problem on getting bi-directional dshot work.
Perhaps I made a little progress by CLI command ?
```
set dshot_bitbang = OFF
save
```
then it forced all motor bi-directional dshot work, here's screenshot
[image]

now I can see the rpm telemetry form all escs, but two of them refreshed in a very low rate, I'm not sure it can work properly, and dare not to have a testflight.

SteveCEvans, 2022-01-25:
@LordGG What version of BLHeli are you using? Have you tried DSHOT300? As you're running a PID loop of below 4kHz with the BMI270 gyro you might as well use that if it works for you.

I've hooked up a logic analyzer and all is working fine with either the BMI270 or MPU6000 gyros selected.

BMI270 (`gyro_to_use = FIRST`) with DSHOT600 updates at ~3.2kHz
[image]

BMI270 (`gyro_to_use = FIRST`) with DSHOT300 updates at ~3.2kHz
[image]

MPU6000 (`gyro_to_use = SECOND`) with DSHOT600 updates at 8kHz
[image]

Maxis666, 2022-01-26:
[image]

Dont work bidirect in 4.3-RC2

LordGG, 2022-01-27:
> @LordGG What version of BLHeli are you using? Have you tried DSHOT300? As you're running a PID loop of below 4kHz with the BMI270 gyro you might as well use that if it works for you.
>
> I've hooked up [...]

I was using BLHeli32 32.8 as it was the only firmware available for Mamba ESCs 128K, no 32.7. I tried DShot300 with no more success.

Did you run your tests with 2 differents FCs ? My Mamba H7 had only one gyro (BMI270), no second MPU6000 gyro, but you probably used a second Mamba with dual MPU6000, right ?

> [image]
> 
> Dont work bidirect in 4.3-RC2

That's exacly what I and other users got : 2 motors with 100% error rate, not working at all with bidir dshot enabled. It is also what Diatone confirmed me. Yet, it is working for @SteveCEvans , maybe because of different ESCS ? What ESCs did you pair with this FC ?

My FC ended up in a draw... the lack of bidir dshot was a huge issue, the buzzer was not working either (same for other folks) and running a low speed loop with additionnal gyro filter was too much. I just replaced the FC and it fixed all the problems.

SteveCEvans, 2022-01-27:
Mine has dual gyros.

I used a Holybro Tekko32 F3, http://www.holybro.com/product/tekko32-f3-metal-esc-65a/

LordGG, 2022-01-27:
> Mine has dual gyros.

Oh ok, good to know ! So it is not the same FC, which might explain the different behaviour.

Maxis666, 2022-01-27:
I have a h743 mamba with bmi270
And esc - mamba 55a esc 48khz with 32.8 sv (own firmware)

AlexRice13, 2022-01-29:
> 

Here is a temporary solution for MAMBA H7 controller.
For me it works
[image]
[曼巴H7飞控双向dshot问题临时解决方案.docx](https://github.com/betaflight/betaflight/files/7963806/H7.dshot.docx)

LordGG, 2022-01-29:
@AlexRice13 Thanks for sharing a temporary solution. I'll pass on that since I have replaced the FC, but it might help others.

Have you also noticed that the buzzer is not working ?

AlexRice13, 2022-01-30:
Oh I forgot to joint my buzzer

Maxis666, 2022-01-30:
> > 
> 
> Here is a temporary solution for MAMBA H7 controller. For me it works [image] [曼巴H7飞控双向dshot问题临时解决方案.docx](https://github.com/betaflight/betaflight/files/7963806/H7.dshot.docx)

It not working for mamba h743 bmi270

howels, 2022-01-30:
Same issue with 2 motors not working on JHEMCU JHH743 AIO, as described in https://github.com/betaflight/betaflight/issues/11309#issuecomment-1025138879

snow520lee, 2022-01-31:
JHH743 doesn’t have a target either in BF.

howels, 2022-01-31:
I confirm that as per https://github.com/betaflight/betaflight/pull/10383#issuecomment-932937466, turning on `DEBUG=GDB` allows the board to work. Try this here?

AlexRice13, 2022-02-01:
> > Here is a temporary solution for MAMBA H7 controller. For me it works [image] [曼巴H7飞控双向dshot问题临时解决方案.docx](https://github.com/betaflight/betaflight/files/7963806/H7.dshot.docx)
> 
> It not working for mamba h743 bmi270

Mine is MAMBA H7 BMI270, yours has different motors not working, so you have to change the motors you are going to remap and so on.

howels, 2022-02-01:
> JHH743 doesn’t have a target either in BF.

Manufacturer was shipping the iFlight H743 target, with some resource remapping.

howels, 2022-02-01:
> I have the same exact issue with a JHEMCU H745. with 4.3.0 only motor 1 and 2 works... 3 and 4 no spinning and 100% error. Tried BLHeli 16.71 and 16.9, same result. turn off bi-directional, all 4 motor works.

If you are able to build 4.3 yourself, try saving the dump then run a build with `make TARGET=IFLIGHT_H743_AIO DEBUG=GDB`and restore the dump you took onto that new firmware. For me, the GDB option changes compiler optimisation and appears to work whereas the default more-optimised build gives the output you show (2 motors not working).

Maxis666, 2022-02-02:
No matter how strange it may sound.  But I'm flying the second week without bidirectional dshot and rpm filtration.  And you know, it's even better and faster)

justinflipflops, 2022-02-09:
I can confirm DEBUG=GDB fixes the issue here with the dual mpu6000 board.

DusKing1, 2022-02-10:
We have tried to reproduce this issue, and at the ESCs range that we have (including IFLIGHT GEPRC HGLRC TMOTOR and old MAMBA ESC with BLHeli_32 or AM32 or bluejay firmware), we can't reproduce any of the issues described above. As none of us has the ESCs as described we come to believe that it is related to the ESC, not betaflight fault. @SteveCEvans @haslinghuis this issue shall be closed.

snow520lee, 2022-02-10:
I don’t think you got at the right conclusion….
Not necessary because you don’t have that FC or ESC that there is no issues to solve….

There are many reports of H7 AIO with the same problem. Some motors are spinning, others not, different firmwares have different motors start or stop working….. this is happening on iFlight H7, Mamba H7, JHEMCU H7 etc…..

Most of them are AIO so the 4 ESC are identical, it’s what drives them that is messed up. 

KR
Marco Borio

LordGG, 2022-02-10:
> We have tried to reproduce this issue, and at the ESCs range that we have (including IFLIGHT GEPRC HGLRC TMOTOR and old MAMBA ESC with BLHeli_32 or AM32 or bluejay firmware), we can't reproduce any [...]

Everyone with that FC (**Mamba H743 BMI270 single gyro**) can reproduce, even at Diatone, they say it is a Betaflight specific issue with 4.3 (works with 4.2.11) :
[image]
Since they cannot provide any link to this PR, I am sure something is wrong on their side, probably a design flaw. But the fact it works with 4.2 but not 4.3 is odd.

This FC does not work as it should. Diatone says it is BF's faults and BF says it is Diatone's ESCs fault, and I am in the middle ^^

DusKing1, 2022-02-10:
I wrote the MAMBAH743 target for them and i have two of that, if you think I don't know what the exact issue is then good luck.

gchedzhemov, 2022-02-10:
It seems that part of the confusion here is that this issue started with a report of bidir dshot not working with aMamba H743, but as the comments progressed, it became clear that other H743 FC/AIOs also suffer from the same (or at least very similar) issue.

As @snow520lee commented, the affected AIOs include iFlight H7, Mamba H7, JHEMCU H7 and others. 

I, and @howels, both have JHEMCU GH743AIOs which exhibit these weird DMA timing issues with bi-directional dshot.

@DusKing1, since I have a brand new spare JHEMCU GH743AIO FC, it would be my pleasure to send it to any member of the Betaflight development team interested in taking a crack at the problem, and to do so at my cost and expense, including covering shipping. Just let me know, and I would be happy to oblige.

If you believe the issue in question is different from, and unrelated to, the Mamba H743 issue discussed here, then I'm perfectly happy opening a new and separate issue for the JHEMCU GH743AIO and possibly other AIOs, though I believe there's at least one other such issue already [Issue 11376](https://github.com/betaflight/betaflight/issues/11376), and we were redirected by @daleckystepan here from there.

LordGG, 2022-02-10:
> I am the one wrote the MAMBAH743 target config for them and made PRs for them, and i have two of that, if you think I don't know what the exact issue is then good luck.

I only blame Diatone for selling a stack that does not work as expected. By no mean I am blaming you or anyone from the BF team.

But if you know what the exact issue is, could you please share it ? I just want to understand why bidir dshot works with some ESCs on 4.2 but not with 4.3 and what is the issue with some ESCs like this Mamba F55_128K BL32. It could help other customers and manufacturers to fix what is wrong with their ESCs.

According to different feedbacks in this thread, it seems there is something going on surrounding H743 and bidir dshot, I have no idea if it can be fixed by software or if it needs hardware fix.

justinflipflops, 2022-02-11:
So I have the MAMBA H743 with dual mpu6000 gyro's. ESC is 65a 128k diatone. Issue is present in 4.3 from the day the STMh743 config was merged 5549a906f67ae7533b9c9a7c1a81b944dbc5c750 and is not present in the current 4.2.11

You are telling me a hardware and ESC design flaw is why one version of the firmware works(with no other variables changed) and not a single build for 4.3 from the initial merge on does without specifying DEBUG=GDB which is only changing... betaflight code?

justinflipflops, 2022-02-11:
So I spent some time today going back to 4.2 and then building forward by month repulling the unified config settings for the H743 and found that up and including 0833aab88664f83a834c2c496470b63d83da8afc everything works as anticipated. This is Sept 12th 2020. While going to commit 0423ad826a390179ffb4d24ead1236a203dfce1c which essentially roles up everything from Sept 13th 2020 and it breaks. The bulk of the commits on the 13th are related to the H7 libraries so I'd say we may have a strong chance thats where the issue actually is. I'm going to try and dig more into it later but I am no expert and the changes happening here may be over my head to diagnose.

justinflipflops, 2022-02-12:
Just pulled in v1.9.1 library and problem still exist. right now i have a side by side diff comparing 1.6 to 1.8 but at this point i'm not sure what function's I would be looking at for changes. If anyone has feedback or a road to travel down, let me know I'll travel :)

howels, 2022-02-12:
> Just pulled in v1.9.1 library and problem still exist. right now i have a side by side diff comparing 1.6 to 1.8 but at this point i'm not sure what function's I would be looking at for changes. If anyone has feedback or a road to travel down, let me know I'll travel :)

Which board did you manage to build? I pulled in the new versions of the Drivers but when going higher than v1.8.0 of the CMSIS and HAL drivers from H743 I see errors:

```
./src/main/drivers/timer_hal.c:1231:61: error: 'TIM_DMADelayPulseCplt' undeclared (first use in this function); did you mean 'TIM_DMADelayPulseHalfCplt'?
 1231 |         htim->hdma[TIM_DMA_ID_CC1]->XferCpltCallback = HAL_TIM_DMADelayPulseCplt;
```
And this callback appears to have been removed in 1.9.0 (https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/9439ae79fd1ead4a3fe75016912e29e8acabf4da/Release_Notes.html) "Add APIs: HAL_TIM_GetActiveChannel, HAL_TIM_GetChannelState and HAL_TIM_DMABurstState and remove TIM_DMADelayPulseCplt callback"
Wondered how you worked around this?

justinflipflops, 2022-02-12:
@howels STM32H743 - so I was referencing the https://github.com/STMicroelectronics/STM32CubeH7 repository which actually turns out to have v1.10.1 merged in it(I compared with meld). So I have merged v1.10.1 CMSIS and HAL into BF4.3 and it compiles/flashes/dshot still works the same for the purpose of testing this. my work:

https://github.com/justinflipflops/betaflight/tree/FirmwareV1101

I just wish I knew what functions I should be examining. I'm thinking of just replacing files with old ones till I hit a working config, then try and dig deeper. 

I may be entirely wrong and completely fubar'd this, but I know the v1.8 library change plays a role in the issue we are experiencing here.

howels, 2022-02-12:
> @howels STM32H743 - so I was referencing the https://github.com/STMicroelectronics/STM32CubeH7 repository which actually turns out to have v1.10.1 merged in it(I compared with meld). So I have [...]

Thanks for the pointer to your branch. I just built the IFLIGHT_H743_AIO_V2 target and applied it to my JHEH743 with the usual resource changes. It worked and all 4 motors show RPM data without needing to disable compiler optimisations.

haslinghuis, 2022-02-13:
ping @jflyper

hydra, 2022-02-18:
In #11387 @howels commented that disabling CPU Data caches fixed a DSHOT output problem - https://github.com/betaflight/betaflight/issues/11376#issuecomment-1044879120

I guess my hunch regarding DMA buffers and caching was correct.  Requires investigation.  I'd start with generating .map files for a build that works and one that doesn't, and then looking at the physical memory address assigned to the dshot buffers.

If the address changed then I'd use `git bisect` and see which commit resulting the memory address changing, generating a .map file and comparing the address at each bisect point to the known-good address.

I recall there were some linker file changes and some gcc __attribute changes that may have resulted in the dma buffers ending up in the wrong memory location, it has happened before, e.g. with led strip and sdcard dma buffers.

other things that can cause this are changes to the dma/dshot caching code and changes to the mpu configuration. these might also be able to be found using git bisect if you can replicate the problem easily.

I don't have time to investigate this, but hopefully these hints will help whoever has time.

justinflipflops, 2022-02-19:
@hydra in my case disabling the caching did not fix the issue. Nor did disabling @SteveCEvans  USE_DSHOT_CACHE_MGMT. However, fully removing the DSHOT BITBANG code by undefining USE_DSHOT_BITBANG for the H7 target did fix it. However, having the code and simply disabling bitbang by setting dshot_bitbang=off results in all motors having 99% errors but they are detected.

SteveCEvans, 2022-02-20:
I've just been back and reviewed the code again to check it's correct. It's probably worth some explanation at this time.

If `USE_DSHOT_CACHE_MGMT` is defined, which it is for H7 by default, then explicit cache management is enabled.

Bit banged DSHOT uses DMA to write out a pattern or 16 x 3 bits to the motor pins. Each triplet of bits is `1x0` where `x` is the binary value to be written. Note that this is inverted id bidirectional DSHOT is enabled.

The buffer to be DMAed is 16x3x32 bits in size. All data DMAed to the output is written to the BSRR register. Each 32 bit write can set/clear any of the port's outputs.

[image]

Reading DMAs data from the port's IDR register which is 16 bits wide.

[image]

In the case of output, the data buffer is contracted by the processor and may reside it it's cache, so **prior** to launching the DMA this data must be flushed from the cache into the memory where the DMA engine can see it, otherwise stale (incorrect) data will be written. This flushing is achieved using the `SCB_CleanDCache_by_Addr()` routine, called from `bbUpdateComplete()`. For each motor a check is made to see if a prior motor uses the same port, and if so doesn't bother to flush/clean the cache as the buffer for that port will already have been cleaned.

In the case of input if the buffer in memory is cached by the processor, then we must invalidate the buffer region to ensure that the the processor doesn't access the old copy of data in cache, but instead fetches the newly DMAed data from memory into cache. This is done **after** the DMA has completed by a call to `SCB_InvalidateDCache_by_Addr()` from `bbUpdateStart()`. Again for each motor a check is made to see if a prior motor is on the same port, so the cache will only be invalidated once.

Using a Mamba H743 with a Holybro Tekko32 F3, http://www.holybro.com/product/tekko32-f3-metal-esc-65a/ connected thus which enables me to capture the motor traces on a logic analyzer. I'm simply not seeing these issues.

[image]

If you don't want to use the cache management described above, then simply undefine `USE_DSHOT_CACHE_MGMT` and the DSHOT buffers will be placed in uncached memory so no cache coherency operations are required. I've just tested this and again see no errors.

It would be very useful to capture the MOTOR waveforms on a bad FC/ESC combo as I have done above.

SteveCEvans, 2022-02-20:
I have requested a Mamba F55_128K BL32 ESC from Diatone in order to be able to wire it up as shown above, and properly investigate this issue.

justinflipflops, 2022-02-20:
@SteveCEvans thanks for the more in-depth explanation of what is going on in the code. I looked at the cost of one of those logic analyzers and it's a bit outside of my kind of budget for now. Instead, all I can offer is time and things I've tried that will hopefully lower the time it takes for troubleshooting. Maybe we can come to a determination before that ESC ever arrives.

It is worth noting, my hardware is a bit different. I have the H743 with the dual MPU6000 and an F65_128K BL32 ESC. But my symptoms are the exact same. 

I did try compiling with USE_DSHOT_CACHE_MGMT undefined and it did not show any change with the default configuration. The one thing I did not try the last time that I did try this time, is that I set dshot_bitbang to off. Which with cache management results in all motors detect with 99% dshot errors. This time, with the cache management disabled, I then set dshot bitbang off via CLI, and I have 4 working motors. 

I also don't know if it's been previously mentioned. but this only seems to be an issue with motors 1-4 resources A00-A03 and does not seem to affect motors 5-8 resources B00,B01,C08,C09. Which is how I originally got around the problem when thinking the motor outputs were possibly just bad. Plugged the ESC into the second port for those resources, and then remapped in BF and had no issues.

Maxis666, 2022-02-21:
Meanwhile... I've been flying mamba h743 with bmi270 for the second month now... with disabled bidirect and rpm filtering.  Who cares, then the video of this flight - https://youtu.be/JFHaNnA-Krg

howels, 2022-02-21:
> Meanwhile... I've been flying mamba h743 with bmi270 for the second month now... with disabled bidirect and rpm filtering. Who cares, then the video of this flight - https://youtu.be/JFHaNnA-Krg

Not arguing about how good or bad RPM filter is, trying to get all features working so 4.3 works no matter what options users choose to enable.

Maxis666, 2022-02-22:
> > А пока... Второй месяц летаю на mamba h743 с bmi270... с отключенным бидиректом и фильтрацией оборотов. Кому интересно, то видео этого полета - https://youtu.be/JFHaNnA-Krg
> 
> Не споря о том, насколько хорош или плох RPM-фильтр, я пытаюсь заставить работать все функции, чтобы 4.3 работал независимо от того, какие опции пользователи выбирают для включения.

Certainly! We all can't wait for you to fix everything. I personally have been looking forward to it.

justinflipflops, 2022-02-28:
Not only does it enable rpm filtering but other features that are RPM based like dynamic idle.

MoppelMat, 2022-03-07:
> It would be very useful to capture the MOTOR waveforms on a bad FC/ESC combo as I have done above.

I have got a fully ulocked Siglent SDS2104X Plus 4Ch. No logic analyzer cable though... May I help with that? 

I have a Mamba 743 with BMI270 Stack in the mail to me.

x4FF3, 2022-03-07:
Hi,

as i've got a MambaH7 laying around and noticed the same problem, i looked at the signal with my DS212 oscilloscope. Motor 4 which has the problem with DShot, gets this signal: 
[image]

all other 3 motors get this signal:
[image]

Also ESC/Motor 4 does restart randomly. I don't know if this helps in any way.

hydra, 2022-03-07:
interesting! seems like the rise time of Motor 4 output isn't as good.  do you have any more zoomed-in scope traces showing the issue in more detail.  Have you reached out to the manufacturers of the ESCs and FCs for comment?

x4FF3, 2022-03-07:
Actually i did, as these screenshots were taken last week already, just didn't have time to post them. Here is a pic they send me today:
[image]

Seems on a proper oscilloscope the signal starts to rise earlier. Also they stated, they tested this already with multiple boards. All the same problem. For ESC i can test another one, but may take some time.

ojwright845, 2022-03-09:
> @SteveCEvans thanks for the more in-depth explanation of what is going on in the code. I looked at the cost of one of those logic analyzers and it's a bit outside of my kind of budget for now. [...]

I have same setup as you, same issue. Can you please tell me what blheli32 version? I am running 32.8.3 testcode. You say switching to motor outputs 5-8 is a workaround, and I agree that should be fine for most people with 4 motors. Thoughts on setting resources a00 - a03 to none or leave as default? I only encountered the loss of 2 motor error after attempting to remap. On defaults it seems to work normally.
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 A01
resource MOTOR 4 A00

ojwright845, 2022-03-09:
+1 with same problem. Dual mpu6000 version. F55 ESC. Occasionally all 4 motors work but upon reboot 2 are gone again.

MoppelMat, 2022-03-09:
I have got my FC and ESC today and would like to help with my Scope. 

Just to clarify: I am supposed to measure at the DShot pins going from the FC to the ESC correct? I am asking because I can not see how that has anything to do with the ESC attached. It is just the FC or not?

I have 4 channels, what should I measure? I saw a ESC-Sensor measurement from @LordGG ...

[56 more replies]

PR #11494 (2022-04-01): Fix [Bidirectional] DShot issues on G4, F7 and H7 by reducing GPIO speed
Fixes: #11155
closed 2022-04-01 by ctzsnooze