betaflight/betaflight #11156: BF 4.3 RC1 GPS coord not shown

AndreaCCIE, 2021-12-24:
### Describe the bug
After having installed the RC1 and got a GPS fix, I noticed the coordinates are not shown on the OSD.
The distance calculation, speed and altitude work correctly.
Connecting the FC to BF, I can see the coordinates are properly reported.

Please see the video:
https://youtu.be/bX-UmM2o8mA

### To Reproduce
Install RC1 on a quad with GPS

### Expected behavior
coordinates should be reported

### Flight controller configuration
```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Dec 23 2021 / 19:24:33 (5933d96bc) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: ec268d11, date: 2021-07-26T13:10:45Z

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id DIAT

# name: LR4

# feature
feature -SOFTSERIAL
feature GPS

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 2 115200 115200 0 115200

# beeper
beeper -RX_LOST_LANDING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# aux
aux 0 0 0 1725 2100 0 0
aux 1 1 3 900 1300 0 0
aux 2 46 4 1700 2100 0 0
aux 3 13 3 1700 2100 0 0
aux 4 26 1 1300 1700 0 0
aux 5 31 1 1700 2100 0 0
aux 6 36 5 1700 2100 0 0
aux 7 39 2 900 1300 0 0

# adjrange
adjrange 0 0 2 900 1325 29 2 0 0
adjrange 1 0 2 1350 1650 29 2 0 0
adjrange 2 0 2 1800 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 800
vtxtable powerlabels 25 100 200 400 800

# master
set gyro_lpf1_static_hz = 500
set gyro_lpf2_static_hz = 0
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = -100,-4,100,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/2
set blackbox_disable_mag = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set motor_output_reordering = 3,2,1,0,4,5,6,7
set failsafe_delay = 3
set failsafe_procedure = GPS-RESCUE
set failsafe_stick_threshold = 15
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 260
set vbat_warning_cell_voltage = 280
set ibata_scale = 156
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 40
set gps_rescue_descent_dist = 150
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_max = 1800
set gps_rescue_ascend_rate = 300
set gps_rescue_throttle_hover = 1400
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set deadband = 6
set yaw_deadband = 6
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 200
set report_cell_voltage = ON
set osd_warn_launch_control = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_dbm_alarm = -100
set osd_cap_alarm = 2500
set osd_tim1 = 2049
set osd_tim2 = 1
set osd_vbat_pos = 97
set osd_link_quality_pos = 14391
set osd_link_tx_power_pos = 14423
set osd_tim_1_pos = 14369
set osd_tim_2_pos = 215
set osd_remaining_time_estimate_pos = 193
set osd_flymode_pos = 432
set osd_throttle_pos = 12385
set osd_vtx_channel_pos = 2132
set osd_crosshairs_pos = 14541
set osd_current_pos = 8609
set osd_mah_drawn_pos = 12737
set osd_craft_name_pos = 364
set osd_gps_speed_pos = 14550
set osd_gps_lon_pos = 8655
set osd_gps_lat_pos = 8623
set osd_gps_sats_pos = 14817
set osd_home_dir_pos = 14414
set osd_home_dist_pos = 14831
set osd_flight_dist_pos = 12790
set osd_compass_bar_pos = 14378
set osd_altitude_pos = 14518
set osd_pid_roll_pos = 2407
set osd_pid_pitch_pos = 2439
set osd_pid_yaw_pos = 423
set osd_warnings_pos = 14505
set osd_avg_cell_voltage_pos = 14401
set osd_disarmed_pos = 12523
set osd_esc_tmp_pos = 428
set osd_log_status_pos = 440
set osd_rate_profile_name_pos = 151
set osd_pid_profile_name_pos = 2401
set osd_efficiency_pos = 12774
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_profile = 3
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 5
set vtx_freq = 5806
set vcd_video_system = PAL
set gyro_1_align_yaw = 1800
set name = LR4

profile 0

# profile 0
set p_pitch = 70
set i_pitch = 126
set d_pitch = 56
set f_pitch = 93
set p_roll = 67
set i_roll = 120
set d_roll = 49
set f_roll = 90
set p_yaw = 67
set i_yaw = 120
set f_yaw = 90
set d_min_roll = 49
set d_min_pitch = 56
set auto_profile_cell_count = 4
set thrust_linear = 25
set feedforward_averaging = 3_POINT
set feedforward_jitter_factor = 12
set dyn_idle_min_rpm = 20
set simplified_master_multiplier = 150
set simplified_d_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 50

rateprofile 1

# rateprofile 1
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 40
set tpa_breakpoint = 1550

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 2
A09: MOTOR 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3 : CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3

# 
```

### Setup / Versions
Mamba F722
BN-220

limonspb, 2021-12-26:
@AndreaCCIE  do you mind going back in commits history (git bisect), flash them and see which PR/change actually started the issue? I believe recently it was working OK on the nightly builds?

if you can't build BF, maybe you can just go back on nightly builds? That would massively help.

Thanks!

AndreaCCIE, 2021-12-26:
@limonspb I don't know how to build a specific PR related build, but will test some of the latest night builds and update here

AndreaCCIE, 2021-12-26:
@limonspb 2579 works, 2580 doesn't

IllusionFpv, 2021-12-26:
It seems related to this https://github.com/betaflight/betaflight/pull/10921

AndreaCCIE, 2021-12-26:
Yes, it looks like @TonyBlit didn’t have all the hardware for regression tests, and no-one else did notice this issue.

atlaware, 2021-12-26:
I confirm this bug for my baby crocodile analogic, board: geprc411-aio, firmware bf 4.3rc1 (gps osd coord working before, with BF 4.2.3).
On OSD I can see altituide, sat N°, but coordinates is always 0.0000 or 0.0000000, no matter how decimal I select in configurator.
If I connect the board to BF configurator, I see coordinates correctly on GPS tab.

Thank you.

limonspb, 2021-12-27:
Hey, @TonyBlit could you please have a look at this issue? Thanks!

daleckystepan, 2021-12-27:
@limonspb @TonyBlit We have fix already: #11168

PR #11168 (2021-12-28): fix for gps coordinates not shown on OSD
Fixes #11156

There is a flag for preventing coordinates to show on OSD until they are valid, and I wasn't updating it.
closed 2021-12-28 by ctzsnooze