betaflight/betaflight #11157: Failure to boot caused by re-use of SPI pins.

hydra, 2021-12-24:
### Describe the bug
Firmware fails to boot when a later unused SPI port uses the same pins as a used SPI port.

### To Reproduce
Configure any H7 target to use PB3/PB4/PB5 for SPI3 and allocate the same pins for SPI6.
Likely also using SPI1 and SPI3 would also exhibit the same issue.

[image]

```
resource SPI_SCK 3 B03
resource SPI_MISO 3 B04
resource SPI_MOSI 3 B05
set gyro_1_spibus = 3
```

after disabling the gyro autodection code that fails, `resource show` still says:
```
B03: SPI_SCK 6
B04: SPI_MISO 6
B05: SPI_MOSI 6
```

`resource` shows:
```
resource SPI_SCK 2 D03
resource SPI_SCK 3 B03
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
```

[image]

after it calls spiInit(SPIDEV_3) the AF's are correct.

[image]

after the call to spiInit(SPIDEV_6) the AF's are incorrect.

[image]

The AF's and pins are correct for both devices, but attempting to use SPI3 for the Gyro results in a lock-up as the SPI read never completes as the GPIO pins have the wrong AF now.

Likely the SPI device should not be initialised if the pins are used by another SPI device as per the resource configuration.

The default resource configuration in the target appears as follows:
```
resource SPI_SCK 2 D03
resource SPI_SCK 3 B03
resource SPI_SCK 6 B03
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MISO 6 B04
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 6 B05
```

The target.h file has this:
```
// SPI3 and SPI6 use the same pins however SPI6 supports independent clock configuration, thus SPI6 used to allow separation between SX1280 and Gyro SPI clocks.
#define USE_SPI_DEVICE_6
#define SPI6_SCK_PIN            PB3
#define SPI6_MISO_PIN           PB4
#define SPI6_MOSI_PIN           PB5
#define SPI6_NSS_PIN            PA15

#define USE_SPI_DEVICE_3
#define SPI3_SCK_PIN            PB3
#define SPI3_MISO_PIN           PB4
#define SPI3_MOSI_PIN           PB5
#define SPI3_NSS_PIN            PA15
```

### Expected behavior
FC shouldn't lockup, SPI3 should work.

However, I'm not 100% certain if it's intended that target definitions can define the same pin for multiple ports.

### Flight controller configuration
```
# version
# Betaflight / SPRACINGH7RF (SP7R) 4.3.0 Dec 24 2021 / 15:02:17 (norevision) MSP API: 1.44

# start the command batch
batch start

board_name SPRACINGH7RF

profile 0

rateprofile 0

# end the command batch
batch end
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
Code use for testing is from https://github.com/betaflight/betaflight/pull/11110#issuecomment-1000166588, which is based on a recent `master` which doesn't have any resource allocation / spi initialisation changes that I'm aware of.  I verified that by comparing the source tree and looking at SPI related/resource files like pg/bus_spi.c, drivers/bus_spi*, drivers/io.c, etc.

hydra, 2021-12-24:
This sequence also results in the same issue:

```
defaults
resource SPI_SCK 3 NONE
resource SPI_MISO 3 NONE
resource SPI_MOSI 3 NONE
resource SPI_SCK 6 NONE
resource SPI_MISO 6 NONE
resource SPI_MOSI 6 NONE

diff

# version
# Betaflight / SPRACINGH7RF (SP7R) 4.3.0 Dec 24 2021 / 15:02:17 (norevision) MSP API: 1.44

# start the command batch
batch start

board_name SPRACINGH7RF

# resources
resource SPI_SCK 3 NONE
resource SPI_SCK 6 NONE
resource SPI_MISO 3 NONE
resource SPI_MISO 6 NONE
resource SPI_MOSI 3 NONE
resource SPI_MOSI 6 NONE

profile 0

rateprofile 0

# end the command batch
batch end

save
```

FC reboots, enter CLI:

```
resource SPI_SCK 3 B03
resource SPI_MISO 3 B04
resource SPI_MOSI 3 B05
set gyro_1_spibus = 3
save
```

`resource show`:

```
B03: SPI_SCK 6
B04: SPI_MISO 6
B05: SPI_MOSI 6
```
^ incorrect, should be as follows:
```
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
```

hydra, 2021-12-24:
On further investigation it seems this issue occurs because the target also uses the same initialisation sequence that `CONFIG_IN_EXTERNAL_FLASH` uses which results in a call to `configureSPIBusses` before `initEEPROM`.

When using (`CONFIG_IN_EXTERNAL_FLASH`) or (`CONFIG_IN_SDCARD` and `USE_SDCARD_SPI`).
Two solutions present themselves:
1) don't specify the same pins for 2 SPI devices.
2) after config is loaded, reset the pins uses for the device that loaded the config to the defaults, ensure those pins aren't used by another device, and if they are remove them from the other device.

In the case of the H7RF, SPI isn't used to load the config, OctoSPI is, so changing the SPI pins should be allowed.

When work more on #11110 I'll investigate this further.

hydra, 2021-12-24:
This can be rescheduled for 4.4.

In the mean time I developed a fix for the H7RF, here:

https://github.com/spracing/betaflight/commit/911c0401a2cb26bfbb6307c1ae181df6d644c025

SteveCEvans, 2021-12-26:
@hydra Could you please explain why you define the same pins for bus 3 and bus 6.

hydra, 2021-12-27:
> @hydra Could you please explain why you define the same pins for bus 3 and bus 6.

So that you can switch between them using just the `gyro_1_spibus` and there are appropriate defaults for the pins, I may need to refresh my memory on whether that was intended use of the resource management system.

Additional background:

Different DMA controllers are used for SPI1/3 and SPI6 so it's useful to be able to balance the DMA resources and clock-speed requirements.  On the H7RF SPI6 can be clocked at exactly 20Mhz because it has a different clock mux than SPI1/3 and uses a different DMA channel, so for the best utilization of the DMA controllers and the lowest latency gyro spi transfers SPI6 should be used for the Gyro, but currently there's no DMA support for SPI6.

hydra, 2022-06-17:
Gonna close this as my expectations don't match the intended design.

closed 2022-06-17 by hydra