betaflight/betaflight #11159: BADRX BetaFlight 4.3.0-RC1 with TBS Tracer

RPassanha, 2021-12-24:
### Describe the bug
After upgrading from BT 4.2 to 4.3 the BADRX message is appearing and it is not possible to arm the engines. The RSSI value is varying from 50 to 0 zero, after a few attempts, it is possible to arm the engines and make the flight. During the flight, the message does not happen.

### To Reproduce
It happens every time before taking off

### Expected behavior
Betaflight 4.2 it doesn't happen.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
# 

# diff

# version
# Betaflight / STM32F405 (S405) 4.3.0 Dec 24 2021 / 18:29:45 (norevision) MSP API: 1.44
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: Bee140

# serial
serial 2 2048 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1300 2100 0 0
aux 1 13 3 1700 2100 0 0
aux 2 35 3 1200 1600 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 0 1 2 3 4
vtxtable powerlabels 25 100 200 400 600

# master
set acc_calibration = -47,173,-448,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set small_angle = 180
set pid_process_denom = 4
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_rssi_pos = 2105
set osd_link_quality_pos = 2098
set osd_tim_2_pos = 2454
set osd_vtx_channel_pos = 2083
set osd_crosshairs_pos = 2253
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2435
set osd_rate_profile_name_pos = 2443
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set gyro_1_align_yaw = 2700
set name = Bee140

profile 0

# profile 0
set p_pitch = 49
set i_pitch = 88
set d_pitch = 57
set f_pitch = 131
set p_roll = 47
set i_roll = 84
set d_roll = 50
set f_roll = 125
set p_yaw = 47
set i_yaw = 84
set f_yaw = 125
set d_min_roll = 37
set d_min_pitch = 42
set simplified_master_multiplier = 105
set simplified_d_gain = 120

rateprofile 0

# rateprofile 0
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 90
set pitch_srate = 90
set yaw_srate = 90

# end the command batch
batch end

# ```
```
PASTE THE OUTPUT OF 'resource show all' HERE
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 3
TIM3:
    CH3 : MOTOR 1
    CH4 : MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# ```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:
FC OMNIBUSF4SD

klutvott123, 2021-12-25:
The only way I'm able to reproduce this with 4.3 RC1 is if I have the radio connected with the arming switch on when power is connected to FC/RX. This is with an aikonf7 and tracer

RPassanha, 2021-12-25:
This is happening because the RSSI signal is dropping to zero several times. Why does this happen?

RPassanha, 2021-12-25:
https://user-images.githubusercontent.com/33128681/147395357-4df71e52-a968-49af-8b7d-76e8e9c9bacc.mp4

you can check the problem in the video

klutvott123, 2021-12-26:
Well that's strange. Is the rx failsafe set to cut or position, and have you saved channel positions for failsafe? BADRX happens when the arm switch is on after it recovers from a failsafe. Why it's failsafing in the first place is hard to say. Could be interference.

haslinghuis, 2021-12-26:
Video is showing RSSI low. Something bad with your antenna?

RPassanha, 2021-12-26:
It doesn't happen with BF 4.2, same quad, same setup. There is no problem with the antenna or other hardware.

RPassanha, 2021-12-26:
https://user-images.githubusercontent.com/33128681/147403281-0018a01a-25d4-46b1-a507-6ee0dfd255be.mp4

Now BF4.2 is installed in the same quad.

RPassanha, 2021-12-26:
I have tested the 2594 and 2595 firmware.
2594 is ok, but in 2595 the problem happened.

closed 2021-12-30 by RPassanha

limonspb, 2022-01-17:
@RPassanha could you please share why the issue is closed?