betaflight/betaflight #11161: GPS Cordinate Error / Not Shown

utharemo, 2021-12-25:
### Describe the bug

### To Reproduce

### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: OmnibusV6 F4
 
 - Other components: Happymodel ELRS 915, TBS Unify HV, Iflight GPS
 
 - How are the different components wired up: RX on Uart1, VTX on Uart2, GPS on Uart3

AndreaCCIE, 2021-12-25:
This is what I reported in #11156

utharemo, 2021-12-25:
Thank you... hope this issue will solve soon

aowi7280, 2021-12-26:
It works OK on Omnibusf4SD

IllusionFpv, 2021-12-26:
Duplicated if https://github.com/betaflight/betaflight/issues/11156, closing

utharemo, 2021-12-26:
Do you using the omnibusV6 with Omnibusf4SD firmware?

closed 2021-12-29 by haslinghuis