betaflight/betaflight #11163: FrSky F.Port Protocol not working on KAKUTEF7 - 4.3.0 RC1

b14ckyy, 2021-12-25:
### Describe the bug
KAKUTEF7 Target does not work with Fport protocol. No RC Signal and no telemetry. Works fine on 2.3.9 but not on 2.4.0 RC1

### To Reproduce
FC: Holybro Kakute F7 V1
Rx: FrSky R-XSR - ACCESS Firmware - F.Port protocol
- Configure INAV 2.3.9 
- Serial RX on UART6 Tx Pad
- Protocol set to FrSky Fport
- - - - RC Channels and Telemetry works flawlessly
- Flash 4.3 RC1 with full chip erase
- Setup same UART6 and Fport
- - - - No RC Channel movement and no Telemetry

### Expected behavior
FrSky F.Port should work 

### Flight controller configuration
#DIFF of 2.3.9 working: 
```
diff all

# version
# Betaflight / STM32F745 (S745) 4.2.9 Apr 27 2021 / 19:34:04 (e097f4ab7) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7
manufacturer_id HBRO
mcu_id 002d00343438510333343635
signature 

# name: Scarlet

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# beeper
beeper -ON_USB

# serial
serial 0 2048 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 46 3 1700 2100 0 0
aux 3 13 5 1700 2100 0 0
aux 4 35 1 900 1300 0 0
aux 5 49 3 900 1300 0 0

# adjrange
adjrange 0 0 6 900 2100 29 6 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
[... 411 lines cut ...]
Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# ```

### Setup / Versions
 - Flight controller: Holybro Kakute F7 V1
 - Other components: BN220 GPS on COM5; Tekko F7 45A ESC; R-XSR Receiver single wire non-inverted F.Port
 - How are the different components wired up: FC and ESC stacked with plug in cable; Camera and VTX go to corresponding pads. VTX Powered from on board VTX regulator.

daleckystepan, 2021-12-25:
Try on BF 4.3.0 RC1
```
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
```

b14ckyy, 2021-12-25:
Oh dear that works.... I totally forgot about the inversion setting. I am used to have that in the configurator in INAV but betaflight hasn't. Sorry for that.

closed 2021-12-25 by b14ckyy