betaflight/betaflight #11165: No post flight statistics BF4.3RC1

aowi7280, 2021-12-26:
### Describe the bug
I have post flight statistics set up, but they do not show after the flight. Flew 6-8 flights, different batteries.

### To Reproduce
Flash BF4.3RC1, configure, fly, disarm, see no post flight stats

### Expected behavior
Flight statistics should show after a flight.

### Flight controller configuration

```
# Betaflight / STM32F405 (S405) 4.3.0 Dec 24 2021 / 18:29:45 (norevision) MSP API: 1.44
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# resources
resource MOTOR 5 NONE
resource SONAR_TRIGGER 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A01
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 38400 0 115200
serial 30 2048 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# led
led 0 6,2::A:6
led 1 6,1::A:6
led 2 6,0::A:6
led 3 7,0::A:6
led 4 7,1::A:6
led 5 7,2::A:6
led 6 8,2::A:6
[... 351 lines cut ...]
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: Airbot Omnibus F4SD
 
 - Other components: Analog camera, R9mm rx, Rush Tank Solo Vtx, Holybro micro GPS, Xing 2207 2450KV, RobotDog ESCs
 
 - How are the different components wired up: Normal. 

Been flying BF4.3 nightly for 6 months ver 2553 was most recent. Just flashed BF4.3RC1 and noticed no post flight stats.

bike797, 2021-12-26:
I have a same issue.
I tsted some firmware version, but all version have issue.
 - FC : Happymodel ELRS X1 AIO 2-4S Flight controller built-in SPI 2.4G ELRS and 12A ESC for Toothpick
 - Firmware  Tested:  
   - betaflight_4.3.0_CRAZYBEEF4SX1280_7c9e5ac8e.zip https://github.com/betaflight/betaflight/pull/11154
   - Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Dec 22 2021 / 23:09:32 (4f0d4b81b) MSP API: 1.44
   - Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Dec 24 2021 / 11:15:50 (5933d96bc) MSP API: 1.44

RPassanha, 2021-12-26:
It doesn't happen with BF 4.2, same quad, same setup. There is no problem with the antenna or other hardware.

daleckystepan, 2021-12-26:
@bike797 @aowi7280 Can you please retest with these FW in the order (you can stop on the first that will not work for you)?

[image]

17-12-2021 ... # 2569
17-12-2021 ... # 2570
20-12-2021 ... # 2575

bike797, 2021-12-26:
17-12-2021 ... # 2569 (OK)
17-12-2021 ... # 2570 (OK)
20-12-2021 ... # 2575 (NG)

daleckystepan, 2021-12-26:
@SteveCEvans Can you please check https://github.com/betaflight/betaflight/pull/11033 regarding OSD post flights statistics ? Thank you.

SteveCEvans, 2021-12-26:
Please try the below images.
[betaflight_4.3.0_STM32F411_ff9d2d2.hex.zip](https://github.com/betaflight/betaflight/files/7777382/betaflight_4.3.0_STM32F411_ff9d2d2.hex.zip)
[betaflight_4.3.0_STM32F405_ff9d2d2.hex.zip](https://github.com/betaflight/betaflight/files/7777383/betaflight_4.3.0_STM32F405_ff9d2d2.hex.zip)

Asizon, 2021-12-26:
Tested on Betafpv F411 toothpick board with 120mhz overclock works really good, also i can see postflight stats. with 108mhz looks a bit slowly.

aowi7280, 2021-12-26:
With the F405 version I don't get video at all. No OSD either. I even reflashed 4.2.11 and used one of my old Diff files, the video is still dead.

Asizon, 2021-12-26:
Then you should revise you video system or configuration, make sure to make test with defaults, no diff.

aowi7280, 2021-12-26:
it was working just fine until I put that fw in.

Asizon, 2021-12-26:
But what is for you no video at all? If you dont get video it is not firmware related.

Asizon, 2021-12-26:
Ok, to test this attached firmware, please select first your board in the selector and then select the local file.

aowi7280, 2021-12-26:
That is what I did. Something in the vtx settings changed.

k-trahan, 2021-12-27:
I have noticed the same issue with no post flight statistics using BETAFPVF4SX1280 4.3.0-RC1.

In addition, Low Power Disarm no longer works after the quad has been armed the first time; this started happening the same time I noticed the post flight statistics issue. Upon powering up, the VTX is at 25 mW, after arming it goes to 350 mW (as set by `set vtx_power = 4`), and after disarming stays at 350 mW. It's almost as if Betaflight isn't registering a disarmed status anymore.

aowi7280, 2021-12-27:
I wired the camera straight to the Vtx and I get a picture (no osd obviously) and there is snow when I unplug the battery. With the camera going through the FC, it is just a gray screen. I can't even upload fonts in the OSD tab because the Font manager is not available. It looks like it killed the OSD chip.

k-trahan, 2021-12-27:
Also see https://github.com/betaflight/betaflight/pull/11154#issuecomment-1000968150 that may relate to my statement "It's almost as if Betaflight isn't registering a disarmed status anymore.":

> A couple other things is my vtx wouldn't switch out of pit mode with high cpu percentage and when arming it doesn't say armed on osd anymore.
>
> I know 4.3 says cpu is a lot higher but every problem I mentioned gets fixed with a lower cpu % beside the arming text on osd.

SteveCEvans, 2021-12-27:
> With the F405 version I don't get video at all. No OSD either. I even reflashed 4.2.11 and used one of my old Diff files, the video is still dead.

Nothing to do with the firmware then. You have a hardware issue.

bike797, 2021-12-27:
Build with https://github.com/SteveCEvans/betaflight/commit/ff9d2d29aa80991574b1999a7a2d319d459b5745 
It works normaly. Flight stats are back.
 - CRAZYBEEF4FR
 - CRAZEBEEF4SX1280

aowi7280, 2021-12-28:
> > With the F405 version I don't get video at all. No OSD either. I even reflashed 4.2.11 and used one of my old Diff files, the video is still dead.
> 
> Nothing to do with the firmware then. You have a hardware issue.

I'm still wondering how new firmware can kill the hardware. Its too much of a coincidence. The OSD and video worked fine. I never even flew the quad, then flashed in your betaflight_4.3.0_STM32F405_ff9d2d2.hex.zip Then the video and osd were not working.

haslinghuis, 2021-12-29:
Fixed in https://github.com/betaflight/betaflight/pull/11169

closed 2021-12-29 by haslinghuis