betaflight/betaflight #11167: Matek F411 OSD slow in refresh from Nightly 2575

AndreaCCIE, 2021-12-26:
### Describe the bug
Starting Nightly 2575 the OSD on my MATEK F411 is laggy/slow to refresh and unusable. All good with 2574.

Please see the video below:
https://youtu.be/WArIzWm4tuA

### To Reproduce
Flash Nightly 2575

### Expected behavior
OSD to works at the correct refresh rate

### Flight controller configuration
```
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Dec 17 2021 / 00:07:16 (a63172cc1) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF411, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411
manufacturer_id MTKS
mcu_id 005000293439510738363239
signature 

# name: GEPPINO

# feature
feature -SOFTSERIAL

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# beeper
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# aux
aux 0 0 0 1725 2100 0 0
aux 1 1 3 900 1300 0 0
aux 2 13 3 1700 2100 0 0
aux 3 36 5 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 1325 29 2 0 0
adjrange 1 0 2 1350 1650 29 2 0 0
adjrange 2 0 2 1800 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5665 5885 5905 5905 5905
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 100 200
vtxtable powerlabels 25 100 200

# rxfail
rxfail 6 s 750
rxfail 7 s 750

# master
set gyro_lpf1_static_hz = 500
set gyro_lpf2_static_hz = 0
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = -4,72,179,1
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set align_board_yaw = 90
set yaw_motors_reversed = ON
set simplified_gyro_filter_multiplier = 200
set report_cell_voltage = ON
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_rssi_dbm_alarm = 0
set osd_tim1 = 2561
set osd_vbat_pos = 97
set osd_link_quality_pos = 14391
set osd_link_tx_power_pos = 14795
set osd_rssi_dbm_pos = 12375
set osd_tim_1_pos = 14369
set osd_tim_2_pos = 215
set osd_flymode_pos = 12792
set osd_throttle_pos = 8289
set osd_vtx_channel_pos = 12779
set osd_current_pos = 8640
set osd_mah_drawn_pos = 12770
set osd_craft_name_pos = 364
set osd_pid_roll_pos = 391
set osd_pid_pitch_pos = 359
set osd_pid_yaw_pos = 423
set osd_warnings_pos = 14505
set osd_avg_cell_voltage_pos = 14401
set osd_disarmed_pos = 12555
set osd_esc_tmp_pos = 428
set osd_log_status_pos = 472
set osd_rate_profile_name_pos = 8343
set osd_pid_profile_name_pos = 8311
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set osd_stat_avg_cell_value = ON
set osd_menu_background = GRAY
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5806
set vcd_video_system = PAL
set gyro_1_align_yaw = 1800
set name = GEPPINO

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_type = PT3
set dterm_lpf1_static_hz = 101
set dterm_lpf2_static_hz = 0
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 65
set i_pitch = 117
set d_pitch = 61
set f_pitch = 175
set p_roll = 63
set i_roll = 112
set d_roll = 54
set f_roll = 168
set p_yaw = 63
set i_yaw = 112
set f_yaw = 168
set abs_control_gain = 20
set d_min_roll = 54
set d_min_pitch = 61
set thrust_linear = 25
set feedforward_averaging = 2_POINT
set simplified_master_multiplier = 140
set simplified_d_gain = 130
set simplified_dmax_gain = 0
set simplified_dterm_filter_multiplier = 135

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 3
set pitch_rc_rate = 3
set yaw_rc_rate = 3
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 90
set pitch_srate = 90
set yaw_srate = 55
set tpa_breakpoint = 1550

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
Matek F411

SteveCEvans, 2021-12-26:
Please try the image below and be sure to set `cpu_overclock=120MHz`.
[betaflight_4.3.0_STM32F411_4a2d6f0.hex.zip](https://github.com/betaflight/betaflight/files/7777284/betaflight_4.3.0_STM32F411_4a2d6f0.hex.zip)

AndreaCCIE, 2021-12-26:
Hi @SteveCEvans, it's slightly (.. slightly ..) better but still... it seems to be in an arcade video game from the 80s :-)

SteveCEvans, 2021-12-26:
profile 3 seemed to be the most problematic and that's working fine for me on a MATEKF411 now. Did you set the CPU overclock?

SteveCEvans, 2021-12-26:
@AndreaCCIE Don't knock 1980's video games; note the name of the author... https://youtu.be/oeqe-hq3TdU

SteveCEvans, 2021-12-26:
@AndreaCCIE  I've tested this on a BetaFPV75X which has a MATEKF411 compatible FC, with `cpu_overclock=120MHz` using your OSD profiles and it seems fine to me, both disarmed and armed.

AndreaCCIE, 2021-12-26:
> Did you set the CPU overclock?

Yes, here's the video while switching OSD profile:
https://youtu.be/JeqvFXUfjOs

0crap, 2021-12-27:
> Please try the image below and be sure to set `cpu_overclock=120MHz`. [betaflight_4.3.0_STM32F411_4a2d6f0.hex.zip](https://github.com/betaflight/betaflight/files/7777284/betaflight_4.3.0_STM32F411_4a2d6f0.hex.zip)

Just as a FYI.
The overclock should not be mandatory.
I have a MATEKF411RX board which drops from the sky regularly, if I use that overclock setting.
(EMAX Tinyhawk 2 Freestyle.)

ctzsnooze, 2021-12-27:
I guess there must be some that don't cope with it, but they are vanishingly uncommon.

AndreaCCIE, 2021-12-27:
@0crap does this fix your th2? I still have laggy OSD

SteveCEvans, 2021-12-27:
Please provide the OSD config which shows the issue.

0crap, 2021-12-27:
> @0crap does this fix your th2? I still have laggy OSD

I'm sorry if my comment did confuse you.
I never had the laggy OSD issue you are seeing.
Not with 4.2 and not with 4.3 (RC1)
Hope your issue can be fixed.

I just read about the overclock setting that triggered me.
It is not always save to use that.
Even if it is "vanishingly uncommon", it should not be suggested without a warning.
I tried and it dropped from the sky...

haslinghuis, 2021-12-28:
@0crap guess we could keep the OC optional as to fit configurations which needs it.

0crap, 2021-12-28:
> @0crap guess we could keep the OC optional as to fit configurations which needs it.

Yes, I would definitely recommend to do something like that for the [4.3 Tuning Notes](https://github.com/betaflight/betaflight/wiki/4.3-Tuning-Notes).

knallex, 2021-12-28:
I do have same laggy OSD on MATEK F411 with Betaflight 4.3 RC1, but Nightly 2574 too. (tried 2597 also)
Although it is only present when having Bidirectional DShot active.
CPU load is at 31% with Bidirectional Dshot OFF.
CPU load is at 59% with Bidirectional Dshot ON.

haslinghuis, 2021-12-28:
CPU load is normal. Please check https://github.com/betaflight/betaflight/wiki/4.3-Tuning-Notes#More-accurate-loop-times

0crap, 2021-12-28:
> I do have same laggy OSD on MATEK F411 with Betaflight 4.3 RC1, but Nightly 2574 too. (tried 2597 also) Although it is only present when having Bidirectional DShot active. CPU load is at 31% with Bidirectional Dshot OFF. CPU load is at 59% with Bidirectional Dshot ON.

I don't recommend to use rpm filtering on a F411 because it taxes the CPU too much. (Just my opinion.)
Anyway, I see the same CPU values as you do.
I only use the (improved) dynamic notch on my F411, works perfect.
However I can't reproduce the laggy OSD. Not with the dynamic notch, nor with Bidirectional DShot active.
¯\_(ツ)_/¯

haslinghuis, 2021-12-28:
Works fine for me @0crap on Matek F411, it can be different on other boards or using high voltage (more noise filtering)

0crap, 2021-12-28:
> Works fine for me @0crap on Matek F411, it can be different on other boards.

I might give it another try on 4.3
But the improved dynamic notch suits my needs on the Tinyhawk 2 Freestyle.

knallex, 2021-12-28:
I also noticed that OSD freezes when end stats should be showing. This also happens without OSD lagging. (RPM Filtering OFF)
CMS works fine.
I'm testing on a GEP-12A-F4 FC from GEPRC.

etracer65, 2021-12-28:
Recommending that people overclock the F411 to 120MHz is risky and ill-advised in my opinion. I've seen lots of F411 boards that won't run with stability at this speed - particularly in warmer weather. 120MHz is a 25% overclock over the design specs for the MCU and really pushing the envelope. In 4.2 I set the default for F411 to a mild 108MHz overclock to increase loop stability for 8/4/rpm but even then I was reluctant to do so. Fortunately most F411 seem to handle the 12.5% overclock (still rare failures) but that's not going to be true with a 25% overclock. Of course people are going to reply with "I do it all the time" and "never had a problem", so whatever. And seems strange that such an overclock is now required when it wasn't previously. Probably should look into and correct that rather than just recommend everybody overclock.

AndreaCCIE, 2021-12-28:
> And seems strange that such an overclock is now required when it wasn't previously.

Agree, if F411 can’t handle BF4.3, or any of its features, it should be clear and written in the documentation

AndreaCCIE, 2021-12-28:
> Please provide the OSD config which shows the issue.

@SteveCEvans it’s the dump I posted at the beginning, OSD profile 2.

SteveCEvans, 2021-12-28:
The issue here is transferring data from the shadow buffer to the MAX7456 frame buffer. When the CPU is overclocked the MAX7456 SPI bus speed is halved, although `max7456_clock=FULL` can help with that. In polled mode the transfer of data is particularly slow and bumping up the clock to 120MHz gives things just enough breathing room. In 4.2 the OSD task way overran its time slot to the detriment of flight performance, just as logging to FLASH used to.

SteveCEvans, 2021-12-28:
@AndreaCCIE With defaults apart from your OSD config and enabling bi-directional DSHOT I see the slow rendering. Setting `max7456_clock=FULL` fixes this for me. Please give that a try as an alternate to raising the CPU overclock. Looking at the logic behind `max7456_clock` I don't actually see it adding much value. The max SPI clock speed for the MAX7456 is 10MHz. Based on a 100MHz clock, the divisor used will be 6.25MHz. Overclocked to 120MHz, this will still be under 10MH at 7.5MHz. Halving that as the current code is doing is pointless.

I'll look into this in more detail and probably raise a PR to remove the `max7456_clock` setting.

knallex, 2021-12-29:
Ok the Nightly #2598 resolves the problems for me at least. No laggy OSD anymore and end stats are showing up when disarming. Thanks.

closed 2021-12-29 by haslinghuis

P-I-Engineer, 2021-12-30:
nightly 2598 has a slow home arrow update on f411. crazybeef4sx1280 build.

rescue still goes in the correct direction. but when i'm flying away it's point away.

SteveCEvans, 2021-12-30:
Do any other elements lag in this way? If not it’ll be the GPS information processing.

P-I-Engineer, 2021-12-30:
not that i noticed. maybe tx power, but it's elrs so i'm not sure.

gps was updating, don't know if accurately.

sjauquet, 2022-01-12:
Just to add my stone: OSD also laggy on BF 4.3 RC2 on GepRCF411(STM32F411)
RPM Filters on, DSHOT 300 CPU load: 28/30%
Thanks for your work :-)

SteveCEvans, 2022-01-12:
@sjauquet can you please set `dshot_bitbang=ON` and see if that helps. Unfortunately doing this will prevent data being DMAed from the gyro but it will still be locked to the scheduler. The upside is that the OSD will then use DMA which will fix any slowness there. Can you please provide `dma show` outputs before and after this setting and also `tasks` run twice.

sjauquet, 2022-01-12:
Hi @SteveCEvans Many thanks, yesterday, after writing this message, I foud the next "issue" opened about this problem and foud your post: https://github.com/betaflight/betaflight/issues/11244#issuecomment-1008037830
I applied it and it seems working (on my quad, the slow OSD bug was not all the time visible), but I didn't had the time to fly it yet because of the weather.
Thanks for your work !

sjauquet, 2022-01-12:
This is the actual status with commands with dshot_bitbang=ON
I have raised pid loop to 4khz with this conf, previously it was at 2kHz because I suspected overload cpu when flying with low osd refresh
i will reset config and re-apply a previous backup so you will have the previous state.

`# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       4       3    0.0%    0.0%         0      0    306       3
01 - (         SYSTEM)    991       5       2    0.4%    0.1%        29      2  27355       2
02 - (           GYRO)   7994      29      13   23.1%   10.3%      2430      0 243184       0
03 - (         FILTER)   3998      39      20   15.5%    7.9%      2361      0 121592       0
04 - (            PID)   3998      77      46   30.7%   18.3%      4924      0 121592       0
05 - (            ACC)    988      23      13    2.2%    1.2%       314     22  27304      13
06 - (       ATTITUDE)     99      16      13    0.1%    0.1%        31      1   2747      13
07 - (             RX)     15      78       3    0.1%    0.0%        36     19   1828      26
08 - (         SERIAL)    100  605689      31 6056.8%    0.3%      1141      0   2625      31
09 - (       DISPATCH)    992       8       3    0.7%    0.2%        24      7  27370       3
10 - (BATTERY_VOLTAGE)     50       6       4    0.0%    0.0%         3      0   1377       4
12 - ( BATTERY_ALERTS)      5       5       4    0.0%    0.0%         0      0    154       4
13 - (         BEEPER)    100       7       3    0.0%    0.0%         4      0   2623       3
14 - (            GPS)    120      48      38    0.5%    0.4%        42     38   3211      38
17 - (       ALTITUDE)     40       6       3    0.0%    0.0%         1      1   1099       3
18 - (      TELEMETRY)    496      46       2    2.2%    0.0%        37     80  13087       2
20 - (            OSD)     10      97      98    0.0%    0.0%        98    100   5037      98
22 - (            CMS)     10       5       4    0.0%    0.0%         0      0    335       4
23 - (        VTXCTRL)      3      82      27    0.0%    0.0%         1     16     68      27
24 - (        CAMCTRL)      3       3       1    0.0%    0.0%         0      0     68       1
26 - (    ADCINTERNAL)      2       5       4    0.0%    0.0%         0      0     15       4
28 - (SPEED_NEGOTIATION)     33       5       2    0.0%    0.0%         1      0   1122       3
RX Check Function                  19       7                        43
Total (excluding SERIAL)                                38.5%

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       1       3    0.0%    0.0%         0      0     17       3
01 - (         SYSTEM)    987       4       2    0.3%    0.1%        30      0   1694       2
02 - (           GYRO)   7994      21      12   16.7%    9.5%      2586      0  13742       0
03 - (         FILTER)   3998      28      20   11.1%    7.9%      2492      0   6871       0
04 - (            PID)   3998      53      46   21.1%   18.3%      5197      0   6871       0
05 - (            ACC)    986      15      13    1.4%    1.2%       333      0   1691      13
06 - (       ATTITUDE)    100      14      13    0.1%    0.1%        33      0    172      13
07 - (             RX)     15      35       3    0.0%    0.0%        38      0    104      26
08 - (         SERIAL)     99   12153      55  120.3%    0.5%      1153      0    170      55
09 - (       DISPATCH)    990       4       2    0.3%    0.1%        25      0   1696       2
10 - (BATTERY_VOLTAGE)     50       5       5    0.0%    0.0%         3      0     86       4
12 - ( BATTERY_ALERTS)      5       2       3    0.0%    0.0%         0      0      9       3
13 - (         BEEPER)     99       4       3    0.0%    0.0%         4      0    170       3
14 - (            GPS)    117      35      38    0.4%    0.4%        45      0    203      38
17 - (       ALTITUDE)     40       4       3    0.0%    0.0%         2      0     69       3
18 - (      TELEMETRY)    495       3       2    0.1%    0.0%        37      0    844       2
20 - (            OSD)     10      43      98    0.0%    0.0%       104      2    289      98
22 - (            CMS)     10       4       3    0.0%    0.0%         0      0     17       3
23 - (        VTXCTRL)      3       2      26    0.0%    0.0%         1      0      3      27
24 - (        CAMCTRL)      3       1       1    0.0%    0.0%         0      0      3       1
26 - (    ADCINTERNAL)      2       3       4    0.0%    0.0%         0      0      1       4
28 - (SPEED_NEGOTIATION)     33       3       2    0.0%    0.0%         1      0     57       2
RX Check Function                  11       7                        46
Total (excluding SERIAL)                                37.6%`

sjauquet, 2022-01-12:
here is without dshot_bigbang set

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10      22      20    0.0%    0.0%         1     16   1956      20
01 - (         SYSTEM)    987      32      20    3.1%    1.9%       225    776 193151      20
02 - (           GYRO)   7994      24       6   19.1%    4.7%      6397      0 1562719       0
03 - (         FILTER)   1999      45      20    8.9%    3.9%      7479      0 390679       0
04 - (            PID)   1999     184     107   36.7%   21.3%     39468      0 390680       0
05 - (            ACC)    988      31      24    3.0%    2.3%       872   1527 193114      24
06 - (       ATTITUDE)    100      38      35    0.3%    0.3%       271     56  19515      35
07 - (             RX)     15      82       8    0.1%    0.0%       296     27  11716      18
08 - (         SERIAL)    100  609035      50 6090.3%    0.5%      1184      0  19449      50
09 - (       DISPATCH)    993      26      19    2.5%    1.8%       200   1241 193191      19
10 - (BATTERY_VOLTAGE)     50      27      18    0.1%    0.0%        23     36   9762      18
12 - ( BATTERY_ALERTS)      5      16       8    0.0%    0.0%         1      3    979       8
13 - (         BEEPER)    100      25      16    0.2%    0.1%        33     91  19448      16
14 - (            GPS)    122      57      50    0.6%    0.6%       421    139  24279      50
17 - (       ALTITUDE)     40      23      20    0.0%    0.0%        14     43   7805      20
18 - (      TELEMETRY)    494      71       4    3.5%    0.1%       101    598  96417       4
20 - (            OSD)     12     104     104    0.1%    0.1%      1318    346  67895     104
22 - (            CMS)     15      24       5    0.0%    0.0%         7    104   3178       5
23 - (        VTXCTRL)      4     103      14    0.0%    0.0%         2     18    841      14
24 - (        CAMCTRL)      4       3       2    0.0%    0.0%         0      5    841       2
26 - (    ADCINTERNAL)      1      22      21    0.0%    0.0%         0      4    188      21
28 - (SPEED_NEGOTIATION)     65      24      19    0.1%    0.1%        16    170  14180      19
RX Check Function                  36       8                       687
Total (excluding SERIAL)                                37.2%

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       3      20    0.0%    0.0%         1      0     20      20
01 - (         SYSTEM)    987      22      20    2.1%    1.9%       227      6   2023      20
02 - (           GYRO)   8000       8       6    6.4%    4.8%      6464      0  16356       0
03 - (         FILTER)   1999      23      21    4.5%    4.1%      7557      0   4089       0
04 - (            PID)   1999     112     109   22.3%   21.7%     39882      0   4089       0
05 - (            ACC)    988      26      21    2.5%    2.0%       881     13   2021      21
06 - (       ATTITUDE)    100      33      32    0.3%    0.3%       274      1    205      32
07 - (             RX)     15      53       8    0.0%    0.0%       299      0    124      17
08 - (         SERIAL)    100   24616      52  246.1%    0.5%      1209      0    205      52
09 - (       DISPATCH)    986      23      21    2.2%    2.0%       203      8   2023      21
10 - (BATTERY_VOLTAGE)     50      10      17    0.0%    0.0%        23      0    103      17
12 - ( BATTERY_ALERTS)      5       4       8    0.0%    0.0%         1      0     10       8
13 - (         BEEPER)    100       9      15    0.0%    0.1%        33      3    205      15
14 - (            GPS)    134      49      50    0.6%    0.6%       425      0    264      49
17 - (       ALTITUDE)     40       4      19    0.0%    0.0%        14      0     82      19
18 - (      TELEMETRY)    497      19      16    0.9%    0.7%       101     11   1013      16
20 - (            OSD)     12      67      25    0.0%    0.0%      1331      4    725      98
22 - (            CMS)     15      21       7    0.0%    0.0%         7      2     27       7
23 - (        VTXCTRL)      4       1      14    0.0%    0.0%         2      0      6      14
24 - (        CAMCTRL)      4       1       2    0.0%    0.0%         0      0      6       2
26 - (    ADCINTERNAL)      1       3      21    0.0%    0.0%         0      0      1      21
28 - (SPEED_NEGOTIATION)     60       8      13    0.0%    0.0%        16      0    114      13
RX Check Function                  35       8                       695
Total (excluding SERIAL)                                38.2%
```

sjauquet, 2022-01-26:
Hello,
Flashed my croc baby with 4.3 RC2 and when I arm I receive the warning RPMFILTER followed by DSHOT_BBANG and the motors do not spin.
If I set dshot_bitbang = OFF, the motors are spinning but the OSD takes 20 seconds to draw the entire OSD ...
Any suggestion please ?
thanks !

sjauquet, 2022-01-27:
and here are the stats/settings:
```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       3       2    0.0%    0.0%         0      1    186       2
01 - (         SYSTEM)    911      27      16    2.4%    1.4%        25     80  17738      16
02 - (           GYRO)   8985      41      12   36.8%   10.7%       609      0 145629       0
03 - (         FILTER)   2247      51      41   11.4%    9.2%       768      0  36407       0
04 - (            PID)   2247     170     158   38.1%   35.5%      4149      0  36407       0
05 - (            ACC)    900      44      26    3.9%    2.3%        90     38  17676      26
06 - (       ATTITUDE)    456      48      41    2.1%    1.8%       109     19   7813      41
07 - (             RX)    142      85       6    1.2%    0.0%        98     25   4056      26
08 - (         SERIAL)     97  614467      52 5960.3%    0.5%      1164      0   1695      52
09 - (       DISPATCH)    911      35      18    3.1%    1.6%        22     45  17771      18
10 - (BATTERY_VOLTAGE)     48      27      20    0.1%    0.0%         2      2    923      20
12 - ( BATTERY_ALERTS)      6      27      13    0.0%    0.0%         0      2     94      13
13 - (         BEEPER)     97      25      24    0.2%    0.2%         3     14   1694      24
14 - (            GPS)    130      56      47    0.7%    0.6%        39     64   2410      47
17 - (       ALTITUDE)     37      21      20    0.0%    0.0%         1     12    719      20
18 - (      TELEMETRY)    451      51      18    2.3%    0.8%        52     73   8538      18
20 - (            OSD)     11     116     116    0.1%    0.1%       110    232   4702     116
22 - (            CMS)     10       9       6    0.0%    0.0%         0      6    224       6
23 - (        VTXCTRL)      4      96      37    0.0%    0.0%         2     10     45      37
24 - (        CAMCTRL)      4       3       2    0.0%    0.0%         0      3     45       2
26 - (    ADCINTERNAL)      3       7       7    0.0%    0.0%         0      1     10       4
28 - (SPEED_NEGOTIATION)     31      18      12    0.0%    0.0%         1     19    854      12
RX Check Function                  43       3                        56
Total (excluding SERIAL)                                64.2%

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10      10       7    0.0%    0.0%         0      5    583       7
01 - (         SYSTEM)    954      36      16    3.4%    1.5%       101    273  56292      16
02 - (           GYRO)   8000      36      25   28.8%   20.0%      2689      0 468227       0
03 - (         FILTER)   1998      74      50   14.7%    9.9%      3215      0 117057       0
04 - (            PID)   1998     199     180   39.7%   35.9%     17966      0 117057       0
05 - (            ACC)    923      44      24    4.0%    2.2%       374    125  55879      24
06 - (       ATTITUDE)    465      52      34    2.4%    1.5%       506     60  28052      34
07 - (             RX)    144      64       9    0.9%    0.1%       876     77  33760      26
08 - (         SERIAL)    100   14981      35  149.8%    0.3%      1193      0   5808      35
09 - (       DISPATCH)    957      36      22    3.4%    2.1%        85    291  56461      22
10 - (BATTERY_VOLTAGE)     50      24      19    0.1%    0.0%         9      6   2912      19
12 - ( BATTERY_ALERTS)      5      18      15    0.0%    0.0%         0      1    292      15
13 - (         BEEPER)    100      23      14    0.2%    0.1%        14     19   5808      14
14 - (            GPS)    115      57      52    0.6%    0.5%       132     44   6588      52
17 - (       ALTITUDE)     40      40      38    0.1%    0.1%         5      6   2328      38
18 - (      TELEMETRY)    463      34      29    1.5%    1.3%        77    122  27981      29
20 - (            OSD)     12     106      97    0.1%    0.1%       358    256  13241      97
22 - (            CMS)     10      24       7    0.0%    0.0%         1      3    583       7
23 - (        VTXCTRL)      2      17       7    0.0%    0.0%         2      0     98       7
24 - (        CAMCTRL)      2       3       2    0.0%    0.0%         0      1     98       2
26 - (    ADCINTERNAL)      1       8       7    0.0%    0.0%         0      0     19       7
28 - (SPEED_NEGOTIATION)     33      22       2    0.0%    0.0%         3     44   1941       2
RX Check Function                  57       6                       269
Total (excluding SERIAL)                                75.3%
```

SteveCEvans, 2022-01-27:
I note that the motor DMA streams are blocking the allocation of the SPI streams so OSD updates will be relatively slow. You may want to try `dshot_bitbang=OFF` to enable SPI DMA for the OSD. Also you might want to try the image at https://github.com/betaflight/betaflight/pull/11340#issuecomment-1022698377