betaflight/betaflight #11174: Some values not shown (show zero)  in OSD BF4.3 RC1 / CrazybeeF4SX1280

LoakAudio, 2021-12-26:
### Describe the bug

Some values a shown as zero in OSD : GPS coordinates but it may be a duplicate of " BF 4.3 RC1 GPS coord not shown #11156 ", but ALSO TX Power Link (0 MW in the picture below)

[image]

### To Reproduce

Select GPS Coordinates and TX Power link in OSD
### Expected behavior

GPS coordinates shown as in the GPS tab of Betaflight-configurator, TX Power link shown as in telemetry sensors of OpenTX
### Flight controller configuration

```
# diff

# version
# Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Dec 24 2021 / 11:15:50 (5933d96bc) MSP API: 1.44

# start the command batch
batch start

board_name CRAZYBEEF4SX1280

# name: Crux3 NLR

# feature
feature GPS

# serial
serial 0 2 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200

# beeper
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ON_USB
beeper -BLACKBOX_ERASE

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1300 2100 0 0
aux 2 46 2 1700 2100 0 0
aux 3 13 1 900 1300 0 0
aux 4 35 1 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# master
set acc_trim_roll = 5
set acc_calibration = -4,-93,-18,1
set motor_pwm_protocol = DSHOT300
set motor_output_reordering = 3,2,1,0,4,5,6,7
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 180
set bat_capacity = 3000
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 310
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_initial_alt = 25
set gps_rescue_descent_dist = 70
set gps_rescue_ground_speed = 700
set gps_rescue_throttle_max = 1900
set gps_rescue_ascend_rate = 200
set gps_rescue_descend_rate = 100
set gps_rescue_throttle_hover = 1560
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 80
set osd_link_quality_pos = 14699
set osd_link_tx_power_pos = 2379
set osd_tim_2_pos = 14710
set osd_flymode_pos = 2391
set osd_throttle_pos = 2359
set osd_crosshairs_pos = 14509
set osd_current_pos = 2336
set osd_mah_drawn_pos = 2369
set osd_gps_speed_pos = 14390
set osd_gps_lon_pos = 2113
set osd_gps_lat_pos = 2081
set osd_gps_sats_pos = 2145
set osd_home_dir_pos = 14446
set osd_home_dist_pos = 2134
set osd_flight_dist_pos = 2166
set osd_altitude_pos = 18582
set osd_warnings_pos = 14602
set osd_avg_cell_voltage_pos = 14689
set osd_disarmed_pos = 14539
set osd_flip_arrow_pos = 2286
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5769
set vcd_video_system = NTSC
set expresslrs_uid = 85,36,152,190,203,181
set expresslrs_rate_index = 3
set name = Crux3 NLR

profile 0

# profile 0
set vbat_sag_compensation = 100

rateprofile 0

# rateprofile 0
set thr_mid = 56
set thr_expo = 33
set roll_srate = 36
set pitch_srate = 36
set yaw_srate = 34

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: PREINIT 2
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Setup / Versions
 
 - Flight controller: CRAZYBEEF4SX1280
  
 - Other components: AIO ELRS+VTX, Happymodel Mini M8N GPS
 
 - How are the different components wired up: ELRS link and VTX are AIO parts, GPS is welded

AndreaCCIE, 2021-12-27:
To show the power uplink, you have to set the wide switch mode in the tx lua

phobos-, 2021-12-27:
Hello @LoakAudio ,

for TX Power you need to set your switch mode to WIDE in your transmitter module. There is no need to configure anything else in the flight controller. 

https://www.expresslrs.org/2.0/software/switch-config/
```
 WideHybrid uses the 8th slot to transmit extra data to the receiver, including the current transmitter power. This is the only switch mode which can show TPwr on the flight controller's OSD.
```

GPS coordinates bug is already being worked on.

LoakAudio, 2021-12-27:
Hoooo, yes, I missed that

Really sorry, I close this no-bug.

LoakAudio, 2021-12-27:
It was a RTFM bug

closed 2021-12-27 by LoakAudio

phobos-, 2021-12-27:
@LoakAudio no worries, please confirm TX power is working after setting switch mode to `WIDE`. I'd like to be sure it is working for you 👍

LoakAudio, 2021-12-27:
@phobos- I confirm, with WIDE switchmode, TX Power link is working in OSD : 250mW max, dynamic mode ON, 10 mW in OSD.