betaflight/betaflight #11178: No CMS / end stats on F405/F411 RC1

CapnBry, 2021-12-27:
### Describe the bug
When switching to CMS mode, the OSD freezes, showing the standard OSD screen but betaflight is definitely in CMS mode, as it can be exited by using the sticks to navigate the invisible menu. Once CMS is exited, the OSD resumes refreshing as expected. During this time, the quad can not be armed, with the Configurator status showing `Arming disable flags: CLI CMS MSP`. Oddly, the OSD will show "CMS" when arming is attempted, despite the rest of the OSD not updating.

Similarly, there's no end stats on disarm. The OSD stops updating at the last frame before the disarm. If I move the sticks enough to clear the end stats, the OSD resumes updating properly.

Betaflight 4.3.0-RC1. Happens on OMNIBUSF4SD (F405) and CRAZYBEEF4FS (F411) at any CPU usage, **but works fine if I `feature -TELEMETRY`**. I've even run 8K/4K on the F11 with no overclock and the OSD runs fine until I try to enter CMS with feature TELEMETRY enabled.

CRAZYBEEF4FS Nightly #2569 WORKS
CRAZYBEEF4FS Nightly #2570 WORKS
CRAZYBEEF4FS Nightly #2571 WORKS 36% CPU at 2K, 68% at 4K, or 52% at 4K 120MHZ overclock
CRAZYBEEF4FS Nightly #2572 Does not work

### To Reproduce
* Flash RC1 with complete flash erase, apply custom defaults.
* Set correct serialrx port, set serial provider
* Enable TELEMETRY feature
* Use stick commands to try to enter CMS, OSD does not update and status shows CMS_MENU as an arming disable flag in the configurator
[image]

### Expected behavior
OSD should render properly? Possible fix in  #11169, see below.

### Flight controller configuration
```
# version
# Betaflight / STM32F411 (S411) 4.3.0 Dec 17 2021 / 23:34:10 (8cbd562dc) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FS, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name CRAZYBEEF4FS
manufacturer_id HAMO

# feature
feature -RX_SPI
feature RX_SERIAL
feature TELEMETRY

# serial
serial 0 64 115200 57600 0 115200

# master
set acc_calibration = -72,-94,-127,1
set serialrx_provider = CRSF
set gyro_1_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: FREE
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 - Flight controller: OMNIBUSF4SD (F405) and CRAZYBEEF4FS (F411)
 - HappyModel EP2 ExpressLRS RX on UART6 / UART1 respectively
 - Other components: different VTXes / ESCs
 - How are the different components wired up: 30AWG wire?

* I don't believe this to be related to #11167 because it doesn't update slowly, it just stops updating when the specific OSD modes are entered.
* Possibly related to #11165 as it does not show post-flight statistics so maybe fixed by #11169 if hex files posted by @SteveCEvans [here](https://github.com/betaflight/betaflight/issues/11165#issuecomment-1001224889) are with that PR's fixes. This does fix the OSD for me on both FCs

Because the CMS wasn't mentioned in any other PR I made a new issue, but if this wants to be flagged as duplicate and closed then go for it.

CapnBry, 2021-12-27:
Fixed via #11169

SteveCEvans, 2021-12-27:
@CapnBry Yes, those hexes contain both fixes.

Asizon, 2021-12-27:
@CapnBry dont forget to configure cpu overclock at 120mhz

CapnBry, 2021-12-27:
> @CapnBry dont forget to configure cpu overclock at 120mhz

Thanks, Asizon. Just realized I wasn't clear in the first post. It works on 8K/4K on standard clock speed with Telemetry off, but doesn't work on 8K/2K at 120MHz with Telemetry on despite the CPU use being less than half (68% vs around 25%). That has me leaning towards it not being task starved for the updates.

The commit log around where this stops working for me seems to have a commit about MSP DisplayPort. Is there a chance that disabling Telemetry somehow disables DisplayPort and therefore CMS works (i.e. CMS only works through DisplayPort?). I guess I could check if there are any MSP packets being sent to the receiver if that would make any sense?

klutvott123, 2021-12-27:
@CapnBry Disabling telemetry will prevent the telemetry displayport initialisation, yes. If I open cms using the stick command I receive frames with this command https://github.com/betaflight/betaflight/blob/5933d96bc3e0718cc4fcd54f23436241d6e1c3c3/src/main/rx/crsf_protocol.h#L58 on my radio everytime I move the sticks. Unfortunately I don't have any fpv equipment so I can't see if it works or not.

The commit in question simply moved the initialisation to be done after cms initialisation which is required for it to work. In the builds before that it was broken and has been since april. Could you try these builds? They have crsfDisplayportRegister() commented out.
[betaflight_4.3.0_STM32F405_norevision.zip](https://github.com/betaflight/betaflight/files/7781969/betaflight_4.3.0_STM32F405_norevision.zip)
[betaflight_4.3.0_STM32F411_norevision.zip](https://github.com/betaflight/betaflight/files/7781970/betaflight_4.3.0_STM32F411_norevision.zip)

CapnBry, 2021-12-28:
> The commit in question simply moved the initialisation to be done after cms initialisation which is required for it to work. In the builds before that it was broken and has been since april. Could [...]

I can confirm both the F405 and F411 do work and allow me to get into CMS with feature TELEMETRY disabled or enabled. On the F411 I can run 8K/4K with no overclock and bidirectional DSHOT at 68% CPU and the CMS works with very minor delay, which is gone at 8K/2K or without bidir dshot. The F405 CMS works with and without feature TELEMETRY, but I can not change the PID denom so I can't test anything else (almost certainly unrelated, but thought I'd mention it).

So apparently `USE_CRSF_CMS_TELEMETRY` means CMS has to use DisplayPort? It seems you're onto the solution here :)

haslinghuis, 2021-12-28:
@CapnBry Tested RC1 with Matek F411 and Crossfire on OpenTx with telemetry enabled in configuration and able to access CMS.

klutvott123, 2021-12-28:
@CapnBry Great! Seems like the first registered cms device becomes the default. This should do the trick
[betaflight_4.3.0_STM32F411_norevision.zip](https://github.com/betaflight/betaflight/files/7782457/betaflight_4.3.0_STM32F411_norevision.zip)

CapnBry, 2021-12-28:
> @CapnBry Great! Seems like the first registered cms device becomes the default. This should do the trick [betaflight_4.3.0_STM32F411_norevision.zip](https://github.com/betaflight/betaflight/files/7782457/betaflight_4.3.0_STM32F411_norevision.zip)

Sorry about the delay, I fell asleep. This works perfectly with everything turned on, the CMS works as expected. I still don't get any end stats though, does that need the fix from the other PR too or should it be working?

@haslinghuis That's bizarre. I wonder how that works and my setup doesn't. Is there something ExpressLRS sends that makes it enable DisplayPort? There may be the possibility we're sending some MSP on connect or something. I know we do for the VTX Administrator but it might be slipping through some others.

phobos-, 2021-12-28:
@CapnBry sending VTX MSP on connect is a non-default value, so I suspect most users have it off 😅

klutvott123, 2021-12-28:
@CapnBry Thanks for testing! PR is up https://github.com/betaflight/betaflight/pull/11189

CapnBry, 2021-12-31:
> @CapnBry Thanks for testing! PR is up #11189

Amazing! With this applied to latest master, everything is working perfectly (CMS and end stats) on my F405 and F411s with ExpressLRS receivers.

JAL513, 2022-01-04:
End stats aren't working on CRAZYBEEF4FR either. Will this fix work for that target as well? Should I report this separately? Let me know how I can help.

JAL513, 2022-01-04:
> End stats aren't working on CRAZYBEEF4FR either. Will this fix work for that target as well? Should I report this separately? Let me know how I can help. 

I can confirm this is fixed in the latest nightly build as of 1/3/21

SteveCEvans, 2022-01-06:
Fixed by: #11189

closed 2022-01-06 by SteveCEvans