betaflight/betaflight #11187: Hglrc 722

ElRouge, 2021-12-28:
### Describe the bug

I updated to 4.3 and I have a consumption of the cpu at 60% while with 4.2.11 it was 4%

### Flight controller configuration
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Dec 24 2021 / 18:42:12 (norevision) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 836acaf9, date: 2021-10-31T19:48:38Z

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# name: SourceOne

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature OSD

# serial
serial 1 65536 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# beacon
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 3 900 1300 0 0
aux 3 35 1 1300 1700 0 0
aux 4 36 2 1625 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 26 29 36
vtxtable powerlabels 25 400 800 1W6

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 34,1,-132,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set dshot_bidir = ON
set yaw_motors_reversed = ON
set small_angle = 180
set osd_link_quality_pos = 2102
set osd_rssi_dbm_pos = 2134
set osd_tim_2_pos = 2519
set osd_throttle_pos = 2113
set osd_craft_name_pos = 2081
set osd_avg_cell_voltage_pos = 2497
set osd_disarmed_pos = 2314
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5740
set name = SourceOne

profile 0

rateprofile 0

# end the command batch
batch end

# 
PASTE THE OUTPUT OF 'resource show all' HERE
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Thanks for the work you do.
Merry christmas and happy new year to all

Asizon, 2021-12-28:
It is how now works, please read tuning notes for 4.3

daleckystepan, 2021-12-28:
That seems to be OK. We have changed the way how CPU usage is calculated. Check https://github.com/betaflight/betaflight/wiki/4.3-Tuning-Notes

ElRouge, 2021-12-28:
ah ok sorry. I'm going to read :-)

KarateBrot, 2021-12-28:
No worries. On my F7 I have about 60% average load as well. You're good, it's normal :)

You can go to the CLI and type "tasks" to get a more detailed list of the CPU load.

closed 2021-12-28 by KarateBrot