betaflight/betaflight #11193: Open location code showing as lat or lon

P-I-Engineer, 2021-12-29:
### Describe the bug

Open  location code not working on Crazybeef4sx1280 28-12-2021 13:08 - 4.3.0
#2598
it shows as lat or lon depending on what you choose in the osd configurator page.

### To Reproduce

set osd element GPS latitidue drop down to "using open location code"

### Expected behavior

I expect to see gps + codes

### Flight controller configuration

```
set osd_gps_lat_pos = 51215
```

```
resources NA
```

### Setup / Versions
 
 - Flight controller:  HM crux35 aio spi elrs
  not important
 - Other components: none
 
 - How are the different components wired up: as expected

this worked in previous nightlies of BF4.3

etracer65, 2021-12-29:
This is as designed. The open location code includes both the latitude and longitude so you only need to include one element.

etracer65, 2021-12-29:
Oops, sorry misread. The decision was made to disable OLC support from this target. See https://github.com/betaflight/betaflight/pull/11134

daleckystepan, 2021-12-29:
Bruce is right. Sorry for that. We needed to fit all new features in small flash space so we have decided to remove this one.

P-I-Engineer, 2021-12-29:
did a bunch of code digging reading, i found the flash size feature cut section common_pre.h

got it, i'll close. I'll do a custom build to re-enable and turn some other stuff off.

thanks for responses

closed 2021-12-29 by P-I-Engineer