betaflight/betaflight #11199: Improper allocations DMA streams for KAKUTEF7 target - BF4.3.0 - RC1

Martivip, 2021-12-30:
### Describe the bug

Default DMA allocations for the KAKUTEF7 are as shown below and it can be seen that ```SPI_MISO 4``` has not been assigned to a stream and that cause status ```gyro 1 locked``` instead of ```gyro 1 locked dma```
The initial analysis and the problem has been described by @SteveCEvans in this PR https://github.com/betaflight/betaflight/pull/11007
The question is whether this problem has impact to all F745 targets or only to KAKUTEF7

### To Reproduce

Install the BF4.3.0-RC1 firmware for KAKUTEF7 target
### Expected behavior

Gyro status should be ```gyro 1 locked dma``` and ```SPI_MISO 4``` should be allocated to the stream 

### Flight controller configuration

```
# 

# diff

# version
# Betaflight / STM32F745 (S745) 4.3.0 Dec 24 2021 / 18:39:17 (norevision) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# resources
resource MOTOR 5 NONE
resource CAMERA_CONTROL 1 B03
resource PINIO 1 C09

# timer
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)

# dma
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# serial
serial 0 2048 115200 57600 0 115200
serial 1 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# led
led 0 4,11::COW:2
led 1 5,11::COW:2
led 2 6,11::COW:10
led 3 7,11::COW:10
led 4 8,11::COW:10

# aux
aux 0 0 0 2050 2100 0 0
aux 1 1 2 1475 1525 0 0
aux 2 46 3 1200 1650 0 0
aux 3 13 3 1900 2000 0 0
aux 4 13 3 1675 1750 0 0
aux 5 13 3 1425 1525 0 0
aux 6 13 3 1200 1275 0 0
aux 7 15 3 2025 2075 0 0
aux 8 15 3 1900 1950 0 0
aux 9 15 3 1775 1825 0 0
aux 10 15 3 1650 1700 0 0
aux 11 15 3 1525 1575 0 0
aux 12 15 3 1425 1475 0 0
aux 13 15 3 1300 1350 0 0
aux 14 15 3 1175 1225 0 0
aux 15 35 2 2050 2100 0 0
aux 16 40 3 1650 1875 0 0
aux 17 40 3 1175 1400 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK I FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6C    X CUSTOM  5658 5695 5760 5800 5880 5917    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 3 h

# master
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 500
set gyro_lpf1_dyn_max_hz = 1000
set acc_calibration = -54,-4,14,1
set align_mag = CW90FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 900
set baro_hardware = AUTO
set fpv_mix_degrees = 25
set serialrx_provider = GHST
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 3
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1300
set vbat_warning_cell_voltage = 340
set ibata_scale = 148
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 35
set gps_rescue_descent_dist = 100
set gps_rescue_throttle_min = 1000
set gps_rescue_throttle_max = 1400
set gps_rescue_throttle_hover = 1255
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_dth = 80
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 200
set osd_warn_batt_not_full = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_alarm = 30
set osd_link_quality_alarm = 30
set osd_rssi_dbm_alarm = -96
set osd_cap_alarm = 1150
set osd_alt_alarm = 120
set osd_tim1 = 0
set osd_tim2 = 1
set osd_vbat_pos = 2168
set osd_link_quality_pos = 2233
set osd_rssi_dbm_pos = 2265
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2497
set osd_anti_gravity_pos = 2145
set osd_throttle_pos = 2113
set osd_vtx_channel_pos = 2101
set osd_current_pos = 2092
set osd_mah_drawn_pos = 2200
set osd_gps_sats_pos = 2088
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2129
set osd_flight_dist_pos = 2121
set osd_altitude_pos = 2241
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2136
set osd_battery_usage_pos = 2505
set osd_disarmed_pos = 2475
set osd_flip_arrow_pos = 2502
set osd_rate_profile_name_pos = 2081
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917
set vcd_video_system = PAL
set camera_control_internal_resistance = 570
set pinio_box = 40,41,42,43
set gyro_1_align_yaw = 2700

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 150
set dterm_lpf1_dyn_max_hz = 300
set dterm_lpf2_static_hz = 300
set anti_gravity_gain = 4500
set p_pitch = 70
set i_pitch = 125
set f_pitch = 163
set p_roll = 67
set i_roll = 120
set d_roll = 39
set f_roll = 156
set p_yaw = 67
set i_yaw = 120
set f_yaw = 156
set d_min_roll = 39
set d_min_pitch = 46
set simplified_d_gain = 130
set simplified_pi_gain = 150
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 130
set simplified_pitch_d_gain = 120
set simplified_dterm_filter_multiplier = 200

rateprofile 0

# rateprofile 0
set thr_mid = 30
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 70
set tpa_breakpoint = 1750

# end the command batch
batch end

# 
```

```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: CAMERA_CONTROL
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: PINIO 1
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: SERIAL_RX 7
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH2 : DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: HolyBro KAKUTE F7 v1.5

SteveCEvans, 2022-01-06:
Try

```
dma SPI_MISO 1 1
save
```

That'll move `SPI_MISO 1` to stream 2
```
# dma SPI_MISO 1 list
# 0: DMA2 Stream 0 Channel 3
# 1: DMA2 Stream 2 Channel 3
```

And that will allow `SPI_MOSI 4` to be allocated to stream 0
```
# dma SPI_MISO 4 list
# 0: DMA2 Stream 0 Channel 4
```

Which should result in
```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 4
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 2
DMA2 Stream 7: DSHOT_BITBANG 5
```

Martivip, 2022-01-06:
@SteveCEvans this CLI command works, but the end user still need the solution where CLI inputs should be not necessary.
Maybe default target config file should be updated ?

SteveCEvans, 2022-01-06:
Yes, agreed

Martivip, 2022-01-06:
> Yes, agreed

Do you agree with the final solution for the end user ? or with update default target config file ? :)
or maybe you have other solution in mind ?

SteveCEvans, 2022-01-06:
I agree with "maybe default target config file should be updated ?"

Martivip, 2022-01-06:
> I agree with "maybe default target config file should be updated ?"

as I remember from https://github.com/betaflight/betaflight/pull/11007 that was your idea and you already performed the test with "updated target" on previouse BF version and with succes and without any negative impact, right ?

SteveCEvans, 2022-01-06:
Recall that changing default behaviour was explored and discounted in https://github.com/betaflight/betaflight/pull/11012. Changing impacted configs seems the right approach here.

Martivip, 2022-01-07:
@SteveCEvans the question is how to allocate the impacted targets

Martivip, 2022-02-01:
The problem still exist in BF 4.3.0 RC3

Martivip, 2022-03-09:
Issue still present

haslinghuis, 2022-04-15:
@SteveCEvans please check my attempt to fix it in https://github.com/betaflight/betaflight/issues/11199

haslinghuis, 2022-04-26:
@Martivip please test https://github.com/betaflight/betaflight/pull/11561

Martivip, 2022-04-28:
> @Martivip please test #11561

@haslinghuis for this test purposes I should download the unified KAKUTEF7 target file , from this : https://github.com/betaflight/unified-targets/tree/master/configs/default production repository ?

haslinghuis, 2022-04-28:
@Martivip yes, but you have to apply this little patch to the config file: https://github.com/betaflight/unified-targets/pull/586

Martivip, 2022-04-28:
@haslinghuis sorry my mistake I mean default config file, not unified Target 
Im asking because @SteveCEvans mention here: https://github.com/betaflight/betaflight/pull/11012 that F745 FC configs thus need to be updated to use :)

haslinghuis, 2022-04-28:
@Martivip KakuteF7 target config is not affected:
https://github.com/betaflight/betaflight/pull/11012#issuecomment-1110179104

Martivip, 2022-04-28:
@haslinghuis ok after upload artifact from https://github.com/betaflight/betaflight/pull/11561 unfortunately still ```SPI_MISO``` 4 has not been assigned to a stream and that cause status ```gyro 1 locked``` instead of ```gyro 1 locked dma```
```
# status
MCU F745 Clock=216MHz, Vref=3.29V, Core temp=37degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4128, max available: 32768
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 14 seconds, Current Time: 2022-04-28T17:58:06.570+00:00
CPU:52%, cycle time: 127, GYRO rate: 7874, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 2
SD card: None inserted
Arming disable flags: RXLOSS CLI MSP RESCUE_SW RPMFILTER

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 5

# timer show

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

Martivip, 2022-04-28:
@haslinghuis ok today is not my day OK after upload artifact from https://github.com/betaflight/betaflight/pull/11561 everything is OK. ```SPI_MISO 4``` has been assigned and gyro status is ```gyro 1 locked dma```
so the issue has been resolved :)

```
# status
MCU F745 Clock=216MHz, Vref=3.29V, Core temp=49degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4128, max available: 32768
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 20 seconds, Current Time: 2022-04-28T18:10:04.932+00:00
CPU:47%, cycle time: 126, GYRO rate: 7936, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: None inserted
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 4
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 5
DMA2 Stream 7: DSHOT_BITBANG 2

# timer show

Currently active Timers:
-----------------------
TIM1:
    CH3 : DSHOT_BITBANG 5
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH4 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
```

SteveCEvans, 2022-04-28:
@Martivip That had me sat here with a FOXEERF745AIO scratching my head wondering why you still had `DMA2 Stream 0: SPI_MISO 1` rather than `DMA2 Stream 0: SPI_MISO 4`. Glad that's now sorted!

closed 2022-04-29 by blckmn