betaflight/betaflight #11201: Holibro Kakute F4v2 Blackbox Flash Write Issues

c3r63ru5, 2021-12-30:
### Describe the bug

When logging blackbox data to onboard flash, logging is aborted randomly.

### To Reproduce

On a Kakute F4v2 FC, enable logging to internal flash (e.g. 2k and GYRO SCALED) and start logging, the resulting logs in flash are incomplete logs, aborted after a few seconds. In the CLI, the command flash_scan results in an error. When comparing to 4.2.11 (and previous), logging and flash_scan are working as expected.

### Expected behavior

flash_scan executed via CLI should result in "successful" and logging should not be interrupted if flash is not defective.

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F405 (S405) 4.3.0 Dec 24 2021 / 18:29:45 (norevision) MSP API: 1.44
# config: manufacturer_id: HBRO, board_name: KAKUTEF4V2, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name KAKUTEF4V2
manufacturer_id HBRO

# master
set blackbox_device = SPIFLASH
set pid_process_denom = 2
set debug_mode = GYRO_SCALED
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: LED 3
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: OSD_CS
B15: INVERTER 3
C00: FREE
C01: FREE
C02: FREE
C03: ADC_BATT
C04: GYRO_CS 1
C05: GYRO_EXTI
C06: FREE
C07: FREE
C08: FREE
C09: BEEPER
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: Holibro Kakute F4 V2
 
 - Other components: R9 FPort on UART1; Vista MSP on UART4
 
 - How are the different components wired up: Nothing special, 4 ESCs connected to Motor 1-4, two UARTs used by RX and Vista MSP, 1000uF ESC on Power lead, powered by 6S

SteveCEvans, 2021-12-30:
Can you please confirm if this FC has a Winbond 25Q128JV flash device fitted.

I've just tested a MATEKF405TE which has the same device albeit in a different package, and that works fine.

I see from the `HBRO-KAKUTEF4V2.config` file that the FLASH and OSD share the same bus. Could I please therefore ask you to disable the OSD and try logging to the FLASH again to see if that improves things. If so this could point to an issue with sharing the bus between devices.

```
set max7456_spi_bus = 3
set flash_spi_bus = 3
```

c3r63ru5, 2021-12-30:
Flash is a Winbond 25Q128JV.

OK, will disable OSD and check again.

c3r63ru5, 2021-12-30:
Tried again, first try with OSD Feature disabled, second one after setting OSD SPI to 0 (set max7456_spi_bus = 0). Same behaviour unfortunately, BB stops logging, also flash_scan fails to verify flash memory.

I also checked the config, and OSD and FLASH were using the same bus before (initial commit).

closed 2022-02-07 by github-actions[bot]