betaflight/betaflight #11203: No more Motors if LED_STRIP option is activated

HiTilt-FPV, 2021-12-30:
### Describe the bug
When turning on the LED_STRIP option in the config tab, in the motors tab, i get a message saying:
"mixer mode problem detected
the quad x model needs 4 motor resources and the current configuration of the firmware provides 0 usable outputs for the selected mode...."

When this happens, BF does not notice any motors, but the motor resources are still set like before.
Turning off the LED_STRIP option again restores the function of the motors.

### To Reproduce
turn on and off the LED_STRIP option in the config tab. 

### Expected behavior
i expect the motors to work, no matter how the LED_STRIP option is set

### Flight controller configuration
diff with LED_STRIP option activated (motors not working):
```
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Dec 24 2021 / 18:42:12 (norevision) MSP API: 1.44
# config: manufacturer_id: BEFH, board_name: BETAFPVF722, version: 150f0b6f, date: 2020-08-16T02:29:24Z

# start the command batch
batch start

board_name BETAFPVF722
manufacturer_id BEFH

# name: TinyTrainer

# feature
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 2 900 1300 0 0
aux 2 35 2 1300 1700 0 0
aux 3 36 1 1700 2100 0 0
aux 4 39 6 900 1700 0 0
aux 5 49 4 1700 2100 0 0

# master
[... 509 lines cut ...]
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 1
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 - Flight controller:  [Betafpv Toothpick F722 2-6S AIO Brushless Flight Controller 35A(BLHeli_S) ](https://betafpv.com/products/toothpick-f722-2-6s-aio-brushless-flight-controller-35ablheli_s)

 - Other components:
tiny trainer build with:
irc ghost zepto receiver,
irc tramp nano vtx,
runcam racer 3 nano mck edition,
five33 tinyturner motors,
[The ManazGCS Tiny Trainer LED Kit](https://flyfive33.eu/product/the-manazgcs-tiny-trainer-led-kit/)

 
uart 3 Tx is ghost receiver, vtx is controlled by ghost receiver (only vid and gnd,vcc used from AIO), LED-kit is wired to dedicated LED pin, rest is wired as labeled/ described in the AIO's pdf wireschematics. AIO has aditional 600uF of 50V SMD C's (each 100uF) wired closely to battery input pads.

I had no problems with BF 4.2.x with this build. If the Bug occurs, i can also not read any of the ESC's with the [ESC Configurator](https://esc-configurator.com/) tool. The Log of the Tool says it downloads data from 0/0 ESC's, so it really is like the AIO has no ESC's at all.

SteveCEvans, 2021-12-30:
@HiTilt-FPV LED_STRIP is using TIM1 and thus blocking DSHOT from also using it.

```
TIM1:
    CH1 : LED_STRIP
```

```
TIM1:
    CH1 : DSHOT_BITBANG 2
    CH3 : DSHOT_BITBANG 1
```

Try setting `dshot_bitbang_timer = TIM8`.

HiTilt-FPV, 2022-01-02:
@SteveCEvans thats a good hint. Problem is that:
```
TIM8:
    CH3 : CAMERA_CONTROL 
```

But since i do not use any camera control features, i freed ```C08: CAMERA_CONTROL``` and now it works finde.

closed 2022-01-02 by HiTilt-FPV