betaflight/betaflight #11204: SA "commands" not working as expected

wildjokerss, 2021-12-30:
### Describe the bug
Since a few versions ago (don't know which one as I just confirmed my suspicions today) VTX / SA stopped working as expected.
I do have VTX power on a switch, problem is, it is no longer updating the VTX
I have 2 quads with this issue.  A 3" with a Rush tiny Tank and a 5" with TBS Unify (5v 800mw, don't now the very exact version)
The problem is the following. I turn on the radio, plug the quad, start flying. Then mid flight I feel like flipping the switch to "kick in full power" on the VTX. I flip the switch, the OSD says it changed (e.g from 25mw to 350mw), but in reality, it didn't change, it stayed on 25mw. As you might guess it took me a lil bit to realize, on the 5" I even thought I had a issue with my pigtail.
Then I investigated further, lipo connected, connected to PC, everything seems fine. If i try to change the power via VTX tab, it doesn't work either.
Then I realized something. I can change the VTX power with the switch, but not after betaflight "initializes".
radio on, switch set to 25mw
plug quad, VTX is on 25mw
change to 350mw 
stays on 25mw

Radio on, switch on 350mw
plug quad, VTX is on 350mw
change to 25mw
stays on 350mw

### To Reproduce
- Using a Betaflight 4.3 version "around" the RC1, I believe RC 1 is affected and the 3rd build from 28th (#2589) is the one I'm currently using and can confirm it has this issue.
- Having a VTX using SA setup and working
- plugging in a quad and try to change the VTX power with a switch or through OSD (this isn't working)

### Expected behavior
Example 1
radio on, switch set to 25mw
plug quad, VTX is on 25mw
change to 350mw 
VTX changing to 350mw

Example 2
Radio on, switch on 350mw
plug quad, VTX is on 350mw
change to 25mw
VTX changing to 25mw

### Flight controller configuration
```
diff

# version
# Betaflight / STM32F405 (S405) 4.3.0 Dec 28 2021 / 21:17:54 (3d9b7cf49) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: acd41165, date: 2021-04-27T20:17:19Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# name: Azulis

# feature
feature SOFTSERIAL
feature TELEMETRY

# serial
serial 2 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# beeper
beeper -BAT_LOW
beeper -ON_USB

# aux
aux 0 0 0 900 1125 0 0
aux 1 1 5 1300 1700 0 0
aux 2 2 5 1700 2100 0 0
aux 3 13 3 1300 1700 0 0
aux 4 31 6 1700 2100 0 0
aux 5 35 3 1850 2100 0 0
[... 284 lines cut ...]
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 2

# 
```

### Setup / Versions
 - Flight controller: Both quads both have the same target "FuryF4OSD". One is a Mamba F405 MK2 and the other is the Mamba F405 MK2 Mini, both bought on banggood 
 - Other components:
As mentioned earlier, The quad where I took the diff from has a Rush tiny tank, the 5" facing the same issue has a TBS unify pro (5v 800mw).
 - How are the different components wired up:
A bit unsure what to add here but SA is connected to TX3

I noticed something odd. I plug a lipo, radio on, connected to configurator, i flip the switch and it doesn't change anything, if I change through the VTX tab it also does nothing, BUT if I go to the CLI, and change "set vtx_power = 1" and press save, it works, and all of a sudden the SA starts working, either using the switch on the radio, or via the VTX tab on the configurator. Sadly after unplugging the lipo and USB cable, if I plug in the lipo again, using the switch no longer works. This does prove that my SA is working and has no issues, but somehow looks like commands aren't being sent to the VTX a few seconds after booting up.

I'd like to say in advance that I've never raised an issue here so please let me know if I messed up anything. If you need further info from me feel free to reach out, would be glad to help.

abwyatt, 2021-12-31:
Same experience here with 3 seperate quads going to RC1. 2 X Speedybee F7 V2 stack, and an iFlight F7 STD AIO. VTX control from switch on all 3 quads works before arming. After arming, VTX power can't be changed from switch.

wildjokerss, 2021-12-31:
think I have a ever so slight different problem, I cannot change before arming. the VTX power will be set when I plug the quad to wherever the switch is at the time I plug it in. plugging in and flipping the switch does not change the vtx power, even if I haven't armed. @abwyatt on your OSD, does it change during flight to the power it should be on? Mine changes on OSD but not on the actual VTX 🙃

abwyatt, 2021-12-31:
> think I have a ever so slight different problem, I cannot change before arming. the VTX power will be set when I plug the quad to wherever the switch is at the time I plug it in. plugging in and [...]

Switch changes VTX power in OSD and actual VTX power output before arming. But after arming, switch will not change power in OSD or actual VTX power.

wildjokerss, 2021-12-31:
Okay, this is odd then, it's not exactly the same as me, but I guess it might be related

abwyatt, 2022-01-01:
Happy to report that 29 Dec nightly seems to have fixed my issue. VTX control on switch working after arming.

wildjokerss, 2022-01-01:
@abwyatt where did you got that? The latest showing up here is from 28th 🙃 
As per https://ci.betaflight.tech/job/Betaflight/, there was no build on the 29th.

abwyatt, 2022-01-01:
> @abwyatt where did you got that? The latest showing up here is from 28th 🙃 As per https://ci.betaflight.tech/job/Betaflight/, there was no build on the 29th.

It's in the RC1 Configurator. 29 Dec #2598
[image]

wildjokerss, 2022-01-01:
[image]
well, now... that's odd :D on mine, build 2598 is from the 28th, not 29th.
Non the less, have you looked at the actual VTX while using the switch to change power modes and see if it actually changes (check the leds or something). I'm on the same build, the OSD shows the power level changing, but it does not reflect on the VTX @abwyatt

kbenne, 2022-01-01:
> Same experience here with 3 seperate quads going to RC1. 2 X Speedybee F7 V2 stack, and an iFlight F7 STD AIO. VTX control from switch on all 3 quads works before arming. After arming, VTX power can't be changed from switch.

Yes. I also see exactly this. Matek f7 mini and Rush Tank.

kbenne, 2022-01-01:
I'm not familiar with the Betaflight code, but this commit might be relevant https://github.com/betaflight/betaflight/commit/a588d23a293906f3facaab0d0e1ad6c14f52129f.

wildjokerss, 2022-01-01:
> I'm not familiar with the Betaflight code, but this commit might be relevant [a588d23](https://github.com/betaflight/betaflight/commit/a588d23a293906f3facaab0d0e1ad6c14f52129f).

I barely understand code, but looking at that, I doubt that that is the commit that caused the issue. Either I'm misunderstanding or that's from October 2020. This stopped working like... idk... since a month ago tops. I do know it was before the "new configurator" cuz my working quad is on the "older" ones and this vtx power is working well

I also noticed that the switch works before the FC beeps, but not after

m4rtin74, 2022-01-01:
> > @abwyatt ezt honnan vetted? A legutóbbi megjelenés itt 28-án történt🙃A https://ci.betaflight.tech/job/Betaflight/ szerint 29-én nem volt építkezés.
> 
> Az RC1 Configuratorban van. december 29. #2598 [image]

I think there is a difference in dates due to time zones.
The vtx power switch doesn't work for me either. It works on one of the drones that has Rush Tank Ultimate Plus on it. It doesn't work on a Rush Tank Solo. I even have a problem with the Real Pit mode not working on the FC. When I plug in the connector, the mode in which the switch on the remote control is set is activated, but I cannot switch afterwards.

> [image] well, now... that's odd :D on mine, build 2598 is from the 28th, not 29th. Non the less, have you looked at the actual VTX while using the switch to change power modes and see if it actually [...]

I think there is a difference in dates due to time zones.
The vtx power switch doesn't work for me either. It works on one of the drones that has Rush Tank Ultimate Plus on it. It doesn't work on a Rush Tank Solo. I even have a problem with the Real Pit mode not working on the FC. When I plug in the connector, the mode in which the switch on the remote control is set is activated, but I cannot switch afterwards.

m4rtin74, 2022-01-01:
> [image] well, now... that's odd :D on mine, build 2598 is from the 28th, not 29th. Non the less, have you looked at the actual VTX while using the switch to change power modes and see if it actually [...]

I think there is a difference in dates due to time zones.
The vtx power switch doesn't work for me either. It works on one of the drones that has Rush Tank Ultimate Plus on it. It doesn't work on a Rush Tank Solo. I even have a problem with the Real Pit mode not working on the FC. When I plug in the connector, the mode in which the switch on the remote control is set is activated, but I cannot switch afterwards.

Naomarik, 2022-01-01:
I've noticed that it doesn't work after first arm for me. I can switch VTX power no problem after plugging in battery, but as soon as I arm it's locked to whatever it was. If I reboot the quad via OSD my switches work again. In my case I went from a late October version of BF to RC1 and something in between introduced this bug.

wildjokerss, 2022-01-07:
This is now fixed in the latest nightly 👍

closed 2022-01-07 by wildjokerss