betaflight/betaflight #11205: Mixer issue on Hex X configuration

snow520lee, 2021-12-31:
### Describe the bug
can't use the Hex X configuration on 4.3.0 RC1, while is possible on 4.2.11.

### To Reproduce
tab MOTOR, choose HEX X, save and reboot. an error message pops out saying not enough resources are allocated:

[image]

### Expected behavior
it should accept the hex configuration as the 4.2.11 does on the same FC.

this is the screenshot of **4.2.11**.. all good here:
[image]
this is the screenshot of 4.3.0.. error after selecting Hex X and back to Motor tab.
[image]

### Flight controller configuration

```
diff

# version
# Betaflight / STM32F745 (S745) **4.3.0** Dec 24 2021 / 18:39:17 (norevision) MSP API: 1.44
# config: manufacturer_id: FLWO, board_name: FLYWOOF745NANO, version: 786f2ef4, date: 2021-06-15T03:31:48Z

# start the command batch
batch start

board_name FLYWOOF745NANO
manufacturer_id FLWO

# feature
feature -RX_PARALLEL_PWM

# mixer
mixer HEX6X

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -16,44,-22,1
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set yaw_motors_reversed = ON
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end

# 
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: MOTOR 6
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 1
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: MOTOR 5
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH2 : DSHOT_BITBANG 3
TIM2: FREE
TIM3: FREE
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 5
```

### Setup / Versions
 - Flight controller: FLYWOO GOKU HEX F745

 - Other components: Happy Model ELRS EP2, TBS Unify Pro32 nano
 - How are the different components wired up: all wiring is normal. no issue on 4.2.11

I tried to make a sense of it... the only thing I can find different between 4.2.11 and 4.3.0 is in the resource show all:
on 4.2.11 is:
Currently active Timers:
-----------------------
TIM1:
    CH1: DSHOT_BITBANG 1
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 5
    CH4: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3

while in 4.3.0 is:
Currently active Timers:
-----------------------
TIM1:
    CH1: DSHOT_BITBANG 1
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 5
    CH4: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3

several DMA and TIM are different.... No idea if it would help.
THANKS GUYS!!!!

haslinghuis, 2022-01-03:
Please try #11198

SteveCEvans, 2022-01-03:
Try:

```
dma SPI_MOSI 1 none
dma SPI_MOSI 4 none
save
```

Crude, but try that and report back.

snow520lee, 2022-01-04:
Hi Steve,
I pasted the two lines and save it. it seems nothing changed in terms of the issue reported... only 4 motors are visible in the motor tab and same request of addressing resources.
[image]

SteveCEvans, 2022-01-04:
`dma show`?

snow520lee, 2022-01-04:
Hi Steve, here it is:

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 5

SteveCEvans, 2022-01-04:
So the issue here is that you need four DSHOT_BITBANG DMA steams. This is because the motor pins are spread across four ports. Quite frankly that's poor FC design. You can make things better though by wiring up to motors 1, 2, 3, 4, 7, and 8. Don't use 5 and 6 because they each would require another stream. You can then use motor remapping to get motors 1 to 6 as you require.

```
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 C09
resource MOTOR 6 A03
resource MOTOR 7 B04
resource MOTOR 8 B05
```

Bit-banged DSHOT uses TIM1 and/or TIM8 which are only available on DMA2.

[image]

Port B (number 2) is handled by DSHOT_BITBANG 2, and Port E (number 5) is handled by DSHOT_BITBANG 5. These are both handled by TIM8 thus. You can see from the above table that these will be on streams 2 and 7 respectively as displayed by `dma show`.

```
TIM8:
CH1 : DSHOT_BITBANG 2
CH4 : DSHOT_BITBANG 5
```

Using the motors as you are now you also need port C (number 3) hence the following which can't be allocated on stream 2 because that's already being used above.
```
TIM1:
CH2 : DSHOT_BITBANG 3
```
By avoiding the use of MOTOR 5 and 6 you'll now be able to get both streams allocated.

snow520lee, 2022-01-04:
Hi Steve,

Thanks for your explanation, I start to understand a little more.

I will try that remapping as suggested and report back. 
What is strange to me is that if I flash the 4.2.11 everything is fine, no need any remapping and I use the pins from 1 to 6.  The hardware is the same, so the difference is in the firmware allocating those DMA in a different way (see the original post the difference between 4.2.11 and 4.3).

SteveCEvans, 2022-01-04:
4.3 uses DMA for the SPI transfers which 4.2 did not, hence the MOSI/MISO assigned streams.

snow520lee, 2022-01-05:
I have set the resources you sent... nothing changes, on the motor tab I still see only 4 motors connected.
same pop window for the Mixer Mode Problem.

# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: MOTOR 6
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 1
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: MOTOR 5
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH2 : DSHOT_BITBANG 3
TIM2: FREE
TIM3: FREE
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 5

#

SteveCEvans, 2022-01-05:
You still have the motor resources defines resulting in the attempted stream allocation.

Use the following to achieve what I suggested.

```
resource MOTOR 5 B04
resource MOTOR 6 B05
resource MOTOR 7 none
resource MOTOR 8 none
```

snow520lee, 2022-01-05:
all right, ok I've done it. 
the pop up message is gone and 6 motors are listed with the 0-100% bidirectional error.....
As the flight controller is connected to the 6in1 ESCs through a pin header, I think I will have to run two wires from S7 and S8 on the FC to the ESC board, right? 

[image]
[image]

SteveCEvans, 2022-01-05:
As the motor 5 & 6 pins are now disabled you should be able to wire the motor 7 output to motor 5 output and motor 8 to motor 6. The motor 7/8 signals will then pass through the motor 5/6 lines to the ESC. That way you’ll not need to disconnect outputs 5 & 6 from the header. Not ideal, but it should work.

snow520lee, 2022-01-05:
yes, I got all the pads breakout.. I was wondering in fact if I should remove the pin from the header.... I will try.
Thanks a lot for your help! do you think there will be a fix to use the original motors in a future release? without need to remapping?

SteveCEvans, 2022-01-05:
No need to touch the header.

snow520lee, 2022-01-05:
fantastic! thanks again Steve

closed 2022-01-05 by snow520lee

ase-dcexpert, 2022-01-09:
I am having the same problem with my Flywoo Hexplorer. Honestly, I am not in the mood resolder the motor PINs. Is there no way to fix that in betaflight? I narrowed it down, that this problem occurs with all 4.3.0 builds after #2584. Meaning that with #2594 the problem started. I will just stay with #2584.

SteveCEvans, 2022-01-09:
@ase-dcexpert The issue is caused by a poor choice of motor pins by the FC designer. The solution is to avoid the use of the badly assigned pins. The solution to this is detailed above requiring the soldering of two short link wires.

snow520lee, 2022-01-09:
> I am having the same problem with my Flywoo Hexplorer. I narrowed it down, that this problem occurs with all 4.3.0 builds after 2584. Meaning that with 2594 the problem started. I will just stay with 2584.

As Steve said... at the end it's just two wires you need to solder and 4 lines of code in the CLI.

ase-dcexpert, 2022-01-10:
@SteveCEvans Thanks, so this is the final fix? I doubt that Flywoo will change anything. I think at least this fix should be documented somewhere in the official BF documentation for other users having this problem.

ase-dcexpert, 2022-01-10:
> As Steve said... at the end it's just two wires you need to solder and 4 lines of code in the CLI.

So to put it in a nuthsell:

For the Flywoo Hexplorer with new BF 4.3.+ version to work you need to:

- solder a bridge betweem motor outputs:  
     7 and 5
     6 and 8

- apply the following CLI commands
`resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 B04
resource MOTOR 6 B05
resource MOTOR 7 none
resource MOTOR 8 none`

Or do I need more CLI commands?

ase-dcexpert, 2022-01-10:
Ok, did it. Works! 😬

SteveCEvans, 2022-01-10:
@ase-dcexpert Thanks for the confirmation.

@flywoo Please note the above. As this is a BNF you should be adding the wire links shown and updating https://github.com/betaflight/unified-targets/blob/master/configs/default/FLWO-FLYWOOF745NANO.config for 4.3 support.

grzbiet, 2025-06-20:
looks like the issue is still there but now it bricks the controller, 

the 4.5.0 still gives me the "6 resources / 4 motors" error as reported by OP

but 4.5.2 and up, if I flash the firmware (don't see difference if I choose less options) and set the motors / mixer / HEX X 
and save and reboot I see quick message "flight controller is ready" but the blue LED does not blink anymore - the usb device is not recognizable until the firmware is re-flashed.

if I do the four console lines first, then set the HEX X all is working good, motors turn no errors from bidirectional DSHOT. 

I'm writing this just as a note to myself, please let me know if anybody is interested in more details.