betaflight/betaflight #11206: Wrong home direction in OSD and during GPS rescue BF 4.3

Xedos9er, 2021-12-31:
### Describe the bug
GPS Home arrow shows wrong direction while distance to home is shown correctly. When activiation GPS Rescue quad tries to fly into the wrong direction and starts to do weird movements.
GPS has solid 20+ sats

### To Reproduce
Fly away from home spot and check home arrow, try GPS rescue

### Expected behavior
right direction is shown, GPS rescue working

### Flight controller configuration
I tried it on 2 quads with different hardware:
-Speedybee F722 V2 and HAKRC F7230D (Target Matek F722SE) 
both show same behaviour

PASTE THE OUTPUT OF 'diff' HERE
```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Dec 24 2021 / 18:42:12 (norevision) MSP API: 1.44
# config: manufacturer_id: SPBE, board_name: SPEEDYBEEF7V2, version: 55694498, date: 2021-05-15T13:56:51Z

# start the command batch
batch start

board_name SPEEDYBEEF7V2
manufacturer_id SPBE

# feature
feature -ESC_SENSOR
feature GPS
feature TELEMETRY

# serial
serial 1 1 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -SYSTEM_INIT
beeper -ON_USB

# led
led 0 4,6::AW:8
led 1 5,6::A:8
led 2 6,6::A:8
led 3 7,6::AW:8

# mode_color
mode_color 6 0 3
mode_color 6 1 6

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1300 2100 0 0
aux 2 46 2 1700 2100 0 0
aux 3 13 3 1700 2100 0 0
aux 4 35 3 1300 1700 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# master
set dyn_notch_count = 2
set dyn_notch_q = 500
set dyn_notch_min_hz = 110
set acc_lpf_hz = 3
set acc_calibration = 11,20,53,1
set min_check = 1030
set max_check = 1990
set rssi_channel = 15
set rc_smoothing_auto_factor = 60
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set ibat_lpf_period = 15
set small_angle = 180
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 45
set gps_rescue_initial_alt = 70
set gps_rescue_ground_speed = 1500
set gps_rescue_throttle_max = 1650
set gps_rescue_ascend_rate = 350
set gps_rescue_throttle_hover = 1300
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2
set osd_warn_visual_beeper = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 15
set osd_rssi_dbm_alarm = -98
set osd_vbat_pos = 194
set osd_rssi_pos = 2049
set osd_link_quality_pos = 65
set osd_rssi_dbm_pos = 97
set osd_tim_2_pos = 88
set osd_flymode_pos = 2542
set osd_crosshairs_pos = 238
set osd_current_pos = 2337
set osd_mah_drawn_pos = 2305
set osd_craft_name_pos = 367
set osd_gps_speed_pos = 2505
set osd_gps_lon_pos = 2065
set osd_gps_lat_pos = 2097
set osd_gps_sats_pos = 2081
set osd_home_dir_pos = 2510
set osd_home_dist_pos = 2511
set osd_altitude_pos = 2514
set osd_warnings_pos = 14760
set osd_avg_cell_voltage_pos = 2273
set osd_disarmed_pos = 267
set debug_mode = GYRO_SCALED
set rpm_filter_lpf_hz = 100

profile 0

# profile 0
set vbat_sag_compensation = 100
set anti_gravity_gain = 4000
set iterm_relax_cutoff = 7
set p_pitch = 65
set i_pitch = 117
set d_pitch = 69
set p_roll = 63
set i_roll = 112
set d_roll = 60
set p_yaw = 63
set i_yaw = 112
set d_min_roll = 45
set d_min_pitch = 51
set dyn_idle_min_rpm = 22
set simplified_d_gain = 150
set simplified_pi_gain = 140

rateprofile 0

# rateprofile 0
set roll_rc_rate = 5
set pitch_rc_rate = 5
set yaw_rc_rate = 5

# end the command batch
batch end

# 
```
PASTE THE OUTPUT OF 'resource show all' HERE
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: MOTOR 2
A10: MOTOR 3
A11: USB
A12: USB
A13: SWD
A14: FLASH_CS
A15: PINIO 1
B00: LED_STRIP
B01: FREE
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 1
    CH3 : DSHOT_BITBANG 3
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH2N: LED_STRIP
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: DSHOT_BITBANG 1
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 3
DMA2 Stream 7: FREE

# 
### Setup / Versions
 - Flight controller:  Speedybee F722 v2
 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) -->
-Caddx Vista
-BN-180 GPS
 - Other components:
 <!-- Details about how all is wired -->
 - How are the different components wired up:
 - connectors

closed 2021-12-31 by github-actions[bot]

haslinghuis, 2022-01-03:
Sober in #11211

hickscorp, 2022-04-21:
Isn't it because `SET_HOMEPOINT_ONCE` is set?