betaflight/betaflight #11209: SPI D8 si receiver lose bind on FW 4.3.0 RC1

Rammy1987, 2021-12-31:
### Describe the bug
updating CRAZYBEEF4FR to 4.3 RC1 I lost bind (SPI RX_support -> FRSKY_D) and I couldn't re-bind it, I sort the issue downgrading to version 4.2.11

### To Reproduce
update FC to FW 4.3.0 RC1

### Expected behavior
the quad shouldn't lose bind

### Setup
 CRAZYBEEF4FR 4.3.0-RC1 - 24-12-2021 09:50
- Configurator version:

```
PASTE THE OUTPUT OF 'diff' HERE
```

haslinghuis, 2021-12-31:
@Rammy1987 please update diff and resource show all using backticks.

DaVinniCode, 2022-01-01:
Same issue here with same FC.  I also flashed the nightly and got the same result.  It is really difficult to get it to bind to begin with and when it finally does it can lose bind mid flight.  Radio is TX16S using latest multi-module firmware.

MJ666, 2022-01-01:
Tested today BetaFPV F4 with MatecF411RX target and RC1. Binding and control in FrSky D8 and FrSkyX (V2LBT, V1FCC) is working here including telemetry. Transmitter for the test is an Jumper T18 OpenTX 2.3.14 with MPM V1.3.3.0. Flight tested all three protocols mentioned above for one battery and found no issues.

Please be aware coming from BF 4.2 the FrSky bind information is not compatible and you need to rebind with BF 4.3. Anyhow this looks not to be your problem. One other issue could be distance between TX and RX. Please have this larger than 1m or more otherwise you may be affected by swamping.

DaVinniCode, 2022-01-01:
Maybe it's something about the CRAZYBEEF4FR .  I was flying over 1m away and then the drone drops out of the sky and the transmitter vibrates due to loss of signal. Sometimes signal would be restored after a few seconds, but then it would disconnect again.
i went back to 4.2.11 to make sure there's no hardware problem, and binding became easy at any distance, and flying became reliable again.

MJ666, 2022-01-02:
Mhhh, looks to be the CRAZYBEEF4FR and the MATEKF411RX are nearly identical (motor pins, sensor ports, OSD) from hardware configuration. The MATEKF411RX has LNA/PA configured. This could be a difference in RX performance but not leading to problems in short range. There must be something else?

Zuldan, 2022-01-08:
I had a lot of issues with my Frsky D until I increased the clock speed. I’m wonder if binding isn’t actually being lost, the RX is just struggling to respond. Run the CPU at 120Mhz and see if that makes a difference.

set cpu_overclock = 120MHZ

DaVinniCode, 2022-01-08:
Ah K, I will try that thanks.  That 'struggling to respond' definitely sounds plausible.  I remember seeing a recommendation to overclock, but I didn't know the command.  I'm also on DSHOT600 because that's what mine shipped with, but I'll go down to DSHOT300.

MJ666, 2022-01-08:
My BetaFPV board is running 108 MHz, 4K loop, DSHOT 600 and no bidiretional DSHOT.

DaVinniCode, 2022-01-08:
I have bidirectional DSHOT enabled, 48kHz PWM, 4K loop.  Default clock speed.  I think overclocking will do the trick.  I'll go DSHOT300 and overclock it..  if it's stable maybe I'll try DSHOT600 again.

MJ666, 2022-01-08:
With bidirectional DSHOT enabled you need to go to DSHOT300 to make it work (4K loop). Overclock to 120MHz is likely be not required and can cause instability on some boards.

DaVinniCode, 2022-01-08:
Thank you sir, I will go DSHOT300 with a conservative overclock.  The way the RX connectivity was erratic definitely points to something like a resource problem.  Makes sense!  I will give it a go.

MLutzAT, 2022-01-11:
Underlying problem should be solved by now!
See #11246