betaflight/betaflight #11213: RSSI and link quality always shows the same values If RSSI Channel is set to Disabled when using S.Bus then generate RSSI signal using frame drop flags from the rx

kvark85, 2022-01-01:
### Describe the bug

RSSI and link quality always show the same values.

### To Reproduce

Case 1 (Betaflight Configuration):
- on the OSD add to screen Link quality;
- on the OSD add to screen RSSI value;
- on the OSD add to screen RSSI dBm value;
- on the Configuration turn off  RSSI ADC;
- on the Receiver disable RSSI Channel;
- in CLI: set rssi_src_frame_errors = **ON**;
- in CLI: save;
- put the transceiver in the microwave;
- turn off the transceiver.

Case 2 (Betaflight Configuration):
- on the OSD add to screen Link quality;
- on the OSD add to screen RSSI value;
- on the OSD add to screen RSSI dBm value;
- on the Configuration turn off  RSSI ADC;
- on the Receiver disable RSSI Channel;
- in CLI: set rssi_src_frame_errors = **OFF**;
- in CLI: save;
- put the transceiver in the microwave;
- turn off the transceiver.

### Expected behavior

Case 1 (rssi_src_frame_errors = ON):
- RSSI value is always 99;
- RSSI dBm value  is always -130;
- link quality is always 9.

Case 1 (rssi_src_frame_errors = OFF):
rssi_src_frame_errors = OFF
- RSSI value  is always 0;
- RSSI dBm value  is always -130;
- Link quality  is always 9.

### Flight controller configuration

```
diff
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F411 (S411) 4.2.5 Nov 22 2020 / 18:38:45 (afdac08b3) MSP API: 1.43
###ERROR: diff: NO CONFIG FOUND###
# start the command batch
batch start

board_name GEPRCF411
manufacturer_id GEPR

# name: TinyGO

# resources
resource BEEPER 1 B02
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 B03
resource MOTOR 6 B10
resource PPM 1 A03
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource LED 1 C13
resource LED 2 C14
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A01
resource GYRO_CS 1 A04
resource USB_DETECT 1 C15

# timer
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dma pin A02 0
# pin A02: DMA1 Stream 0 Channel 6
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature OSD

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 2,4::CTO:6
led 1 3,4::CTO:6
led 2 4,4::CTO:6
led 3 5,4::CTO:6
led 4 6,4::CTO:6
led 5 7,4::CTO:6
led 6 8,4::CTO:6
led 7 9,4::CTO:6
led 8 10,4::CTO:6
led 9 11,4::CTO:6
led 10 12,4::CTO:2
led 11 13,4::CTO:2

# aux
aux 0 0 0 900 1300 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# vtx
vtx 0 3 0 0 1 900 1300
vtx 1 3 0 0 2 1400 1600
vtx 2 3 0 0 3 1700 2100

# rxfail
rxfail 5 s 1850

# master
set gyro_lowpass2_hz = 350
set dyn_lpf_gyro_min_hz = 280
set dyn_lpf_gyro_max_hz = 700
set acc_calibration = -66,100,14,1
set mag_bustype = I2C
set mag_i2c_device = 1
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 1
set baro_hardware = NONE
set rssi_src_frame_errors = ON
set serialrx_provider = SBUS
set dshot_burst = AUTO
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = AUTO-LAND
set align_board_yaw = 180
set current_meter = ADC
set battery_meter = ADC
set beeper_inversion = ON
set beeper_od = OFF
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 180
set osd_rssi_alarm = 98
set osd_vbat_pos = 2401
set osd_rssi_pos = 2168
set osd_link_quality_pos = 2200
set osd_rssi_dbm_pos = 2135
set osd_tim_2_pos = 2422
set osd_flymode_pos = 2104
set osd_vtx_channel_pos = 2081
set osd_current_pos = 2368
set osd_warnings_pos = 14633
set system_hse_mhz = 8
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5880
set max7456_spi_bus = 2
set dashboard_i2c_bus = 1
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set name = TinyGO

profile 2

rateprofile 2

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: GEP-12A-F4 Flight Controller 12A ESC (GEPRC)
 
 - Other components: tranceiver and receiver: GEPRC GR8 2.4GHz 8CH FHSS, VTX: GEP-VTX200-Whoop, Camera: Caddx Loris 4K 60fps
 
 - How are the different components wired up:  GR8 receiver connerced to flight controller with three wires: GND, 5V, and UART2_RX (Receiver) 

I am talking about ready to flight kit GEPRC TinyGO 4K FPV Whoop RTF

m4rtin74, 2022-01-01:
Hi, Maybe update the firmware.
The 4.2.5 release is more than a year old.

kvark85, 2022-01-01:
Firmware was updated to 4.2.11, still reproducible.

### Flight controller configuration

```
diff

# version
# Betaflight / STM32F411 (S411) 4.2.11 Nov  9 2021 / 20:28:23 (948ba6339) MSP API: 1.43
# config: manufacturer_id: GEPR, board_name: GEPRCF411, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

board_name GEPRCF411
manufacturer_id GEPR

# name: TinyGO

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0

# feature
feature LED_STRIP

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 2,4::CTO:6
led 1 3,4::CTO:6
led 2 4,4::CTO:6
led 3 5,4::CTO:6
led 4 6,4::CTO:6
led 5 7,4::CTO:6
led 6 8,4::CTO:6
led 7 9,4::CTO:6
led 8 10,4::CTO:6
led 9 11,4::CTO:6
led 10 12,4::CTO:2
led 11 13,4::CTO:2

# aux
aux 0 0 0 900 1300 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# vtx
vtx 0 3 0 0 1 900 1300
vtx 1 3 0 0 2 1400 1600
vtx 2 3 0 0 3 1700 2100

# rxfail
rxfail 5 s 1850

# master
set gyro_lowpass2_hz = 350
set dyn_lpf_gyro_min_hz = 280
set dyn_lpf_gyro_max_hz = 700
set acc_calibration = -74,96,-92,1
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = AUTO-LAND
set align_board_yaw = 180
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 180
set osd_rssi_alarm = 98
set osd_vbat_pos = 2401
set osd_rssi_pos = 2168
set osd_link_quality_pos = 2200
set osd_rssi_dbm_pos = 2135
set osd_tim_2_pos = 2422
set osd_flymode_pos = 2104
set osd_vtx_channel_pos = 2081
set osd_current_pos = 2368
set osd_warnings_pos = 14633
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5880
set dashboard_i2c_bus = 1
set name = TinyGO

profile 2

rateprofile 2

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

# 
```

klutvott123, 2022-01-02:
> Expected behavior
>
> Case 1 (rssi_src_frame_errors = ON):
>
>     RSSI value is always 99;
>     RSSI dBm value is always -130;
>     link quality is always 9.
>
> Case 1 (rssi_src_frame_errors = OFF): [...]

Is this what happens or is it what you expect to happen?
I don't have goggles to see what happens in the OSD, but if i set rssi_src_frame_errors = on it shows 100% in the configurator. When I turn off the radio it drops to 0%

kvark85, 2022-01-02:
### Actual behavior
Case 1 (rssi_src_frame_errors = ON):
- RSSI value is always 99;
- RSSI dBm value  is always -130;
- link quality is always 9.

Case 1 (rssi_src_frame_errors = OFF):
rssi_src_frame_errors = OFF
- RSSI value  is always 0;
- RSSI dBm value  is always -130;
- Link quality  is always 9.

### Expected behavior
There must be a way to observe the quality of the data transfer.
It doesn't matter if it shows as RSSI or as link quality.

kvark85, 2022-01-02:
@klutvott123, are you saying that the ``` set rssi_src_frame_errors = on ``` setting is working correctly for you? Is your transmitter and receiver GEPRC GR8 2.4GHz 8CH FHSS?

klutvott123, 2022-01-02:
It sure looks like it's working. frsky XSR sbus/smartport

kvark85, 2022-01-02:
I guess the issue is reproducible only with [GEPRC GR8 2.4GHz 8CH FHSS](https://aliexpress.ru/item/1005001840337405.html) remote controller and receiver

daleckystepan, 2022-01-11:
@kvark85 You probably need to use one of the AUX channels to transmit RSSI values from TX to your RX (FC).
You can try something like
https://www.youtube.com/watch?v=t-evOAS9Mkg
https://oscarliang.com/rssi-ppm-channel-taranis/

kvark85, 2022-01-26:
as @daleckystepan advised, I bought another transmitter and another receiver. So it is impossible to get RSSI and link quality for transceiver and receiver: GEPRC GR8 2.4GHz 8CH FHSS. If you have one, just buy another which supports RSSI, for instance, FrSky

kvark85, 2022-01-26:
@daleckystepan, thanks, your links were helpful. I bought FrSky receiver and now I have RSSI in OSD. I guess the ticket may be closed because I don't have an issue anymore.

daleckystepan, 2022-01-26:
@kvark85 Glad that it helped you.

closed 2022-01-26 by kvark85