betaflight/betaflight #11216: Pinio stuck after arm

armandogallegos1990, 2022-01-02:
### Describe the bug
Using pinio to power on and off the Menace RC COB LED kit. When running the latest RC (24-12-2021 01:50 -4.3.0-RC1) the led kit will get stuck in the on or off state after arming and can no longer be toggled. When running an older nightly (31-08-2021 06:53 - 4.3.0 #2500) with the same diff the bug is not present. 

NOTE: Bug is not present when plugged in to usb. Only when usb is unplugged and running on battery power. 

### To Reproduce
Wire a vtx switch or led switch to a uart and remap to PINIO 1. Diff will set USER 1 as the mode to control the switch. toggle the switch on and off after arming. 

### Expected behavior
After arming the pinio device will not function as intended. 

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
# resources
resource SERIAL_TX 4 NONE
resource PINIO 1 A00

# aux
aux 0 0 0 1700 2100 0 0
aux 1 40 0 1700 2100 0 0

# master
set serialrx_provider = GHST
set vtx_band = 0
set vtx_channel = 0
set vtx_power = 0
set vtx_freq = 0
set pinio_box = 40,40,40,40

profile 0

rateprofile 0

# end the command batch
batch end
```
PASTE THE OUTPUT OF 'resource show all' HERE
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: PINIO 1
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: LED_STRIP
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH3 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

### Setup / Versions
 
 - Flight controller: BETAFPVF405, purchased from pirodrone and five33. Happens on 4 of them
 
 - Other components: IMRC GHST ATTO, MENACE RC COB LED KIT
  
 - How are the different components wired up: COB LED KIT wired to vbat, GND, and TX4

abwyatt, 2022-01-02:
I think I am having a similar or related problem. Using PINIO to control a Caddx Peanut camera from a switch (USER 1). Was working fine on 4.3 nightlies before RC1. Since RC1, the switch works before arming but stops working after arming.

m4rtin74, 2022-01-02:
I have a similar problem. I control the switching on and off of VTX with USER1 (also called realpit mode). The difference is that it doesn't work for me even before arming. It sticks to the position it is in when the drone is plugged in. I use 4.3.0 # 2598

JDRotor, 2022-01-02:
I have same issue with PINIO 1 operating vtx.  I traced the problem to nightly release #2570 17-12-21 and all later nightlies (latest tested #2599).  After an arming cycle (on and off) PINIO will no longer cycle - (but will function correctly during 1st Arming but before 1st dis-arming command).  I have experienced the issue on multiple F7 FCs - AikonF7 and LUX F7 (MATEKF722SE).

limonspb, 2022-01-02:
> I have same issue with PINIO 1 operating vtx. I traced the problem to nightly release #2570 17-12-21 and all later nightlies (latest tested #2599). After an arming cycle (on and off) PINIO will no [...]

ok so looks like the issue introduced here then:
https://github.com/betaflight/betaflight/pull/10813

abwyatt, 2022-01-03:
> > I have same issue with PINIO 1 operating vtx. I traced the problem to nightly release #2570 17-12-21 and all later nightlies (latest tested #2599). After an arming cycle (on and off) PINIO will no longer cycle - (but will function correctly during 1st Arming but before 1st dis-arming command). I have experienced the issue on multiple F7 FCs - AikonF7 and LUX F7 (MATEKF722SE).
> 
> Thanks for checking different versions. ok so looks like the issue introduced here then: #10813

Could this task timing issue also be causing the SA issue here 'SA "commands" not working as expected #11204' ?

JDRotor, 2022-01-03:
Thanks Steve.

limonspb, 2022-01-03:
> > > I have same issue with PINIO 1 operating vtx. I traced the problem to nightly release #2570 17-12-21 and all later nightlies (latest tested #2599). After an arming cycle (on and off) PINIO will no longer cycle - (but will function correctly during 1st Arming but before 1st dis-arming command). I have experienced the issue on multiple F7 FCs - AikonF7 and LUX F7 (MATEKF722SE).
> > 
> > 
> > Thanks for checking different versions. ok so looks like the issue introduced here then: #10813
> 
> Could this task timing issue also be causing the SA issue here 'SA "commands" not working as expected #11204' ?
> 
> In both cases, switching stops working after arming.

i think so.
please chekc these hex files from the comment with the fix:
https://github.com/betaflight/betaflight/pull/11198#issuecomment-1003842143

these are unified target hex files, there is one extra step when flashing them:
**1. In the configurator select your board, then 4.2 or 4.2 firmware and click download.**
2. only after that select local file and flash
3. Agree to apply custom defaults as usual.

JDRotor, 2022-01-03:
Loaded the Hex file as suggested and on a bench test the issue appears to be resolved.

 

Too late to test fly but I see no issues at this point.   Thanks for the quick fix!

 

Regards,

Joe

 

#11198 (comment) <https://github.com/betaflight/betaflight/pull/11198#issuecomment-1003842143>

abwyatt, 2022-01-03:
> > > > I have same issue with PINIO 1 operating vtx. I traced the problem to nightly release #2570 17-12-21 and all later nightlies (latest tested #2599). After an arming cycle (on and off) PINIO will no longer cycle - (but will function correctly during 1st Arming but before 1st dis-arming command). I have experienced the issue on multiple F7 FCs - AikonF7 and LUX F7 (MATEKF722SE).
> > > 
> > > 
> > > Thanks for checking different versions. ok so looks like the issue introduced here then: #10813
> > 
> > 
> > Could this task timing issue also be causing the SA issue here 'SA "commands" not working as expected #11204' ?
> > In both cases, switching stops working after arming.
> 
> i think so. please check these hex files from the comment with the fix: [#11198 (comment)](https://github.com/betaflight/betaflight/pull/11198#issuecomment-1003842143)
> 
> these are unified target hex files, there is one extra step when flashing them: **1. In the configurator select your board/target** 2. only after that select local file and flash 3. Agree to apply custom defaults as usual.

Thanks for quick work. Much appreciated.

limonspb, 2022-01-03:
> Thanks for quick work. Much appreciated.

@SteveCEvans  is the MAN

m4rtin74, 2022-01-03:
The PINIO and VTX power switches already work perfectly for me. Thanks!

SteveCEvans, 2022-01-06:
Fixed by: #11219

closed 2022-01-06 by SteveCEvans