betaflight/betaflight #11221: F411 opens some settings that make the CPU usage too high to stutter unable to connect to the Betaflight Configurator-4.3.0 RC1

Xiang-FPV, 2022-01-03:
### Describe the bug
In the 4.3.0 RC1.The Mamba f411 open 8k PWM loop frequency, DShot600 and Bidirectional DSHOT（Other defaults） at the same time, it will be due to high CPU usage and stutter, resulting in the inability to connect to the Betaflight Configurator.Of course, this can only be solved by holding down the boot button to re-update the firmware.（Probably the same will happen to other f411 flight controllers）

### To Reproduce
1.F411 Flight Controller re-updates firmware to BF4.3.0 RC1 (enables full chip erase)
2.Select the corresponding firmware and start loading online
3.Flash firmware
4.Click connect after the update is successful and select "Apply Custom Defaults"
5.Connet
6.Set the 8k PWM loop frequency—Save. Set the DShot600 and turn on Bidirectional DSHOT（Other defaults）
7.Save and reconnect
8.Then there is the slow connection and the CPU usage is too high

### Expected behavior
I want f411FC to prompt us when these settings are enabled at the same time, or to disable f411FC to turn on these settings at the same time to ensure that flight controller does not fail to connect to betaflight Configurator. Maybe I will avoid it, but there is no guarantee that others will not open it by mistake, especially some novices

### Flight controller configuration
```
# diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Dec 24 2021 / 18:32:13 (norevision) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF411, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

board_name MAMBAF411
manufacturer_id DIAT

# master
set pid_process_denom = 1
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: ADC_BATT
A01: ADC_CURR
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: GYRO_EXTI
B01: FREE
B02: BEEPER
B03: MOTOR 1
B04: MOTOR 2
B05: FREE
B06: MOTOR 3
B07: MOTOR 4
B08: FREE
B09: FREE
B10: INVERTER 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : MOTOR 1
TIM3:
    CH1 : MOTOR 2
TIM4:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 - Flight controller: Mamba f411  Purchase on the DIATONE website
 - Other components: No external equipment soldered
 - How are the different components wired up:

Asizon, 2022-01-03:
@Xiang-FPV what is too hihg?? <75% is good, please read new 4.3 tuning notes.

Xiang-FPV, 2022-01-03:
@Asizon CPU Load is too high （Approximately 93%）

haslinghuis, 2022-01-03:
It does not make sense to run DShot 600 on F411. Please use Dshot 300 instead.
```
F4 processors should run a 4k PID loop.
F411 users should use DShot300, not DShot600, and the F411 chip should be overclocked to 120Mhz.
```
Please see Tuning Notes: https://github.com/betaflight/betaflight/wiki/4.3-Tuning-Notes#betaflight-43-introduction

Xiang-FPV, 2022-01-03:
@haslinghuis I know that, I also look at the wiki, but what if it was opened by mistake, or if it was accidentally set up by a newbie?

Xiang-FPV, 2022-01-03:
Just out of curiosity: how much CPU would you use if you turned these high-load Settings on? As a result, when I saved and reconnected, the connection was slow (i.e. no connection) and I couldn't roll back the Settings. So I mean even if I accidentally set it wrong, I can go ahead and connect to BetaFlight Configurator and reset it. But now the situation is that I "accidentally" set 8K, two-way Dshot, Dshot600, found that the flight control does not work, I want to connect the ground station to reset, but can not connect

haslinghuis, 2022-01-03:
Good call. Should we disable Dshot 600 on F411 targets completely or not?

haslinghuis, 2022-01-03:
My Matek F411 is running at 40%
Please run tasks in cli

Xiang-FPV, 2022-01-03:
In short, if the setup error can be connected to the computer after the reset I want to disable DShot600 is not necessary, and I am generally 8KPWM, DSHOT600 does not have Bidirectional Dshot turned on

Xiang-FPV, 2022-01-03:
If only two or one of these three Settings are the same as the one I described, the flight control can be used normally. Therefore, I think it is possible to disable the simultaneous enabling of 8K PWM Dshot600 and Bidirectional DShot, Or f411 enabling all three Settings at the same time to give us a hint. This is just one of my ideas. If it is not reasonable, you do not need to adopt it. Of course, if you think of a better method than mine, I will definitely support it.

MJ666, 2022-01-03:
I'm also running my BetaFPV F411 RX board with DSHOT600 without issues. 4k PID loop and bidirectional DSHOT off. CPU load is 32% with 108MHz clock. I would not recommend to disable DSHOT600 for the F411 in general. Likely a combination of 8K PID loop and bidirektional DSHOT could be disabled to avoid this problem.

Xiang-FPV, 2022-01-03:
@MJ666 That's good advice!

phobos-, 2022-01-04:
@haslinghuis maybe we should set the default clock to 120MHZ for F411?
> F4 processors should run a 4k PID loop. F411 users should use DShot300, not DShot600, and the F411 chip should be overclocked to 120Mhz.

SteveCEvans, 2022-01-04:
No, 120MHz is pushing it a bit. With other changes we’ve made (notably https://github.com/betaflight/betaflight/pull/11197 and https://github.com/betaflight/betaflight/pull/11222) we need to consider changing the advice you quote from the tuning guide as 108MHz should be OK.

turboed13b, 2022-01-05:
120mhz causes problems on spi boards. I had someone else with this issue on the betafpv 1s elrs board. Enabled dshot600 and couldn't connect to betaflight after that.

phorensic, 2022-01-07:
I have an iFlight SucceX Micro F4 FC which is F411 and an iFlight SucceX Micro 15A BLheli_S ESC which is DSHOT600 capable, which I always ran (DSHOT600) on bf 4.2.x. With 8k/8k on 4.2 I was at 16% CPU load. Flashed 4.3 and tried 8k/8k DSHOT600 - 61% CPU load and super laggy in configurator and supper lagging throttle on the quad (I actually stuck to the ceiling at full throttle for almost one second while the FC processed what was going on and dropped the throttle, lol). Flight controls were somewhat laggy, but the throttle was unflyable. Dropped to 4k PID loop, CPU dropped to 30%, configurator has no lag, throttle is almost back to normal. Either it is normal now, still _slightly_ laggy, or I'm looking for the problem now and feeling a soft of placebo effect lag, hard to tell. I noticed the option "Auto Smoothness" (now renamed "Auto Factor") default was increased from 10 to 30 and it mentions input lag. So that's my next setting to try and tune out the last bit of lag.

In any case, glad to see this is being discussed. From previous research I saw that 4.3 just showed more CPU utilization without actually using more CPU, as in the number in configurator was just more accurate. After testing now I feel like it's maybe the opposite, 4.3 is actually using more CPU, it's a bit heavier. That's fine, as long as I can configure around it. I don't need 8k PID loops anyway, I just fly LOS with low skill. Overlocking the F411 sounds fun, too!

SteveCEvans, 2022-01-07:
@phorensic in 4.2 the CPU load was based on how many tasks were waiting, ready to run. 4.3 instead reports the % of time for which the CPU is busy. 4.3 enforces rigid timing to ensure everything is being done on time. As result filtering is much more precise etc. It’s simply not possible for an F411 to execute the gyro, filter and PID tasks, handle bidirectional DSHOT, and RX or OSD in 125us (8kHz loop time). In 4.2 this timing would overrun (a lot) to the detriment of flight performance. Running at 4kHz PID loop means only one of the filter or PID tasks are run in any 125us cycle, leaving time for the (rewritten for 4.3) RX/OSD tasks to run.

phorensic, 2022-01-07:
Thanks for the explanation. Looks like you (the devs) have been quite hard at work under the hood. With major software changes it is often the case where the old hardware support has to be dropped in favor of getting new features to work, and it looks like that may be the case here with an F411. Just not enough power for the next evolution. I'm not up to speed on FC hardware that will fit in my insanely small Tadpole 3", but it's time to start researching what might have a bit more CPU horsepower. The browser tabs are growing!!

SteveCEvans, 2022-01-07:
Overclocking the F411 to 120MHz makes quite a difference.

hydra, 2022-01-09:
How about something in the FC that detects if serial latency is too high and disables or de-prioritizes something so that the serial link works again?  There is already the `MSP` and `CPU_LOAD` arming prevention states that could possibly be used.

Forcing the user to erase their entire config so that they can boot and reconnect from the UI is a bad user experience.

Xiang-FPV, 2022-01-09:
@hydra @SteveCEvans Yeah, now that you mention it, it does happen,For example, when the FC CPU load is very high (60% ~ 70%), I feel the refresh in the BetaFlight Configurator is very slow。 What's the reason? Can it be solved？

JAL513, 2022-01-10:
@SteveCEvans, I'm running an F411 with 4kHz PID loop, bidirectional DSHOT300, overclock 120mhz.  I was getting issues with real slow updates to the OSD menus, VTX power on a switch real slow and "quirky", and my OSD profile adjustments were also really slow to update and caused flight issues after arming.  When things were moving slow, I could feel lag in the sticks.  

I just tried lowering the PID loop to 2kHz and EVERYTHING is fixed, running absolutely amazing.  I just flew the best 6 packs of my life!  I feel no sacrifice from the lower PID loop frequency.

Just wanted to let you know in case it helps troubleshoot some of the issues I'm reading about people having.  Thanks for all your work, all the BF devs are heroes in my eyes! Let me know if I can help is some way.

SteveCEvans, 2022-01-10:
@JAL513 Are you running RC2? I’m curious to know what your `dma show` gives. Are you using timer based or bitbang DSHOT?

JAL513, 2022-01-10:
Sorry, should have specified that!  I just moved to RC2 on the Configurator and BF.

Here is my dma show 
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: MOTOR 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

I'm not smart enough to know how to answer that last one... Let me know how to check.

JAL513, 2022-01-10:
> Sorry this post is a mess.
> 
> Please indicate if I should report this elsewhere or how I can help.
> 
> -pitts.mo
> 
> 
> 
> I have had issues with OSD update rate in 4.3.0 builds and have skimmed over multiple 4.3.0x issue reports that I suspect may be rooted in a single underlying cause of task priority and timing that may depend upon configuration and may manifest a variety of undesirable behaviors.
> 
> 
> 
> **My current guess is that @hydra may be onto a valid global solution.** 
> 
> (oops, after re-read some of the posts here I think @hydra was only referring to configurator connection. I still feel there may be a need to reconfigure some of the tasks to perform better at lower priorities or add auto/user configurable task priorities ?)
> 
> 
> 
> **my reasoning:**
> 
> I found the most predictable improvements in OSD update rate for my test build only occurred after I tested setting my ELRS TX at lower rates. ELRS @50hz allowed my OSD updates to feel almost normal and the unpredictable lag in OSD updates was observed to increase as I increased the ELRS TX rate. However, on Betaflight 4.2.11 I normally run 250hz without any problem.
> 
> 
> 
> some key details about my 4.3.0x test build:
> 
> I began testing 4.3.0 builds since RC1 and I have been unable to get timely OSD updates.
> 
> EMAX Tinyhawk II w/MATEKF411RX
> 
> adjrange 0 0 1 900 2100 29 1 0 0 (OSD selection)
> 
> set dshot_bidir = ON
> 
> set serialrx_provider = CRSF (normally use ERLS 250hz)
> 
> set motor_pwm_protocol = DSHOT300
> 
> set rpm_filter_harmonics = 1
> 
> 
> 
> Some other things I tired that appeared to affect the timeliness of OSD updates but not actually improve the OSD usability much:
> 
> Adding the official ELRS 250 preset
> 
> set pid_process_denom = 2 (tried 4/8)
> 
> set cpu_overclock = 108MHZ (tried 120)
> 
> set max7456_clock = NOMINAL (tried HALF/DOUBLE)
> 
> set dshot_bitbang = AUTO (tried OFF/ON )
> 
> set tasks.c -increasing TASK_PERIOD_HZ and priority for [TASK_OSD]
> 
> set osd.h #define OSD_FRAMERATE_MIN_HZ 10 and #define OSD_FRAMERATE_DEFAULT_HZ 20
> 
> 
> 
> 
> 
> 
> 
> 

@pitts-mo try moving PID loop to 2khz that made everything work as expected for me. Curious if you see the same.

JAL513, 2022-01-10:
@pitts-mo Sorry it didn't work out.  
I'm running Tracer, so fixed 250hz without issue, interesting you have to go down to 50hz on ELRS.  The devs will get it sorted out.

Good luck!

phorensic, 2022-01-11:
@pitts-mo Bidirectional DSHOT doubled the CPU usage on my F411. That's the only thing I didn't see you try turning off. I decided I didn't have the horsepower on my F411 to run it (on 4.3.x)

pitts-mo, 2022-01-12:
Thanks all,
I have found a compromise that allows me a usable 4.3.0x test experience. I can now use dshot300, bidirectional, bitbang and ELRS @150hz :-) with only 30% CPU usage. I still observe the odd OSD lag even while monitoring CPU to be around 30% but updates are able to sneak in just reliable enough to be usable.

I will add an update to #11274 as some of my testing suggests my issue has some odd relation to adjustments.

My travels across issue reports still makes it feel like an underlying issue... but maybe we'll just need find and rework all the individual task events that seem to be causing problem.

hydra, 2022-01-14:
If you guys were experiencing an issue where the configurator becomes unusable after you use it to make configuration changes (i.e. what it was designed to do) then a bug report should be raised.  To me, it feels like there's more of a general problem that needs to be addressed in the firmware when the CPU load is too high.

SteveCEvans, 2022-01-14:
@pitts-mo Thanks. You’ll see that I’ve pushed a further commit to that PR to allow the OSD task to be scheduled if it’s too overdue. It will typically be well behaved, but on processors which are heavily loaded both the RX and OSD tasks can be expedited at the expense of accurate timing.

hydra, 2022-02-14:
@pitts-mo Have you tried RC3?

pitts-mo, 2022-03-21:
Sorry I missed your inquiry @hydra,
Yes, I have since installed RC3 on all my quads. I can not recall observing the quirky control/OSD lag during flight with RC3. However, I do still regularly see weird control/OSD lag occurring prior to arming RC3. I am still utilizing serial ELRS 2.4ghz @150hz as prior to RC3 I found higher RC rates associated to notable increase in lag/weirdness.

I have not tested nightly builds since RC3 but I did observe updates slip through from @hydra and @SteveCEvans I suspect may continue to reduce the weird instances of control/OSD lag I experience prior to arming.

closed 2022-04-28 by github-actions[bot]