betaflight/betaflight #11227: Board Name identify by betaflight 10.8.0 configurator but not board name found by "auto detect"

Ls4life, 2022-01-04:
### Describe the bug
 I am attempting to upgrade my flight controller with board name "JHEMCUF405" when running the upgrade board procedure betaflight auto detect do not recognize or identify the board name.
I seached through the list of board name on the drop down and the board name is not there. 

### To Reproduce
 
1 - Open betaflight
2 - Connect drone or FC with board name "JHEMCUF405"
3 - On betaflight configurator click on the "Upgrade Firmware" tap located on the upper right hand of the screen and to the left of the tap "disconnect"
4 - if necessary click "auto detect" if no board name that match your board name come up, open the drop down and search for the board Name. 
5 - The board name is not listed.
### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
###ERROR: diff: NO CONFIG FOUND###
# start the command batch
batch start

board_name JHEMCUF405
manufacturer_id JHEM

# name: Ls4inchQAV-S

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 B06
resource PPM 1 B08
resource PWM 1 B08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 2 C13
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05
resource PULLDOWN 1 B08

# timer
timer B08 AF3
# pin B08: TIM10 CH1 (AF3)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer B11 AF1
# pin B11: TIM2 CH4 (AF1)

# dma
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0
dma pin A02 0
# pin A02: DMA1 Stream 0 Channel 6
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin B11 0
# pin B11: DMA1 Stream 7 Channel 3

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD

# map
map TAER1234

# serial
serial 1 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led
led 0 8,6::CB:8
led 1 9,6::CB:8
led 2 10,6::CB:8
led 3 11,6::CB:8

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_calibration = 37,-38,-51,1
set mag_bustype = I2C
set mag_i2c_device = 2
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 2
set baro_hardware = NONE
set serialrx_provider = CRSF
set blackbox_device = SPIFLASH
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 2
set current_meter = ADC
set battery_meter = ADC
set beeper_inversion = ON
set beeper_od = OFF
set small_angle = 180
set gps_provider = UBLOX
set gps_set_home_point_once = ON
set osd_vbat_pos = 14570
set osd_crosshairs_pos = 14541
set system_hse_mhz = 8
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 4
set vtx_freq = 5769
set max7456_spi_bus = 2
set dashboard_i2c_bus = 2
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW90
set gyro_1_align_yaw = 900
set name = Ls4inchQAV-S

profile 0

rateprofile 0

# end the command batch
batch end

```
PASTE THE OUTPUT OF 'resource show all' HERE
```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 B06
resource PPM 1 B08
resource PWM 1 B08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 2 C13
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05
resource PULLDOWN 1 B08

### Setup / Versions
 
 - Flight controller: board_name JHEMCUF405
 - manufacturer_id JHEM

 
 - Other components:
 
 - How are the different components wired up:

 The only issue is that the board name is not on your list so it can be updated.

haslinghuis, 2022-01-04:
Has it changed to JHEF?

https://github.com/betaflight/unified-targets/pull/475

[image]

Ls4life, 2022-01-04:
> Has it changed to JHEF?
>
> betaflight/unified-targets#475
>
> [image: Screenshot from 2022-01-04 19-39-03]
> <https://user-images.githubusercontent.com/8344830/148107684-28b67c5b-092f-415c-ac62-73250c4becc6.png>

Is this what I am supposed to do? Select the maketek board?

haslinghuis, 2022-01-04:
No no, it's just an example. Please save your current diff before flashing. I was suggesting to try JHEF405
Ask your supplier / reseller for the right target

Ls4life, 2022-01-04:
I did picked from the drop down list the jhemf405 but betaflight provide a
message as the one you used as example indicating that my board name and
the board I selected had a name missed match and if I wanted to continue. I
was not sure if I should proceed.

haslinghuis, 2022-01-04:
Can't find the codeowner for this target. Do they still exist? Please ask your reseller for the right target.

Ls4life, 2022-01-04:
Thanks I will contact them to find out and let you know.

Ls4life, 2022-01-11:
Hello, just to follow up on this issue, I spoke to getfpv from where I
bought the board in question. They confirmed that the manufacturer who made
the board is JHEM. I guess is a new board. Do you think you will add this
board to the list in betaflight firmware update? The board name is
JHEMCF405. Or do you think I could run it with the default firmware version
it comes with and use the betafight configurator 10.8.0. It seem the board
was configured from with firmware 4.2.0 for what I can tell when I connect
the board to betaflight.

haslinghuis, 2022-01-11:
Bad luck. We don't have JHEM in unified targets. The manufacturer should supply needed config files for their targets by creating a PR in the unified targets repository as we don't have the hardware samples to do so.

Ls4life, 2022-01-12:
Thank. Let me check on that.

Ls4life, 2022-01-12:
Here is what I found on their website. I am attaching a capture so that you
see what appears to be the firmware they are using.  It seem It is call
betaflightF4.Hex

haslinghuis, 2022-01-12:
This is not official betaflight firmware. Closing this issue as the manufacturer is not even listed in our database.

closed 2022-01-12 by haslinghuis

Ls4life, 2022-01-12:
Ok, thanks.