betaflight/betaflight #11233: VTX management via OSD don't work - BETAFPVF4SX1280 - 4.3.0 RC-1

Martivip, 2022-01-06:
### Describe the bug

Vtx management (channel, band, power) via OSD don't work on BETAFPVF4SX1280 target.
Probably the issue is related to ELRS integration in BF https://github.com/betaflight/betaflight/pull/10788#issuecomment-1001550363
Is worth to mention that Smartaudio on this target is managed via soft serial.
Also I've noticed that changing the VTX parameters in ELRS system via ELRS Lua v2 and VTX Administrator functionality is working but on OSD VTX information element values stays unchanged.

### To Reproduce

Instal the RC-1  BETAFPVF4SX1280 target on BETAFPV SPI ELRS AIO Boards

### Expected behavior

Changing the VTX parameters via OSD should take the effect and changing the parameters via ELRS VTX Administrator should be considered under the OSD VTX information element.

### Flight controller configuration

```
# 

# diff

# version
# Betaflight / BETAFPVF4SX1280 (BHER) 4.3.0 Dec 24 2021 / 17:08:11 (norevision) MSP API: 1.44

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# name: Meteor65-ELRS

# feature
feature -LED_STRIP
feature -ANTI_GRAVITY

# serial
serial 0 1 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1425 1575 0 0
aux 2 2 1 1925 2075 0 0
aux 3 35 2 1300 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 350

# master
set gyro_notch1_hz = 400
set gyro_notch2_hz = 200
set acc_calibration = 108,-127,111,1
set max_check = 2000
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set current_meter = NONE
set ibata_scale = 179
set small_angle = 180
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 193
set osd_link_quality_pos = 2273
set osd_rssi_dbm_pos = 2305
set osd_tim_1_pos = 386
set osd_tim_2_pos = 2455
set osd_remaining_time_estimate_pos = 33
set osd_flymode_pos = 2423
set osd_throttle_pos = 2401
set osd_vtx_channel_pos = 2080
set osd_crosshairs_pos = 2253
set osd_current_pos = 343
set osd_craft_name_pos = 2441
set osd_home_dist_pos = 98
set osd_flight_dist_pos = 130
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 289
set osd_disarmed_pos = 2315
set osd_esc_tmp_pos = 163
set osd_esc_rpm_pos = 150
set osd_core_temp_pos = 278
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = NTSC
set expresslrs_uid = 179,193,51,58,47,191
set expresslrs_rate_index = 1
set expresslrs_switch_mode = WIDE
set name = Meteor65-ELRS

profile 2

# profile 2
set p_pitch = 51
set i_pitch = 99
set d_pitch = 42
set f_pitch = 105
set p_roll = 46
set i_roll = 94
set d_roll = 39
set f_roll = 99
set p_yaw = 50
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 25
set d_min_pitch = 28

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end

# 
```
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: BEEPER
A15: RX_SPI_CS
B00: FREE
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_EXPRESSLRS_RESET
B10: MOTOR 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : SERIAL_TX 11
TIM2:
    CH1 : MOTOR 2
    CH3 : MOTOR 3
TIM3: FREE
TIM4:
    CH2 : MOTOR 4
    CH3 : MOTOR 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: 
 
  BetaFpv F4 1S 5A AIO Brushless Flight Controller (ELRS 2.4G)
 - Other components:
 BetaFpv M03 25-350mW 5.8G VTX
 
 - How are the different components wired up:
 BetaFpv M03 25-350mW 5.8G VTX, smart audio hooked up to SA port on the board, SA on this board is managed via Soft serial functionality

wildjokerss, 2022-01-07:
i believe it will work if you install the latest nightly, I had the same problem. non the less if you set it via OSD and restart the FC, it should change

betafpv-engineer, 2022-01-07:
A passage from phobos- : In your case EEPROM_WRITE is causing the CPU to starve the RX task, and it's losing the connection briefly due to that. There is nothing that can be done to prevent it. It is described as a known issue in the [#10788 (comment)](https://github.com/betaflight/betaflight/pull/10788).

You can, as @wildjokerss said, after the OSD is set, reapply the power to the FC.

Martivip, 2022-01-10:
Resolved by RC2

Martivip, 2022-02-04:
It doesn't work with RC3

SteveCEvans, 2022-02-05:
@Martivip  can you please confirm if, when using TX1 you’re able to configure the VTX using the configurator. If so then the SA is OK.

Doing this via the OSD and the ELRS link is adding other things to the equation. We have a known issue where writing to FLASH following a config change results in ELRS dropping the connection, but that shouldn’t stop the application of that configuration.

Martivip, 2022-02-05:
@SteveCEvans this is really good question because I have really interesting test observations.
This is exactly dropped ELRS connection during the VTX change but unfortunatelly this cause change to be effective even if I repower the FC.
But this behavior is really interesting, when my ELRS TX is switched OFF (no link between ELRS TX and ELRS SPI RX) I'm capable to effectively change VTX settings via Configurator, but when my RC link between ELRS TX ans ELRS SPI RX is established then when I trying to change ie. VTX channel then I see in the googles that channel is changing but for a while it automatically reverting to previous channel, in other words configurator trying to change but this change is not stored in EEPROM. Another scenario : ELRS Tx is Off, I'm changing via configurator VTX channel and the change **is effective**, next I switching ON the ELRS Tx and the RC link is establisched and to my surprise VTX by themselve changing (reverting) channel to previously set.

CMartin249, 2022-02-19:
> ... Another scenario : ELRS Tx is Off, I'm changing via configurator VTX channel and the change **is effective**, next I switching ON the ELRS Tx and the RC link is establisched and to my surprise VTX by themselve changing (reverting) channel to previously set.

I believe you have VTX Admin configured in ELRS LUA. When connecting, it configures VTX to what is specified in LUA.

Martivip, 2022-02-19:
@CMartin249 yes of course, in general no matter via which method I will use to change VTX channel ELRS Lus VTX administrator, OSD, BF Lupa, right now channel is charging but when I repower the FC it reverting the VTX channel to previously set

CMartin249, 2022-02-19:
I was thinking the vtx admin feature was what was resetting it. It doesn't save the config, but will set the config to what is specified there every time the rx connects.

Martivip, 2022-03-13:
Hi @SteveCEvans i've tested today with the latest Development build no. 2648 and still VTX management via ELRS RC together with ELRS SPIrx ELRS AIO FC board don't work as it should.
If there in no RC Link (TX is off) then I can apply VTX changes via BF Configurator.
If RC Link is established then if I change channel from R1 to R2 via VTX Administrator Lua script then change it becomes effective and the channel is set to R2 but after the FC repower RC Link is established immediately but channel comes back to previously set R1 channel an after 8 seconds is restores to requested above R2. If I connect FC to BF Configuration after applying this VTX channel change to R2 via VTX Administrator Lua script but with no RC link established  it **shows R1 channel set** definitely there is something in EEPROM writes to FC

haslinghuis, 2022-04-02:
@Martivip please test again with latest development firmware and report back

CMartin249, 2022-04-03:
I applied build 2662. Still having issues. ELRS VTX Admin feature does not get saved after changing the channel. This issue and #11448 appear to be related. A comment on that thread indicates saves are currently disabled on ELRS SPI due to disconnect issues.

Martivip, 2022-04-03:
@haslinghuis with Dev. build no. 2662 still unfortunately don't work via
**OSD :** when changing a channel from R1 to R2 I see in the googles that something going's on but after second or two it comes back to previously set R1. Another weird thing is if I repower FC after that above change, then I confirm in the googles  that channel is set to R2 but for few seconds is restores to R1 so definitely something is wrong.
**ELRS LUA - VTX Administrator :** channel change form R1 to R3 is effective, but when I repower the FC it starts at channel R2 (probably remember setting form above change via OSD) and for a few seconds it restores to channel R3.
**BF LUA VTX Settings**: the same behavior as above ELRS LUA - VTX Administrator 

So generally there is something wrong.

klutvott123, 2022-05-01:
This should solve the issue when changing vtx settings through the betaflight lua scripts at least https://github.com/betaflight/betaflight/pull/11573

Martivip, 2022-05-03:
@klutvott123 ok after upload firmware from #11573 
With VTX Administrator band set to OFF the VTX management works well via OSD and BF Lua script 👍 
And the VTX Administrator works too but right now it causes LQ drop to zero in few seconds interval :(

Generally good job 👍 another step ahead:) it look like only the VTX Administrator need to be fixed

Martivip, 2022-06-03:
@klutvott123 after merge of the #11573 and #11597 I've uploaded the latest dev. build
and only there is still an issue with VTX Admin, during the disarm LQ RC link drops to zero and back in few seconds interval

haslinghuis, 2023-12-17:
Closing as inactive

closed 2023-12-17 by haslinghuis