betaflight/betaflight #11244: F411 OSD still laggy on the last nightly build - 2606

AndreaCCIE, 2022-01-07:
### Describe the bug
Using the latest nightly build, 2606, the OSD on different F411 (Matek and GepRc) are laggy.
I have set the CPU to 108MHz and max7456_clock = DOUBLE.

Please check the video:
https://youtu.be/_e9bGEfYH7M

### To Reproduce
Install the latest build on one of the affected FCs (it looks to me that not all the brands are subjected to this issue as I'm the only one reporting it)

### Expected behavior
OSD should refresh correctly

### Flight controller configuration

```
# 

# diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jan  7 2022 / 03:13:16 (4f8579afc) MSP API: 1.44
# config: manufacturer_id: GEPR, board_name: GEPRCF411, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

board_name GEPRCF411
manufacturer_id GEPR

# name: CineLog25

# resources
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B04
resource MOTOR 4 B05
resource MOTOR 5 A00
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A00
resource SERIAL_RX 11 B10
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource PINIO 1 A08
resource FLASH_CS 1 B03

# timer
timer C06 AF2
# pin C06: TIM3 CH1 (AF2)
timer C07 AF2
# pin C07: TIM3 CH2 (AF2)

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin C06 0
# pin C06: DMA1 Stream 4 Channel 5
dma pin C07 0
# pin C07: DMA1 Stream 5 Channel 5

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1550 2100 0 0
aux 1 1 3 900 1300 0 0
aux 2 13 3 1700 2100 0 0
aux 3 35 3 1700 2100 0 0
aux 4 36 5 1700 2100 0 0
aux 5 39 2 900 1300 0 0

# adjrange
adjrange 0 0 2 900 1300 29 2 0 0
adjrange 1 0 2 1300 1700 29 2 0 0
adjrange 2 0 2 1700 2100 29 2 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 BAND_D   D CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# rxfail
rxfail 7 s 750

# master
set gyro_lpf1_static_hz = 450
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = 0,29,-119,1
set min_check = 1010
set max_check = 1990
set rc_smoothing_auto_factor = 90
set rc_smoothing_auto_factor_throttle = 50
set rc_smoothing_setpoint_cutoff = 15
set rc_smoothing_feedforward_cutoff = 15
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set align_board_yaw = 45
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set simplified_gyro_filter_multiplier = 180
set report_cell_voltage = ON
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_dbm_alarm = -100
set osd_tim1 = 2561
set osd_vbat_pos = 8289
set osd_link_quality_pos = 14391
set osd_link_tx_power_pos = 12747
set osd_rssi_dbm_pos = 12375
set osd_tim_1_pos = 14369
set osd_tim_2_pos = 215
set osd_flymode_pos = 12792
set osd_throttle_pos = 8289
set osd_vtx_channel_pos = 12779
set osd_current_pos = 8640
set osd_mah_drawn_pos = 12770
set osd_craft_name_pos = 364
set osd_pid_roll_pos = 391
set osd_pid_pitch_pos = 359
set osd_pid_yaw_pos = 423
set osd_warnings_pos = 14505
set osd_avg_cell_voltage_pos = 14401
set osd_disarmed_pos = 12555
set osd_esc_tmp_pos = 428
set osd_log_status_pos = 472
set osd_rate_profile_name_pos = 151
set osd_pid_profile_name_pos = 119
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 5
set vtx_freq = 5806
set vcd_video_system = PAL
set max7456_clock = DOUBLE
set gyro_1_sensor_align = CW90FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 900
set name = CineLog25

profile 2

# profile 2
set dterm_lpf1_type = PT3
set dterm_lpf2_static_hz = 0
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 16

rateprofile 0

# rateprofile 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 40
set tpa_breakpoint = 1550

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: PINIO 1
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: PREINIT 2
B04: MOTOR 3
B05: MOTOR 4
B06: MOTOR 1
B07: MOTOR 2
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 3
    CH2 : MOTOR 4
TIM4:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 1
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 2
DMA1 Stream 4: MOTOR 3
DMA1 Stream 5: MOTOR 4
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
# Betaflight / STM32F411 (S411) 4.3.0 Jan  7 2022 / 03:13:16 (4f8579afc) MSP API: 1.44
# config: manufacturer_id: GEPR, board_name: GEPRCF411, version: 7b156dec, date: 2021-07-26T13:19:23Z
# board: manufacturer_id: GEPR, board_name: GEPRCF411

pitts-mo, 2022-01-08:
Is lag or delay of OSD a known issue in the current 4.3.0 nightly builds or should @AndreaCCIE or I try to dig at this a bit more on our own?

more detail:
Today I started my first migration from Betaflight 4.2.11 to Betaflight 4.3.0. I can still fly my Emax TinyHawk II w/MATEKF411RX target on 4.3.0 RC1 or nightly 2606 builds but the OSD is oddly/excessively delayed and the OSD menus are just not usable. I have not tried to strip my config to test with minimal "defaults" yet.

pitts-mo, 2022-01-08:
@AndreaCCIE / @SteveCEvans
I did noticed my OSD lag was more flaky and less consistent that the video posted here but does "set cpu_overclock = 120MHZ" improve the OSD speed on this GEPRCF411 build?

background:
A reset to defaults and minimal config appeared to resolve the OSD lag I was seeing. I reloaded my config and changed "set dshot_bidir = ON" to "set dshot_bidir = OFF" on a hunch to find CPU usage dropped from 60 to 35 and also appeared to resolve my issue with OSD lag... Searching some more I found #11167 which lead me to:
# get cpu_overclock
cpu_overclock = 120MHZ
Allowed values: OFF, 108MHZ, 120MHZ
Default value: 108MHZ

AndreaCCIE, 2022-01-08:
@pitts-mo accotding to PR #11197, more recent than #11167, the overclock to 120MHz is not needed anymore.
Moreover, I don't want to get rid of DSHOT to have a working OSD.

pitts-mo, 2022-01-08:
Ok, Thank you for pointing me to the additional background. I agree that RPM Filters is critical and would prefer not to utilize any overclock.

I test with set dshot_bidir = OFF and set cpu_overclock = 120MHZ. Then circle back on the max7456_clock and other max7456 settings. 

Thank you @AndreaCCIE

sasoer, 2022-01-08:
I also have a problem with slow/laggy OSD on couple of F411 boards (matek and betafpv aio). I tested a bunch of builds. Last build everything works smoothly is #2569. Starting with build #2570 and up I have more or less same laggy OSD. This only happens woth BiDirectional DShoot enabled.

I didn't find any newer build that would work as expected.

SteveCEvans, 2022-01-08:
Working fine on my F411 boards. Reducing `rpm_filter_harmonics` and `dterm_notch_hz` will help if you're still seeing lag.

sasoer, 2022-01-08:
I'm not using rpm filter at all. Only reason I have bidirectional dshot enabled is dynamic idle. What does have dterm_notch_hz with anything? Issue started with build #2570 so It's not a config issue.

AndreaCCIE, 2022-01-08:
> Working fine on my F411 boards. Reducing `rpm_filter_harmonics` and `dterm_notch_hz` will help if you're still seeing lag.

Is this a cpu related problem?

SteveCEvans, 2022-01-08:
The OSD driver used to trample all over the timing of the flight code, negatively impacting flight performance. Now the OSD code is well behaved and the flight code takes precedent as it should. If you have an OSD display setup with many elements which update frequently then the OSD update rate will automatically drop so as not to impact flight performance. You will get better OSD performance if DMA is enabled, which on most F411 FCs will require you to set `dshot_bitbang=ON`. That prevents gyro DMA from working on an F4 if the gyro is on SPI bus 1, but it’s not a huge flight performance impact.

SteveCEvans, 2022-01-08:
@AndreaCCIE I see from your timer assignments that you don’t have `dshot_bitbang=ON`. The OSD crawling update in your video is due to the transfer from a buffer in RAM to the MAX7456 device running too slow when done using polled rather than DMA transfer. You’ll see from the `dma show` output that you don’t have streams assigned for SPI_MOSI 2 and SPI_MISO 2 (or 3 depending on which bus the OSD is on), so polled mode is being used. Setting `dshot_bitbang=ON` will free up the MOTOR resources on DMA1 and then DMA to the MAX7456 will be possible. Using 120MHz over clock speed up the polled operation enough to avoid such issues typically, but enabling DMA will definitely fix it.

AndreaCCIE, 2022-01-08:
Thanks @SteveCEvans. I'm not sure why this doesn't happen on the other F411 I'm flying, but regardless: shouldn't this be a default setting for these FCs?

SteveCEvans, 2022-01-08:
@AndreaCCIE Did `dshot_bitbang=ON` work for you?

Ideally we want the best flight performance, and use of gyro DMA is best for that as it will give lower jitter and thus better filter performance for which timer based DSHOT is best. Many F411 FCs, for example those based on the MATEKF411 have well thought out motor pin assignments allowing the use of timer based DSHOT, and taking the FC in my BetaFPV75X as an example, there are other resource conflicts that mean that timer based DSHOT is preferred over bit banged. Specifically the choice of LED_STRIP pin requires the use of timer TIM1, which would also be required for bit banged DSHOT, so if I were to choose bit banged DSHOT (by making it the default) people would complain that their LEDs no longer work. The current defaults offer the best compromises here.

These problems stem from an F4 errata which means that we get port corruption if we use SPI DMA using DMA2 (which is used for SPI bus 1, most commonly used for the gyro) at the same time as DMAs to a port as used by bit-banged DSHOT. The consequence of this is that we have to use timer based DSHOT if we want to use DMA on the gyro's SPI bus. That in turn requires 4 DMA streams for MOTOR 1 to MOTOR 4 as per your config below.

```
Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 1
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 2
DMA1 Stream 4: MOTOR 3
DMA1 Stream 5: MOTOR 4
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

By comparison (when all motors are on the same port) bit banged DSHOT only requires a single DMA stream as shown below, you can see that streams 3 and 4 (used for OSD DMA) can't be allocated above because of a conflict with the MOTOR allocations.

```
# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

It's all about compromises on the F411, but fortunately F7/H7/G4 processors are less constrained as this errata doesn't apply to them.

In your case however there were still issues, although it's not clear if these were related to the FC architecture or some other aspect of your configuration and bit banged DSHOT will improve OSD performance for you.

AndreaCCIE, 2022-01-08:
@SteveCEvans yes, it works fine now. But to understand why I have to read your post a few times 🥇

closed 2022-01-08 by SteveCEvans