betaflight/betaflight #11246: F411 Frsky: No stable RC connection since #2604 (on Mobula6, Crazybee F4FR)

MLutzAT, 2022-01-08:
### Describe the bug
I update my Mobula6 since a few weeks to almost every development build. Since #2604 my RC connection is no more stable, it comes and goes and sometimes (!) after about 2mins a stable connection will establish. I flashed all builds since #2604, and the problem stays the same. If I go back to below #2604 all is fine again.
Of course I tried to rebind etc, also to change the protocol from D16 to D8, nothing changes...

### To Reproduce
Flash build >=#2604 on HAMO-CRAZYBEEF4FR, and try to get stable Frsky D8 or D16 connection

### Expected behavior
stable link as before #2604

### Flight controller configuration

```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jan  8 2022 / 01:08:20 (b8c58abf0) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FR
manufacturer_id HAMO
mcu_id 003a001e3437510a32363736
signature 

# name: Mobula 6-21

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature -ANTI_GRAVITY
feature TELEMETRY

# serial
serial 1 2048 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -ARMED
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 2 2 1300 2100 0 0
aux 3 4 3 1700 2100 0 0
aux 4 13 3 1300 1700 0 0
aux 5 19 1 1700 2100 0 0
aux 6 28 2 1300 2100 0 0
aux 7 30 2 1700 2100 0 0
aux 8 35 3 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 350

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_max_hz = 900
set acc_trim_roll = 3
set acc_calibration = 33,10,-17,1
set baro_hardware = NONE
set fpv_mix_degrees = 20
set rx_spi_protocol = FRSKY_D
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set bat_capacity = 300
set vbat_max_cell_voltage = 435
set vbat_min_cell_voltage = 325
set vbat_warning_cell_voltage = 335
set ibata_scale = 875
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_visual_beeper = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_cap_alarm = 300
set osd_vbat_pos = 2433
set osd_rssi_pos = 2306
set osd_link_quality_pos = 225
set osd_rssi_dbm_pos = 257
set osd_tim_1_pos = 2455
set osd_tim_2_pos = 2423
set osd_remaining_time_estimate_pos = 2220
set osd_flymode_pos = 2393
set osd_anti_gravity_pos = 2103
set osd_g_force_pos = 2135
set osd_throttle_pos = 2327
set osd_vtx_channel_pos = 2059
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2112
set osd_mah_drawn_pos = 2370
set osd_motor_diag_pos = 2295
set osd_craft_name_pos = 2441
set osd_home_dist_pos = 98
set osd_flight_dist_pos = 130
set osd_compass_bar_pos = 2090
set osd_power_pos = 2145
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 289
set osd_pit_ang_pos = 2241
set osd_rol_ang_pos = 2155
set osd_battery_usage_pos = 35171
set osd_disarmed_pos = 2346
set osd_nheading_pos = 2125
set osd_esc_tmp_pos = 2105
set osd_esc_rpm_pos = 2167
set osd_rtc_date_time_pos = 6
set osd_flip_arrow_pos = 2286
set osd_core_temp_pos = 2081
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_rpm = ON
set cpu_overclock = 120MHZ
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5865
set frsky_spi_tx_id = 190,58,1
set frsky_spi_offset = 30
set frsky_spi_bind_hop_data = 0,30,60,91,120,150,180,210,5,35,65,95,125,155,185,215,10,40,70,100,130,160,190,221,15,45,75,105,135,165,195,225,20,50,80,110,140,170,200,230,25,55,85,115,145,175,205,0,0,0
set gyro_1_align_yaw = 900
set name = Mobula 6-21

profile 0

# profile 0
set dterm_lpf1_static_hz = 0
set iterm_relax_cutoff = 25
set p_pitch = 51
set i_pitch = 92
set p_roll = 49
set i_roll = 88
set p_yaw = 49
set i_yaw = 88
set level_limit = 45
set auto_profile_cell_count = 1
set thrust_linear = 20
set feedforward_averaging = 3_POINT
set feedforward_jitter_factor = 12
set dyn_idle_min_rpm = 35
set simplified_pi_gain = 110

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_expo = 50
set roll_rc_rate = 5
set pitch_rc_rate = 5
set roll_expo = 25
set pitch_expo = 30
set roll_srate = 50
set pitch_srate = 35

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 - Flight controller: CRAZYBEEF4FR
 - Other components: stock Mobula6
 - How are the different components wired up: stock Mobula6

I flash and fly since a few years, but this is my first bug report. Sorry if not perfect. I did everything to rule out that the problem is on my side. I guess the SPI changes introduced in #2604 are the cause.

AndreaCCIE, 2022-01-08:
Duplicate of #11125?

MLutzAT, 2022-01-08:
Sorry sorry, although I searched... I searched not good enough...
But I found the "cause" now: The problems appear only when I enable bidirectional DSHOT...

I flashed build 2607 again, and started from scratch. When the install is "clean", D8 works after a rebind (or cli pasting the spi bind parameters). But: After enabling bidirectional DSHOT the connection gets unstable. After disabling the bidirectional DSHOT, all is fine again!

So, i conclude that the changes affecting DSHOT introduced in build 2604 are related to our problem!!
Before 2604 I used bidirectional DSHOT without encountering any problems (as far as I could "see or detect" them)...

MLutzAT, 2022-01-08:
added this info as an edit to the first post:
_edit: I now found the issue: The problem appears only, when bidirectional DSHOT is activated. Before build 2604 I used it for months, and even flashed the proper ESC telemetry firmwares (JESC) to make use of the RPM filtering...
Today I updated from a build from Dec28 to the actual one, and the problems started to appear. After flashing every build I could nail the problem down to "since 2604", and now I discovered that when I do not use bidirectional DSHOT all is fine. If I activate it the RC gets unstable. If I go back to build 2603 again everything is fine...

So I think that since 2604 there is a problem with bidirectional DSHOT and RC._

AndreaCCIE, 2022-01-08:
@MLutzAT no need to sorry, mine was just a question!

turboed13b, 2022-01-08:
Try using dshot300/2k. Might be overloading cpu like other spi targets.

MLutzAT, 2022-01-09:
> @MLutzAT no need to sorry, mine was just a question!

Thank you - I do not want to cause any unnecessary work or attention...
I usually figure most problems out for myself, in this case I thought I should share my thoughts, maybe this is useful for somebody or helps to solve problems (if they exist, and are no "user error")...

@turboed13b Thanks for your suggestion - my CPU utilization is below 50%. When I switch to 2K it goes around 35%, and link stability does not seem to improve. 
CPU performance might to have something to do with it, I've seen it in #11229 and can confirm that set the CPU clock to 120MHz seem to improve the situation sometimes (but NOT reproducible), as it "feels" that with 108MHz the situation is even worser. 
But as I said, going back to a build before 2604 solves all my problems. I can use bidirectional DSHOT600/4k, RPM filtering, and my Mob6 flies like a charm... Also the OSD works flawlessly with proper update rate. I will watch the situation closely, and I'm sure that these SPI problems will be addressed and sorted out soon... Because as it seems there are a few problems related to SPI as root cause...

If I can help or contribute in any way (more then buying your merch etc. and propagating/teaching BF to my friends) please let me know. I use BF now for over 2 years, and would be more then happy if I could do some "good"!

Best regards, Markus

SteveCEvans, 2022-01-09:
Try `dshot_bitbang=ON`.

MLutzAT, 2022-01-09:
@SteveCEvans Thank you!! This makes it "usable" with 2K PID loop, as the CPU load gets now below 30%. With bitbang off the CPU load sits around 37%, with 4K around 50%, and the SPI RC connection goes on and off...
I'm used to that around 50% CPU should be no problem, but this seems no longer to be the case. At least until this gets probably fixed?

What would you recommend?
Use build below 2604 until this gets sorted out, or use the actual ones (2610 in my case now) with lower PID loop freq., overclock to 120 and bitbang on?

Sorry for the questions, I would understand if you do not have the time to answer.

turboed13b, 2022-01-09:
> @SteveCEvans Thank you!! This makes it "usable" with 2K PID loop, as the CPU load gets now below 30%. With bitbang off the CPU load sits around 37%, with 4K around 50%, and the SPI RC connection [...]

Anything below 4.3 should be working fine with spi d8.  I run 3.4.1 personally it just flies the best. Do not overclock on old builds. I dont recommend 120hz either on 4.3 I have ran into stability/twitching issues.

SteveCEvans, 2022-01-10:
Can you please try the image below.

[betaflight_4.3.0_STM32F411_1725766.hex.zip](https://github.com/betaflight/betaflight/files/7836381/betaflight_4.3.0_STM32F411_1725766.hex.zip)

limonspb, 2022-01-10:
> Can you please try the image below.
> 
> [betaflight_4.3.0_STM32F411_1725766.hex.zip](https://github.com/betaflight/betaflight/files/7836381/betaflight_4.3.0_STM32F411_1725766.hex.zip)

This hex fixes similar problem on my whoop: running SPI Redpine protocol 8k/4/dshot300/bidir=on, F411 (BETAFPVF411RX)
Issue: rc commands are being update only once per second (approximately) since the build 2604 (2603 - all good)

MLutzAT, 2022-01-10:
> Can you please try the image below.
> 
> [betaflight_4.3.0_STM32F411_1725766.hex.zip](https://github.com/betaflight/betaflight/files/7836381/betaflight_4.3.0_STM32F411_1725766.hex.zip)

Hello Steve, thank you very much for your attention!

Seems your image solves the problem also for me! I can use bidirectional DSHOT and 4K PID loop again with around 50% CPU. And the most important thing: SPI Frsky RC link is stable again!

Just because I'm curious: What did you change in your build? 

Again, thank you!

SteveCEvans, 2022-01-10:
@MLutzAT Please see the explanation in https://github.com/betaflight/betaflight/pull/11269

closed 2022-01-11 by haslinghuis

mitcht, 2022-04-30:
I'm using the same hardware, just built a hex from master, and i'm getting dropouts at 20 feet still.