betaflight/betaflight #11250: board not support unified target OMNIBUSF7NANOV8

link-gamer, 2022-01-08:
hello good day to everyone who is reading.
i am interested to modify my flight control to use and test betaflight 4.3.0 but unfortunately i required the config file for my board or i must need to find a generated firmware previously done, i try the unified version for the  target OMNIBUSF7NANOV7 but is not working and i must modify manually values to load Gyros and UART's
its there possibility to add this board to the existents to get the firmware or al test locate correctly the config for this one and be capable to compile correctly a version for myself

actual version 
# version
# Betaflight / OMNIBUSF7NANOV8 (O7V8) 4.2.0 Mar 31 2020 / 09:02:54 (5f8fc1e6f) MSP API: 1.43

board_name OMNIBUSF7NANOV8
manufacturer_id 

testing version with unified target 
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.11 Nov  9 2021 / 20:29:32 (948ba6339) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF7NANOV7, version: 14929e39, date: 2020-08-31T10:05:16Z

# start the command batch
batch start

board_name OMNIBUSF7NANOV7
manufacturer_id AIRB

Asizon, 2022-01-08:
@link-gamer this is a sample test board from Airbot? 
@x4FF3 do you now something about this nano V8 board?

link-gamer, 2022-01-08:
> @link-gamer this is a sample test board from Airbot? @x4FF3 do you now something about this nano V8 board?

until as i know, yes it is  from Airbot, i am requiring from the supplier some more information about the factory and manufacture, I live in Thailand and this supposed to be "original" and was installed betaflight version 4.2.0 from factory 
[image]
http://www.fpvonly.com/product/3553/new-original-airbot-ominibus-f7-nano-v8-usb-c-flight-control-%E0%B8%9A%E0%B8%AD%E0%B8%A3%E0%B9%8C%E0%B8%94%E0%B8%84%E0%B8%A7%E0%B8%9A%E0%B8%84%E0%B8%B8%E0%B8%A1-fpv-racing-drone

it have the STM32F722 MCU, the description in the seller page its wrong in several technical points that i want to let them know 

[image]
[image]

x4FF3, 2022-01-20:
Hi,

i checked back with airbot and it shouldnt have gone on public sale...i made a preliminary target for this board in the past, for testing the board. but yeah...if i find time for this ill make a target for it.

closed 2022-02-27 by github-actions[bot]

Yanghuoling, 2025-01-04:
Is there a problem with osd? What about this flight control information? Do you have wiring and parameters?