betaflight/betaflight #11258: No Stable RC Link on Mobula7 with CRAZYBEEF4FS

saimat, 2022-01-09:
### Describe the bug
After booting the FC with BTFL 4.3 RC2 the RC connection is not stable, it comes and goes. No arming possible (BADRX).
After some minutes a stable connection will establish and is then stable until the end of the flight.
When flashing back to BTFL 4.2.11 the problem does not occur.

### To Reproduce
Flash BTFL 4.3 RC1 or RC2 on CRAZYBEEF4FS

### Expected behavior
Stable link as in BTFL 4.2

### Flight controller configuration

```
# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jan  8 2022 / 01:08:20 (b8c58abf0) MSP API: 1.44
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FS, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FS
manufacturer_id HAMO
mcu_id 005a00323437511535313736
signature 

# name: Mobula7

# feature
feature -AIRMODE
feature MOTOR_STOP

# serial
serial 0 2048 115200 57600 0 115200

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -ARMING_GPS_NO_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1800 2100 0 0
aux 3 13 3 1525 1725 0 0
aux 4 28 4 1300 1700 0 0
aux 5 35 2 1800 2100 0 0

# adjrange
adjrange 0 0 5 900 2100 12 5 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 UNKNOWN  U CUSTOM     0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860    0
vtxtable band 5 RACEBAND R CUSTOM     0    0 5732 5769 5806 5843    0    0
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 1
vtxtable powerlabels 25

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_trim_roll = 8
set acc_calibration = 28,0,40,1
set mag_hardware = NONE
set baro_hardware = NONE
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 330
set ibata_scale = 1175
set yaw_motors_reversed = ON
set small_angle = 180
set osd_vbat_pos = 2369
set osd_rssi_pos = 2081
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2457
set osd_throttle_pos = 2425
set osd_current_pos = 2401
set osd_mah_drawn_pos = 2433
set osd_craft_name_pos = 2443
set osd_avg_cell_voltage_pos = 2135
set osd_flip_arrow_pos = 2282
set osd_rate_profile_name_pos = 2113
set osd_stat_max_fft = ON
set rate_6pos_switch = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_freq = 5806
set gyro_1_align_yaw = 900
set flysky_spi_tx_id = 949835362
set flysky_spi_rf_channels = 21,112,74,159,4,107,58,146,41,101,47,131,33,93,63,124
set name = Mobula7

profile 0

# profile 0
set iterm_relax_cutoff = 10
set p_pitch = 70
set i_pitch = 126
set d_pitch = 71
set f_pitch = 187
set p_roll = 67
set i_roll = 120
set d_roll = 62
set f_roll = 180
set p_yaw = 67
set i_yaw = 120
set f_yaw = 180
set angle_level_strength = 55
set horizon_level_strength = 55
set horizon_transition = 100
set level_limit = 85
set d_min_roll = 62
set d_min_pitch = 71
set simplified_master_multiplier = 150
set simplified_d_gain = 140
set simplified_dmax_gain = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rateprofile_name = ACTUAL85
set throttle_limit_type = SCALE
set throttle_limit_percent = 85

rateprofile 1

# rateprofile 1
set rateprofile_name = ACTUAL

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: RX_SPI_EXTI
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Setup / Versions
 
 - Flight controller: CRAZYBEEF4FS
 
 - Other components: Stock Mobula7
 
 - How are the different components wired up: Stock Mobula7

swilkens, 2022-01-09:
What PID frequency are you using? It's recommended to go to 4 Khz on F4. Even then, the [4.3 tuning notes](https://github.com/betaflight/betaflight/wiki/4.3-Tuning-Notes) suggest overclocking the F4 MCU to 108 or 120 Mhz.

What is the MCU load during these first few minutes? Maybe a `status` in `cli` would be usefull here.

saimat, 2022-01-09:
I'm using 4 kHz PID loop frequency, CPU Load is at 58%. 

Right after boot:
```
# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.28V, Core temp=35degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3978, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000
OSD: MAX7456
System Uptime: 45 seconds, Current Time: 2022-01-09T13:22:30.648+00:00
CPU:58%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP
```

After 2 min when the link is established:
```
# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.28V, Core temp=38degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3978, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000
OSD: MAX7456
System Uptime: 196 seconds, Current Time: 2022-01-09T13:25:01.210+00:00
CPU:61%, cycle time: 126, GYRO rate: 7936, RX rate: 259, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: CLI MSP
```

saimat, 2022-01-09:
Overclocking to 120Mhz reduces the load to 52% but does not solve the problem.

````
# status
MCU F411 Clock=120MHz (PLLP-HSE), Vref=3.28V, Core temp=39degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3978, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000
OSD: MAX7456
System Uptime: 27 seconds, Current Time: 2022-01-09T13:45:18.181+00:00
CPU:52%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP
````

SteveCEvans, 2022-01-10:
Can you please try the image below.
[betaflight_4.3.0_STM32F411_1725766.hex.zip](https://github.com/betaflight/betaflight/files/7836382/betaflight_4.3.0_STM32F411_1725766.hex.zip)

MLutzAT, 2022-01-10:
Hello @saimat !
I guess you have the same issue as we have/had here: #11246 

The new image from Steve solved the problem for me and @limonspb though, so I guess this will be soon sorted out! :)
All kudos to Steve!! :)

saimat, 2022-01-10:
Hi @SteveCEvans thank you, this image solves my problem. RC link is established and stable directly after FC boot.

closed 2022-01-11 by haslinghuis