betaflight/betaflight #11260: CAMERA CONTROL on BF 4.3, difficult to enter the menu, or select the right line in the OSD.

corrado-c, 2022-01-09:
### Describe the bug
after upgrading 3 quads to BF4.3, I found that CAMERA CONTROL works bad or not working (with BF4.2 is ok). Trying to change camera_control_key_delay = 100 there were improvements. When I enter the menu, it sometimes goes directly to the next submenu. the menu goes up with a high stick but does not go down with a low stick. Sometimes with the stick in the center it scrolls up for a few seconds. I will post the settings of one of the 3 quads .
I hope I compile everything correctly :)

### To Reproduce
simply try to change the camera settings from the remote control.

### Expected behavior
correct and simple operation of camera control from OSD via radio control.

### Flight controller configuration

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan  9 2022 / 11:51:40 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: FOXE, board_name: FOXEERF722V2, version: baa1d34a, date: 2021-08-16T07:35:36Z

# start the command batch
batch start

board_name FOXEERF722V2
manufacturer_id FOXE

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# serial
serial 1 64 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 2 3 1700 2100 0 0
aux 2 13 2 1700 2100 0 0
aux 3 26 2 900 1300 0 0
aux 4 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# master
set gyro_notch1_hz = 260
set gyro_notch1_cutoff = 220
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_min_hz = 120
set dyn_notch_max_hz = 400
set acc_calibration = 28,107,-307,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1000
set max_check = 2000
set fpv_mix_degrees = 20
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_idle_value = 450
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set bat_capacity = 1300
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_alt_alarm = 50
set osd_tim1 = 1280
set osd_tim2 = 1793
set osd_vbat_pos = 2051
set osd_link_quality_pos = 2104
set osd_link_tx_power_pos = 2420
set osd_rssi_dbm_pos = 2072
set osd_tim_2_pos = 2091
set osd_flymode_pos = 2452
set osd_anti_gravity_pos = 2253
set osd_g_force_pos = 2478
set osd_throttle_pos = 2272
set osd_vtx_channel_pos = 2484
set osd_current_pos = 2058
set osd_mah_drawn_pos = 2098
set osd_power_pos = 2066
set osd_pidrate_profile_pos = 2458
set osd_avg_cell_voltage_pos = 2083
set osd_esc_tmp_pos = 2471
set osd_flip_arrow_pos = 2221
set osd_core_temp_pos = 2465
set osd_stat_max_spd = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5806
set vcd_video_system = PAL
set camera_control_key_delay = 110
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 35
set thr_expo = 20
set rates_type = BETAFLIGHT
set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 110
set roll_expo = 9
set pitch_expo = 9
set roll_srate = 76
set pitch_srate = 76
set yaw_srate = 73
set tpa_breakpoint = 1650
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

# end the command batch
batch end

# 
```
### Resource

```
resource
resource BEEPER 1 A04
resource MOTOR 1 A09
resource MOTOR 2 A08
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 C06
resource MOTOR 6 C07
resource PPM 1 B07
resource LED_STRIP 1 A15
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C15
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 B03
resource ADC_BATT 1 C00
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C02
resource FLASH_CS 1 B12
resource OSD_CS 1 C03
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 B02

# 
```

### Setup / Versions

 - Flight controller: FOXEERF722V2
 - RX: R9MM ELRS
 - CAMERA: FOXEER FALKOR
 - 

<!-- Add any other context about the problem that you think might be relevant here. -->

KarateBrot, 2022-01-09:
Try BF 4.3 RC2 and see if the problem still persists.

corrado-c, 2022-01-09:
> Try BF 4.3 RC2 and see if the problem still persists.
I tried now, the same

KarateBrot, 2022-01-09:
@SteveCEvans Do we need a PR to make cam control work? Some `schedulerIgnoreTask` stuff?

SteveCEvans, 2022-01-09:
@corrado-c Can you please provide `timer` and `timer show` outputs. The only change here has been to the priority of the `CAMCTRL` task. The image below bumps this up from `TASK_PRIORITY_LOWEST` to `TASK_PRIORITY_LOW`. Could you please give that a try
[betaflight_4.3.0_STM32F7X2_norevision.hex.zip](https://github.com/betaflight/betaflight/files/7835889/betaflight_4.3.0_STM32F7X2_norevision.hex.zip)
.

corrado-c, 2022-01-10:
> @corrado-c Can you please provide `timer` and `timer show` outputs. The only change here has been to the priority of the `CAMCTRL` task. The image below bumps this up from `TASK_PRIORITY_LOWEST` to [...]

here I am! I was able to put my hand only now.
if I am able, I am happy to do all the tests I can

Below are the results of TIMER AND TIMER SHOW (sorry but I don't understand why it looks like this)

I tried the firmware but it doesn't seem to change anything, same behavior

timer
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer C07 AF3
# pin C07: TIM8 CH2 (AF3)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)

# 

TIMER SHOW

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

# 

[timer and timer show.txt](https://github.com/betaflight/betaflight/files/7841133/timer.and.timer.show.txt)

corrado-c, 2022-01-11:
why closed?

KarateBrot, 2022-01-11:
It was auto closed because the linked PR got closed, I reckon. Reopening as it seems the problem was not solved.

SteveCEvans, 2022-01-11:
@corrado-c Can you please check if this works using defaults plus receiver settings only. I note that your config includes `set camera_control_key_delay = 110`. Please test the fix again with defaults.

corrado-c, 2022-01-11:
Sure I have tried several times differently, using defaults plus receiver settings only, using backup, using full chip erase, and I have also tried different values of camera_control_key_delay 100 to 200. 
to be safe I repeated the flash now, Unfortunately same situation.

`diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan  9 2022 / 20:02:51 (norevision) MSP API: 1.44
# config: manufacturer_id: FOXE, board_name: FOXEERF722V2, version: baa1d34a, date: 2021-08-16T07:35:36Z

# start the command batch
batch start

board_name FOXEERF722V2
manufacturer_id FOXE

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# serial
serial 1 64 115200 57600 0 115200

# master
set serialrx_provider = CRSF
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DISABLED
set motor_pwm_rate = 480
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end

# `

corrado-c, 2022-03-05:
hi, but can I do some other tests?

haslinghuis, 2022-04-02:
@corrado-c please check with latest development firmware available. (Don't forget to update configurator nightly first)

KarateBrot, 2022-04-03:
> @corrado-c
hi, but can I do some other tests?

In the new Betaflight Discord Server, channel `#roles`, you can subscribe as a `@Testers` to get news when something needs testing.

haslinghuis, 2022-04-15:
Please test with RC5

haslinghuis, 2024-09-10:
Stalled.

closed 2024-09-10 by haslinghuis