betaflight/betaflight #11262: Crossfire failsafes

dnielsen4, 2022-01-09:
### Describe the bug
When flashing BTFL 4.3 RC1 or RC2 onto the DarwinFPVF722HD, crossfire LQ and SNR on the OSD occasionally drops to zero and -30 for a split second and then immediately returns. This occurs consistently around every 7-8 seconds when running in 150 Hz mode, and 22-24ish seconds when running in 50 Hz mode. This does not occur when flashing back to BTFL 4.2.11.

### To Reproduce
Flash BTFL 4.3 RC1 or RC2 on DarwinFPVF722HD

### Expected behavior
Stable link

### Flight controller configuration
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan  9 2022 / 11:51:40 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: DAFP, board_name: DARWINF722HD, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name DARWINF722HD
manufacturer_id DAFP
mcu_id 003a00393538510835343631
signature 

# feature
feature -AIRMODE
feature GPS
feature LED_STRIP

# serial
serial 0 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# map
map TAER1234

# led
led 0 0,0::AT:13
led 1 1,0::AT:13
led 2 2,0::AT:13
led 3 3,0::AT:13

# mode_color
mode_color 6 0 2
mode_color 6 1 2
mode_color 7 0 7

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 46 2 1700 2100 0 0
aux 3 13 1 1100 1150 0 0
aux 4 13 1 1475 1550 0 0
aux 5 13 1 1850 1900 0 0
aux 6 15 3 900 1025 0 0
aux 7 26 0 1700 2100 0 0
aux 8 28 1 1300 2100 0 0
aux 9 35 1 2000 2050 0 0
aux 10 35 1 1600 1650 0 0
aux 11 35 1 1200 1250 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 999

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 250
set dyn_notch_count = 2
set dyn_notch_q = 400
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 250
set acc_calibration = -132,-16,-289,1
set rc_smoothing_setpoint_cutoff = 15
set rc_smoothing_feedforward_cutoff = 15
set serialrx_provider = CRSF
set crsf_use_rx_snr = ON
set dshot_idle_value = 700
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_auto_baud = ON
set gps_set_home_point_once = ON
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1500
set gps_rescue_throttle_max = 1750
set gps_rescue_throttle_hover = 1350
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 50
set ledstrip_race_color = WHITE
set ledstrip_beacon_period_ms = 50
set ledstrip_beacon_percent = 100
set osd_units = IMPERIAL
set osd_warn_link_quality = ON
set osd_tim1 = 2563
set osd_vbat_pos = 33
set osd_link_quality_pos = 2081
set osd_link_tx_power_pos = 65
set osd_rssi_dbm_pos = 2113
set osd_tim_1_pos = 2423
set osd_current_pos = 2411
set osd_mah_drawn_pos = 2401
set osd_gps_speed_pos = 2241
set osd_gps_lon_pos = 49552
set osd_gps_lat_pos = 51591
set osd_gps_sats_pos = 2093
set osd_home_dir_pos = 2129
set osd_home_dist_pos = 2124
set osd_flight_dist_pos = 2135
set osd_altitude_pos = 18648
set osd_avg_cell_voltage_pos = 2103
set osd_efficiency_pos = 2273
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set osd_stat_min_rssi_dbm = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set vcd_video_system = NTSC
set rpm_filter_q = 450

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 63
set dterm_lpf1_dyn_max_hz = 127
set dterm_lpf1_static_hz = 127
set dterm_lpf2_static_hz = 127
set yaw_lowpass_hz = 75
set thrust_linear = 25
set feedforward_smooth_factor = 0
set feedforward_jitter_factor = 15
set dyn_idle_min_rpm = 15
set simplified_dterm_filter_multiplier = 85

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
```
```
# 
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: PREINIT 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: LED_STRIP
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 - Flight controller: FPVCYCLE DARWINFPV WHOOP STACK FLIGHT CONTROLLER (from FPVCycle)
 - RX: CRSF nano RX and CRSF nano diversity RX (on version 6.13)
 - TX: CRSF Micro V2
 - How are the different components wired up:
 - RX attached to UART 2 on the FC, also tried UART 4 with same issue

limonspb, 2022-01-10:
Does it only drops LQ or it also does a real failsafe that you can feel/see?

dnielsen4, 2022-01-10:
> Does it only drops LQ or it also does a real micro failsafe that you can feel/see?

I just took a blackbox log (while it popped up LQ twice on the OSD) but can't find any evidence of a real micro failsafe. It seems like it may just be an OSD thing.

Zuldan, 2022-01-18:
@dnielsen4 try the latest nightly

closed 2022-02-26 by github-actions[bot]

klutvott123, 2022-03-07:
@dnielsen Do you have both tx and rx wired to the flight controller? https://intofpv.com/t-psa-tbs-crossfire-tracer-rx-wiring-always-connect-the-rx-ch2-telemetry-wire