betaflight/betaflight #11267: HGLRC F722 does not show Barometer on 4.3 RC2 even after enabled in configurator

shyam4learn, 2022-01-10:
### Describe the bug

On installing Betaflight 4.3 RC2 (or RC1 for that matter), HGLRC F722 board does not recognize the Barometer even after it is enabled.

### To Reproduce

1. Flash Betaflight 4.3 RC2 (or RC1) on HGLRC F722 board
2. Enable Barometer in Configuration page

### Expected behavior

Betaflight configurator should recognize the Barometer and show it enabled on the top right section.

### Flight controller configuration

```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan  9 2022 / 11:51:40 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 836acaf9, date: 2021-10-31T19:48:38Z

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature OSD
feature ESC_SENSOR

# serial
serial 1 2 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 900 1700 0 0
aux 1 1 0 1025 1175 0 0
aux 2 46 0 1225 1375 0 0
aux 3 13 0 1725 1850 0 0
aux 4 31 0 1875 1975 0 0
aux 5 35 0 1425 1600 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# master
set gyro_lpf1_static_hz = 500
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 2
set dyn_notch_q = 500
set dyn_notch_min_hz = 120
set gyro_lpf1_dyn_min_hz = 500
set gyro_lpf1_dyn_max_hz = 1000
set acc_calibration = 12,13,-43,1
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set rc_smoothing_auto_factor = 40
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_idle_value = 450
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set failsafe_delay = 7
set failsafe_procedure = GPS-RESCUE
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_set_home_point_once = ON
set gps_rescue_min_sats = 7
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 200
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 1100
set osd_vbat_pos = 6552
set osd_rssi_pos = 2477
set osd_link_quality_pos = 430
set osd_tim_2_pos = 6604
set osd_throttle_pos = 2209
set osd_current_pos = 6583
set osd_mah_drawn_pos = 6561
set osd_craft_name_pos = 6369
set osd_gps_lon_pos = 6160
set osd_gps_lat_pos = 6145
set osd_gps_sats_pos = 6232
set osd_home_dir_pos = 6221
set osd_home_dist_pos = 6223
set osd_altitude_pos = 6210
set osd_warnings_pos = 2369
set osd_avg_cell_voltage_pos = 6529
set debug_mode = GYRO_SCALED
set rpm_filter_fade_range_hz = 100

profile 0

# profile 0
set dterm_lpf1_dyn_expo = 10
set vbat_sag_compensation = 100
set p_pitch = 59
set i_pitch = 106
set d_pitch = 58
set f_pitch = 158
set p_roll = 51
set i_roll = 92
set d_roll = 53
set f_roll = 138
set d_min_roll = 43
set d_min_pitch = 46
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 10
set dyn_idle_min_rpm = 30
set dyn_idle_p_gain = 35
set simplified_pids_mode = RP
set simplified_d_gain = 145
set simplified_pi_gain = 115
set simplified_dmax_gain = 70
set simplified_feedforward_gain = 115
set simplified_pitch_d_gain = 95
set simplified_pitch_pi_gain = 110

rateprofile 0

# rateprofile 0
set thr_mid = 37
set thr_expo = 35
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 58
set pitch_srate = 58
set yaw_srate = 50
set throttle_limit_type = SCALE
set throttle_limit_percent = 90

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller:  HGLRC F722 20x20 FC, Bought directly from HGLRC
 
 - Other components: Crossfire nano receiver, DJI Caddx Vista, TBS M.8 GPS
 
 - How are the different components wired up: Crossfire RX connected to UART1, Caddx Vista connected to UART4, GPS connected to UART2

The Barometer functionality worked without issues on all version of Betaflight 4.2 with the same component and wiring setup as described above.

alienwaredream, 2022-01-11:
flash BF4.3RC2 and CLI
“
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
save
”

alienwaredream, 2022-01-11:
I think there is a little mistake in HGLRC F722 unified-targets

mooiweertje, 2022-11-02:
Why is it not working out of the box? How is someone to know to set baro to spi in CLI?!??!