betaflight/betaflight #11268: 4.3.0RC2 Motors Tab "Mixer Mode Problem Detected" BF thinks motors not tied to resources, when they are. FC is BetaFPVF722 UPDATE: when LED_STRIP enabled, I get the motors tab error.

recyclerealestate, 2022-01-10:
### Describe the bug
-----------------------UPDATE: 
As soon as I enable LED_STRIP in the configuration tab, then I get the error in the motors tab.  If I disable the LED_STRIP, then am back to good.  So the bug is actually that there is some sort of conflict detecting the motor resources when the LED_STRIP is enable.
------------------------ENDUPDATE
when I go to the motors tab it gives me an error "Mixer Mode Problem Detected" Quad X requires 4 motor resources and has 0 ....
When I go to CLI and type "resources show all" I can see all 4 motors are assigned.

### To Reproduce
-------------UPDATE:
1) Freshly upload new firmware 4.3 rc2
2) accept defaults;
3) enable LED_STRIP on configuration tab; reboot
4) go to motors tab; set ESC/motor protocol: "DSHOT300", motor direction: "reversed"
And if you  disable the LED_STRIP and reboot the motors tab will work fine.
-------------END UPDATE:
Go to motors tab;  Since others obviously have this working I wonder if it is something specific to the flight controller, the betaFPV board, Board: BEFH/BETAFPVF722(STM32F7X2), version: 0
-- i did just flash the new 4.3.0RC2 onto this controller
 -- then I set VTX using the presets
  -- note RPM sensing doesn't work on this board...
  -- had set motors to just DSHOT300  WITHOUT bidirectional, because RPM sensing isn't possible on this board (as far as I know).

### Expected behavior
be able to test motor configuration and have a quad X working fine

### Setup

- Configurator version:  10.8.0 (c9b7ed8f)  

Below is the diff all and the resource list -- in the resource list you can see 4 motors are defined.
```
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan  9 2022 / 11:51:40 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: BEFH, board_name: BETAFPVF722, version: 150f0b6f, date: 2020-08-16T02:29:24Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFPVF722
manufacturer_id BEFH
mcu_id 00170012345650042030324b
signature 

# name: Diablo2

# feature
feature MOTOR_STOP
feature GPS
feature LED_STRIP

# serial
serial 0 2 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 36
vtxtable powerlabels 25 100 400 MAX

# master
set acc_calibration = -86,30,-90,1
set serialrx_provider = CRSF
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set gps_auto_baud = ON
set gps_set_home_point_once = ON
set report_cell_voltage = ON
set vtx_band = 3
set vtx_channel = 6
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5905
set name = Diablo2

profile 0

# profile 0
set feedforward_averaging = 3_POINT
set feedforward_smooth_factor = 70
set feedforward_jitter_factor = 8

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 70

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# resource MOTOR
###ERROR IN resource: INDEX NOT BETWEEN 1 AND 8###

# resource MOTOR 1
###ERROR IN resource: PARSING FAILED###

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 4
A01: MOTOR 3
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: CAMERA_CONTROL
C09: FREE
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: PREINIT
C14: BEEPER
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : CAMERA_CONTROL
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

# 
```

[BTFL4.3_cli_Diablo2_VA3SEZ_20220110_084241__DiffAll_ResourceList.txt](https://github.com/betaflight/betaflight-configurator/files/7839638/BTFL4.3_cli_Diablo2_VA3SEZ_20220110_084241__DiffAll_ResourceList.txt)
[image]

SteveCEvans, 2022-01-10:
Bitbanged DSHOT has to use either `TIM1` or `TIM8`.

As `TIM1` is being used for `LED_STRIP` and `TIM8` is assigned for `CAMERA_CONTROL` you have a problem. Consequently streams for `DSHOT_BITBANG 1` and `DSHOT_BITBANG 2` required for the motor pins on ports A and B respectively can't be assigned.

```
Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : CAMERA_CONTROL
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE
```

You have four choices. Spoiler alert - you won't like 1 or 2.

1. `set dshot_bitbang=OFF`
This will result in the following.
```
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2:
    CH1 : MOTOR 4
    CH2 : MOTOR 3
TIM3:
    CH3 : MOTOR 1
    CH4 : MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : CAMERA_CONTROL
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: MOTOR 4
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE
```

Unfortunately the `MOTOR 4` assignment (which can't be moved) has trampled on top of `DMA1 Stream 5: SPI_MOSI 3`. This will prevent DMA accesses on SPI 3 used for the baro and MAX7456.

```
set baro_spi_device = 3
set max7456_spi_bus = 3
```

So let's discount that option.

Your other options are:

2. Disable `LED_STRIP` which will give you the following working configuration, but I suspect you want to use LEDs, so let's rule that one out too.
```
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
    CH3 : DSHOT_BITBANG 1
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : CAMERA_CONTROL
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: DSHOT_BITBANG 1
DMA2 Stream 7: FREE
```

3. `set camera_control_mode=SOFTWARE_PWM` to free up it's hardware timer.
```
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 2
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: DSHOT_BITBANG 1
```

4. `resource CAMERA_CONTROL 1 none` if you don't care about this feature, which you don't mention, so I presume you don't care about it.
```
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

# dma show

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 2
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: DSHOT_BITBANG 1
```

closed 2022-01-10 by SteveCEvans

recyclerealestate, 2022-01-10:
You are amazing, totally solved my problems. thank you.

ipushkar, 2024-07-08:
Having same issue when turning on Servo_tilt and LED_Strip. Enabling both causes the motors to stop working. Any suggestions ?

haslinghuis, 2024-07-08:
Did you try option 4 in previous comment?

ipushkar, 2024-07-09:
Sure I did, in my case I wasn't using this feature, so it never helped me.