betaflight/betaflight #11271: Failsafe when switching OSD profile on CRAZYBEEF4SX1280 target

AndreaCCIE, 2022-01-11:
### Describe the bug
I have set a switch to change between different OSD profiles. On the CRAZYBEEF4SX1280 when I pull the switch, it failsafe:
https://youtu.be/ywxDmCEGrZQ
The link doesn't recover itself, I have to power-cycle it.

No issue with the same configuration on other FC brands (but this is the only SPI I can test)

### To Reproduce
Apply the config below

### Expected behavior
OSD profile should change without failsafe

### Flight controller configuration

```
diff

# version
# Betaflight / CRAZYBEEF4SX1280 (HAMO) 4.3.0 Jan 11 2022 / 01:37:18 (b22c825ef) MSP API: 1.44

# start the command batch
batch start

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO

# name: ELRS-Whoop75

# feature
feature -AIRMODE
feature -ANTI_GRAVITY

# serial
serial 1 2048 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1400 2100 0 0
aux 1 1 3 900 1250 0 0
aux 2 28 4 1650 2100 0 0
aux 3 35 3 1650 2100 0 0

# adjrange
adjrange 0 0 2 900 1200 29 2 0 0
adjrange 1 0 2 1200 1800 29 2 0 0
adjrange 2 0 2 1800 2100 29 2 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# master
set gyro_lpf1_static_hz = 500
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 5
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 800
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = 5,-50,-67,1
set min_check = 1010
set max_check = 1990
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 300
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set simplified_gyro_filter_multiplier = 200
set report_cell_voltage = ON
set ledstrip_profile = RACE
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_rssi_dbm_alarm = -100
set osd_tim1 = 769
set osd_vbat_pos = 8289
set osd_rssi_pos = 57
set osd_link_quality_pos = 14390
set osd_link_tx_power_pos = 12747
set osd_rssi_dbm_pos = 14422
set osd_tim_1_pos = 14369
set osd_tim_2_pos = 215
set osd_flymode_pos = 12792
set osd_throttle_pos = 12375
set osd_vtx_channel_pos = 12779
set osd_current_pos = 8640
set osd_mah_drawn_pos = 12770
set osd_craft_name_pos = 364
set osd_pid_roll_pos = 391
set osd_pid_pitch_pos = 359
set osd_pid_yaw_pos = 423
set osd_warnings_pos = 14505
set osd_avg_cell_voltage_pos = 14401
set osd_disarmed_pos = 12555
set osd_esc_tmp_pos = 428
set osd_log_status_pos = 472
set osd_rate_profile_name_pos = 8343
set osd_pid_profile_name_pos = 8311
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set osd_stat_avg_cell_value = ON
set osd_menu_background = GRAY
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5806
set vcd_video_system = PAL
set max7456_clock = DOUBLE
set expresslrs_uid = 58,170,169,65,53,21
set expresslrs_switch_mode = WIDE
set name = ELRS-Whoop75

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_type = PT3
set dterm_lpf1_static_hz = 105
set dterm_lpf2_static_hz = 0
set iterm_relax = RPY
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set throttle_boost = 2
set p_pitch = 70
set i_pitch = 126
set d_pitch = 69
set f_pitch = 187
set p_roll = 67
set i_roll = 120
set d_roll = 60
set f_roll = 180
set p_yaw = 67
set i_yaw = 120
set f_yaw = 180
set d_min_roll = 45
set d_min_pitch = 51
set thrust_linear = 25
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 40
set dyn_idle_min_rpm = 30
set simplified_master_multiplier = 150
set simplified_dterm_filter_multiplier = 140

rateprofile 0

# rateprofile 0
set roll_rc_rate = 3
set pitch_rc_rate = 3
set yaw_rc_rate = 3
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 90
set pitch_srate = 90
set yaw_srate = 55
set tpa_breakpoint = 1550

# end the command batch
batch end

# ```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: PREINIT 2
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
CRAZYBEEF4SX1280 on the latest master plus #11229

<!-- Add any other context about the problem that you think might be relevant here. -->

haslinghuis, 2022-01-11:
@AndreaCCIE please try again with latest `master` plus #11229

SteveCEvans, 2022-01-11:
@AndreaCCIE Please try the image below from #11274 

[betaflight_4.3.0_STM32F411_c486c9d.hex.zip](https://github.com/betaflight/betaflight/files/7849655/betaflight_4.3.0_STM32F411_c486c9d.hex.zip)

haslinghuis, 2022-01-11:
@SteveCEvans please compile make TARGET=CRAZYBEE4SX1280

AndreaCCIE, 2022-01-11:
@haslinghuis same issue with 8d52a3615. Waiting for @SteveCEvans's one

SteveCEvans, 2022-01-11:
@AndreaCCIE 
[betaflight_4.3.0_CRAZYBEEF4SX1280_c486c9d.hex.zip](https://github.com/betaflight/betaflight/files/7849697/betaflight_4.3.0_CRAZYBEEF4SX1280_c486c9d.hex.zip)

AndreaCCIE, 2022-01-11:
@haslinghuis I built myself master + #11229 + #11274 and it works. Thanks @SteveCEvans

closed 2022-01-11 by AndreaCCIE

haslinghuis, 2022-01-11:
Just trying to keep up with https://github.com/betaflight/betaflight/pull/11229#issuecomment-1010369927

@AndreaCCIE Now the question is did you even need #11229. Could you please try latest master build with #11274 only.

AndreaCCIE, 2022-01-11:
I will test it as soon as possible, but to be honest #11229 didn't improve the situation that much in my case.

AndreaCCIE, 2022-01-11:
@haslinghuis just tested master and #11274 only: it works well, same behaviour with and without #11229