betaflight/betaflight #11273: Eachine Novice-II FC+VTX: Can't control VTX via Smartaudio protocol when using softserial

GregorPotrebujes, 2022-01-11:
### Describe the bug
I cannot change any VTX parameters via smartaudio over softserial. 
Fact no.1: VTX is properly configured, because smartaudio works well when connected to a real UART.
Fact no.2: softserial is properly configured, because I've been using it for my receiver (Frsky R9MX, over F.PORT) and it's working flawlessly.
What's interesting: in the Betaflight -> Video transmitter menu it says "Device ready - Yes".
Setting serialrx_halfduplex and serialrx_inverted to any combination/permutation doesn't change that status. It always says "Yes".But the VTX commands don't seem to go through in any inverted/halfduplex combination.

### To Reproduce
Connect the VTX to a FC via LED_STRIP pin, Remap LED_Strip pin to TX 11 and configure it to be used as smart audio in the ports tab in Betaflight.

### Expected behavior
If you make any changes in the "Video transmitter" tab and save it, the change should affect the actual VTX settings. The VTX should change channel, band or power according to the configuration. But it doesn't.
To be honest a bad smartaudio implementation in this VTX is probably the culprit. But I guess there are many more VTXs with the same issue... so there might be many grateful users if it gets fixed somehow... 

### Flight controller configuration
Here is my diff:
```
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.8 Feb 15 2021 / 12:09:04 (101738d8e) MSP API: 1.43

# config: YES

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# name: NOVICE-II

# resources
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A00

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature -RX_SPI
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234
[... 295 lines cut ...]
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
Hardware:
FC: Nano X F4 (from Eachine Novice-II)
VTX: Happy model diamond (whoop version) (also from Eachine Novice-II)
I have to use softserial for the VTX, because this FC only has two UARTs and I use them for GPS and receiver.

When searching for the solution I found that this might have been fixed in the latest iNav firmware... I would be very happy if someone could apply the same fix in the Betaflight, because I don't want to switch to iNav just because of it.
https://github.com/iNavFlight/inav/issues/7455
https://github.com/iNavFlight/inav/pull/7641

daleckystepan, 2022-01-11:
@GregorPotrebujes Try it with
```
resource SERIAL_TX 13 A00
```

GregorPotrebujes, 2022-01-12:
I tried:
```
resource SERIAL_TX 11 none
resource SERIAL_TX 13 A00
```
but I get the following message:
###ERROR: resource: INDEX NOT BETWEEN 1 AND 12###

Then I tried "resource SERIAL_TX 12 A00" and then configured TBS smartaudio for SOFTSERIAL2 in the Ports tab, but then I get this message in the Video Transmitter tab:
"Attention: Your VTX is not configured or not supported...." 
Since softserial can only be 11 or 12 in my case as far as I understand, I guess it will have to stay at Serial_TX 11...
Thanks you for your tip anyways! Any other suggestion?

daleckystepan, 2022-01-12:
@GregorPotrebujes Sorry man. I got the idea because of
https://github.com/betaflight/betaflight/blob/master/src/main/io/vtx_smartaudio.c#L497-L512
but I haven't realized that there are only 2 SOFTSERIALs. Maybe we need some fix for that as `default:` case cannot be used.

haslinghuis, 2022-01-12:
Please also disable feature `LED_STRIP`
```
feature -LED_STRIP
save
```

daleckystepan, 2022-01-12:
You can try this patched FW. It enables startbit for all serials including softserials.
[betaflight_4.2.11_STM32F411_norevision.hex.zip](https://github.com/betaflight/betaflight/files/7855869/betaflight_4.2.11_STM32F411_norevision.hex.zip)

GregorPotrebujes, 2022-01-12:
@haslinghuis: I already have it disabled. My diff doesn't say that, but that's probably because it's disabled by default I guess?
@daleckystepan : I flashed your patched firmware and it works!!! :) 
I can confirm that all communications work (receiver, GPS and VTX/smartaudio). Thank you very much!
After flashing I clicked the "apply defaults (or something)" button and left everything at default... all I did was: remapped the A00 pin to serial_tx 11, configured it to be used for smartaudio, loaded the vtx table and wolla! If I make a change in the video transmitter tab and hit save, the change gets applied on the VTX. After that I also confired receiver and GPS just to be sure that everything works.
I hope you will apply your solution to the official release too... Maybe by adding some sort of a variable, so that you can "apply this fix" or not (as with iNav).

haslinghuis, 2022-01-13:
@daleckystepan where is your PR for this patch. Please submit.

daleckystepan, 2022-01-13:
@haslinghuis @GregorPotrebujes Here it is. Pleas try: https://github.com/betaflight/betaflight/pull/11287 . Play with `vtx_softserial_alt ` and tell us which combination work for which VTX.

GregorPotrebujes, 2022-01-13:
Ok, so far I tested this PR on the the same hardware configuration as before (smartaudio over softserial) and it functions as intended ie. smartaudio works only with vtx_softserial_alt = ON. 
I will soon (probably tomorrow) flash my other two drones with this PR. One with Tramp-type VTX wired to a real UART and the other with SmartAudio-type VTX wired to a LED pin (softserial). Since both of these configurations worked even before that fix I'm expecting both will work with vtx_softserial_alt = OFF. I'll report back as soon as I manage to test that.

daleckystepan, 2022-01-13:
@GregorPotrebujes Thank you for the testing. Please be aware that PR is BF 4.3 and you need the latest Configurator to prevent flyaways etc.

daleckystepan, 2022-01-13:
@haslinghuis It will solve only issue if the FC is connected to VTX using SOFTSERIAL.

haslinghuis, 2022-01-13:
I'm aware of that. So thanks to @GregorPotrebujes for testing!

haslinghuis, 2022-01-14:
You can find it here: https://www.youtube.com/watch?v=nkOrVXiqefc

closed 2022-01-16 by blckmn