betaflight/betaflight #11276: OSD post flight menu

xqusme, 2022-01-11:
### Describe the bug
Post flight stats do on come on OSD after disarming.

### To Reproduce
Arm and Disarm

### Expected behavior
All of POST FLIGHT stats I selected should show up in OSD upon disarming.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Dec 24 2021 / 18:32:13 (norevision) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF411RX, version: a074838b, date: 2021-06-14T21:36:54Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411RX
manufacturer_id MTKS
mcu_id 003e002a3030510131333433
signature 

# name: MeteorBlue

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0

# feature
feature -AIRMODE
feature MOTOR_STOP
feature TELEMETRY
feature LED_STRIP

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

[... 521 lines cut ...]
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 MTKS/MATEKF411RX(STM32F411), version: 0
 - Flight controller: 
 Betafpv F4 whoop C01 camera
 - Other components:
 
 - How are the different components wired up: internal

    Happens on 3 different  whoops

daleckystepan, 2022-01-11:
@xqusme Should be already fixed. Try BF 4.3 RC2 please.

Xayfpv, 2022-01-20:
> @xqusme Should be already fixed. Try BF 4.3 RC2 please.
> 
> * [Reduce number of OSD bytes transferred in polled mode #11169](https://github.com/betaflight/betaflight/pull/11169)

Is there a specific Betaflight configurator for the RC2? or it works on all? I'm actually using the 10.8.0-debug-71fb7a68

haslinghuis, 2022-01-20:
Configurator RC2: https://github.com/betaflight/betaflight-configurator/releases/tag/10.8.0-RC2

Instead of flashing RC2 please choose development and grab the latest development build as it's has addressed some issues.

daleckystepan, 2022-02-07:
Is it working for you with RC3?

closed 2022-03-18 by github-actions[bot]