betaflight/betaflight #11284: OSD Slow to Update on MAMBA F411 Target

Bhorski, 2022-01-12:
### Describe the bug

The OSD is very slow to update, taking 25 sec to clear the startup screen with block-like display or clearing. OSD elements are displayed a few characters per second. For example, the post-fight statistics take about 15 sec to display after disarm, with individual characters appearing slowly.
### To Reproduce

### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

closed 2022-01-12 by github-actions[bot]

haslinghuis, 2022-01-12:
Try 
```
dshot_bitbang=ON
save
```

Bhorski, 2022-01-12:
Thanks! This worked.
Sorry that I could not get the request formatted right.  Glad that you still were able to see it.
Steve