betaflight/betaflight #11286: Power & Battery - Auto Calibration Issue

Blue-RR, 2022-01-13:
### Describe the bug
When using the auto calibration function for Voltage and Amp scale in power & battery page ,once the measured voltage and amperage in input the subsequent page does not show the calculated values. Once apply is pressed the old values still remain.

This issue was present on RC1 as well.

### To Reproduce
Happens all the time, irrespective of what FC is connected.

### Expected behaviour
Functionality should be as per 4.2.x. Once the measured values are input, the subsequent page should show the calculated values and once apply is pressed they should be input into the main page.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```

PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
- Flight controller: 
Happens on all FC listed below:
-- MATEKF411
-- KakuteF7miniV3
-- GHF13AIO

 - Other components:
-- R-XSR and ELRS Rx components for Rx.
-- Various VTx with Smartaudio.
-- Some have GPS functionality.

 - How are the different components wired up:
-- Serial communication.

klutvott123, 2022-04-16:
Just tested it with RC5 and it works fine. are you using the latest configurator?

closed 2022-04-16 by haslinghuis

Blue-RR, 2022-04-17:
Sorry, forgot to update this. Correct, this issue can be closed. It works fine from RC3 on wards.

randyyBro, 2022-04-24:
Hello, Everyone. Sorry out of topic

Anyone have dump file.tx cli for jhemcu ghf13aio? My FC is No Gyro Issue after update firmware and forgot to backup my last firmware. If anyone have, may i have it?