betaflight/betaflight #11293: Misaligned PID and filter settings after moving sliders with iPhone connection

lethuer, 2022-01-15:
### Describe the bug

On my 5" Freestyler I have a Diatone Mamba F722 APP FC that has Bluetooth with a board. I have now noticed with both BF 4.2.9 and 4.3 RC2 that the PID and filter settings are completely misaligned as soon as I move the PID slider with the iPhone... 

### To Reproduce

Connect with iPhone Speedy Bee App 
BF4.2.9: Move PID sliders, FF of Pitch axis will react weird
BF4.3 RC2: Move PID sliders and Filter settings will become weird

### Expected behavior

Just only the expected value(s) of corresponding slider should change, nothing else.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
tbd (currently on the phone)
```

```
PASTE THE OUTPUT OF 'resource show all' HERE
tbd (currently on the phone)

```

### Setup / Versions
 
 - Flight controller: Diatone Mamba F722 APP FC
 
 - Other components: TBS Tango 2, Rush Tank, GPS Mate, Beitan BN220 
 
 - How are the different components wired up:
Normal way, soldered copper wires

haslinghuis, 2022-01-15:
Speedy Bee app is not compatible with 4.3. Not sure about previous versions.

lethuer, 2022-01-15:
How do you know ? Only thing I could find is „>= 3.2“
https://www.speedybee.com/speedy-bee-app/
On the webpage 4.2.9 images are shown so at least with that version I shouldn‘t have problems.

DusKing1, 2022-01-15:
We are not the maintenance team for speedy bee. Go ask them. Not here.

abwyatt, 2022-01-15:
> How do you know ? Only thing I could find is „>= 3.2“ https://www.speedybee.com/speedy-bee-app/ On the webpage 4.2.9 images are shown so at least with that version I shouldn‘t have problems.

As others have pointed out, this is a Speedybee problem. They have updated their app for RC1 which fixed the tuning sliders problem, at least for Android phones. Not sure for iPhone since I don't use iPhone.

closed 2022-01-15 by lethuer

lethuer, 2022-01-17:
Response from speedy bee:
„ Sorry for the late reply. Yes, there's a bug in the iOS app, the fixed revision is still under development. It will be available in a few days. I'll let you know when it's available.“

lethuer, 2022-01-27:
New information from Speedybee:

„We've finally released our latest iOS app that supports BF 4.3. It's still under the App store's review process and will probably be available in 2 days. Please keep an eye on it.“