betaflight/betaflight #11295: Instructions/Manual for SITL

xuhao1, 2022-01-15:
Hi guys,
I am trying to develop add helicopter mode for Betaflight. My target is to use Betaflight on F3C/3D helicopters.
The first step I am trying to do is to set up software-in-the-loop (SIL) or hardware-in-the-loop (HITL).
I have already written scripts to interact with RealFlight G9.5s (and also a naive FC in python), which provides API for SIL/HITL.
I found Betaflight looks has [SITL target](https://github.com/betaflight/betaflight/blob/master/make/mcu/SITL.mk), but I can't find any instructions or manual to make Betaflight.
Can anyone help me with this?

DusKing1, 2022-01-15:
https://github.com/betaflight/betaflight/blob/master/src/main/target/SITL/README.md

xuhao1, 2022-01-15:
> https://github.com/betaflight/betaflight/blob/master/src/main/target/SITL/README.md

Thanks! I have implemented the link from RealFlight G9 to Betaflight.

xuhao1, 2022-01-16:
I am now working on improving the SITL of betaflight https://github.com/betaflight/betaflight/pull/11303.

closed 2022-01-16 by xuhao1