betaflight/betaflight #11298: Allow per-axis static notches

bluehallu, 2022-01-15:
### Is your feature request related to a problem? Please describe.

Static notch filters run unnecessarily on all axis. This adds unnecessary delay. 

In my GepRC Mark 4 HD5 there is a very prominent frame resonance at 193Hz on roll and another on 145hz on pitch. I can completely eliminate these with extremely narrow notches (e.g centered at 193hz and cutoff at 183hz or even 188hz), but the firmware forces me to run each of these on all axis which is wasteful.

### Describe the solution you'd like

Allow users to specify what axis they want a static notch to run

### Describe alternatives you've considered

I have considered relying on dynamic notches but this has some drawbacks:

* Running the FFT code is unnecessary in my case as the resonances are always at the same frequency across all throttle positions
* The resonances on roll/pitch have different amplitudes but I'm forced to pick a single Q value for all axis and all notches
* Generally, multiple narrow notches are better than a wide notch. Removing resonances with static notches should allow the dynamic notch to be concentrate on motor noise when RPM filtering is not available, to be disabled with RPM filtering or to be left with an even higher Q value to deal with other moving sources of noise for an overall lower filter latency

bluehallu, 2022-01-15:
This is what my roll axis gyro scaled looks like for reference, you can see the first notch is irrelevant. The pitch axis is the opposite.

[image]

UhlsenFPV, 2025-02-01:
I had the same idea and found this post so I want to give some attention to this again because i see the same problems mentioned in this post. Also found that multiple narrow notches are way more effective so the limitation of notch filters is pretty annoying and also seems unessecary to me. As well as the fact they are applied on all axis even though they are often only needed on one. I see a lot of potential in advanced notch filtering because it can be very effective without intoducing much delay especialy on the D-Term.

mateuszmiskowiak, 2025-12-29:
I agree, I also have a use cases for static notch with different setup per axis.