betaflight/betaflight #11304: Better SITL.

xuhao1, 2022-01-16:
### Is your feature request related to a problem? Please describe.
Currently SITL not working and only support Gazebo.
Personally, I think SITL is very important to develop flight controllers

### Describe the solution you'd like
I submit PRs https://github.com/betaflight/betaflight/pull/11302 https://github.com/betaflight/betaflight/pull/11303 to improve SITL of betaflight, making it work with [RealFlight](https://www.realflight.com/) with my own [bridge](https://github.com/xuhao1/RealFlightBridge), one of the best commercial RC simulators with accurate airplane, helicopter, and multirotor simulations.

I am still working on improving it and adding documents.

### Describe alternatives you've considered
Maybe move some open-source simulator here.

DusKing1, 2022-01-16:
AirSim maybe?

xuhao1, 2022-01-16:
> AirSim maybe?

Yes, possible in the future.

closed 2022-06-29 by blckmn