betaflight/betaflight #11306: Altitude hold

bakwc, 2022-01-16:
Hi. I would be glad to continue discussion in previous issues: #6675 and #10511 but it looks like discussion there is closed. I have a few questions:
1) The Alt hold mode should never be implemented in betaflight because this feature does not aligns with the vision of the project owners?
2) I would like to implement altitude hold for multi-rotors (I don't have plane - I can't test it there). Is it make any sense? Or you can accept this feature only if it works on all sort of vehicles?

haslinghuis, 2022-01-17:
https://www.te.com/usa-en/industries/sensor-solutions/applications/drone-sensors.html

bakwc, 2022-01-17:
> https://www.te.com/usa-en/industries/sensor-solutions/applications/drone-sensors.html

Sorry, have you read the question? How is this sensor related to a missing "Alt hold" mode

mituritsyn, 2022-01-17:
> 1. The Alt hold mode should never be implemented in betaflight because this feature does not aligns with the vision of the project owners?

Alt hold does not aligns with the philosophy of the project as a whole. Betaflight is for acro flying.

> Sorry, have you read the question?

I suppose this is a hint that not many FCs has a barometer and even fewer people will use it in a BF to keep altitude

bakwc, 2022-01-17:
> 
> Alt hold does not aligns with the philosophy of the project as a whole. Betaflight is for acro flying.
> 

With all the respect, I would like to hear this statement from one of the beta-flight official maintainers.

danik-guitar, 2022-01-17:
Лично мне нравится betaflight  и я хотел бы что бы удержание высоты было реализовано

haslinghuis, 2022-01-17:
@bakwc Please see background discussion here: https://github.com/betaflight/betaflight/issues/6675

See also this feature request: https://github.com/betaflight/betaflight/issues/4663

bakwc, 2022-01-17:
Thank you for references! 
#6675 - I read all the discussion but it is not clear if is it about broken old implementation, or about "philosophy" of the project. Obviously, I can try to implement more robust version, but I can't do anything with "philosophy". Don't want to spend time working on feature that will be rejected.

#4663 - Looks like a good feature request, I like that feature and want to see it in betaflight too. I see there is a few caveats that can make a good implementation of "ALT hold" more complicated, but I dont't think that it is' impossible. As far as i see - need to improve altitude estimation and need to tune PID filters to handle more vehicles gracefully.

etracer65, 2022-01-17:
The "philosophy" part is that flash space on popular flight controllers is limited and we'd prefer to use it for features that are consistent with the project's goals of high performance pilot-controlled flight. Autonomous flight features like alt-hold and gps-hold, robust RTH, etc. are not consistent with these goals. There are other projects that emphasize autonomous flight (like INAV).

bakwc, 2022-01-17:
> flash space on popular flight controllers is limited
Flash space issue could be solved using defines and disabling "heavy" additional features on flight controllers without much memory.
Anyway, I got the answer, thank you for clarification.

haslinghuis, 2022-01-17:
FYI Major brands are phasing out F411 and F722 this year so this limitation is only on these targets.

bakwc, 2022-01-17:
Actually the answer to my question is this:

> Autonomous flight features like alt-hold and gps-hold, robust RTH, etc. are not consistent with these goals.

It's just a matter of core team opinion.

I like "high performance pilot-controlled flight" but I also need a "pause" function, robust RTH function and some autonomous features from time to time.

INAV guys also does not support robust RTH - in fact they don't handle setups with missing / glitching compass. Ironically, they send their users to betaflight, see here: https://github.com/iNavFlight/inav/issues/4999

As the result we have users suffering because of "philosophy" of both projects.

haslinghuis, 2022-01-17:
That's why I'm personally open to this on > 512K targets. This might not reflect core team opinion, but we need to keep an open mind.

bakwc, 2022-01-17:
Cool! Would be great to hear some other members opinions.

limonspb, 2022-01-18:
i like the idea of alt hold. Especially if you say you can try to do it!
Flash space is not a problem, can target the code only for the specific versions of the chips and cover it with ifdef ALT_HOLD or something like that.

ctzsnooze, 2022-01-18:
I doubt that alt hold would add an appreciable amount of code. We already calculate altitude. We already have that code.. Altitude hold, from that data, only need a fairly simple PID controller. I made a simple P only altitude limiter that works quite well and it added very little to flash space.

Unfortunately, a baro or GPS based altitude hold is very approximate only,  the quad wobbles up and down, and the tuning requires some adjustment to avoid oscillations. In the worst situation the quad can bounce up and down potentially oscillating out of control.

If we included the Z axis acceleration to the control loop we could get much tighter control. But that would require another set of PIDs for the ‘inner loop’ using accelerometer data, a bit like level mode itself.  And that has not been done yet.

Simple baro/gps alt hold itself isn’t a problem, so long as expectations are low, but the coding side and, importantly, the tuning side, is nontrivial. And then there will be support requests etc. 

And of course altitude hold isn’t position hold, the wind blows the quad away, so people then want GPS hold as a feature creep, and so it goes to more and more work.

I’m very open to people working on altitude hold, but that first we actively need to overhaul the entire level/horizon accADC/attitude part of the code, to get the fundaments right.  There are aliasing and smoothing issues to resolve.

Anyone making an effective altitude control PR can provide binaries to make it easier for others to test..

bakwc, 2022-01-18:
Made a simple simulator in google colab (python), anyone can make a copy and tweak some values.
https://colab.research.google.com/drive/1gJL-LcBGRtqEyIFeYtQ3f2QA4oUnh-qW?usp=sharing

Currently struggling with correct PID values - tried different values but non of them leads to altitude stabilization. May be we really need a second PID filter for acceleration here.

ledvinap, 2022-01-20:
@bakwc : Did you try using altitude error as speed PI(D) setpoint? It makes more sense physically ...

ledvinap, 2022-01-20:
You can lookup double integrator system control, but most references are quite math heavy.

(Sorry, this will be confusing, I'm lacking proper math/control theory background)
Problem is with controller that tries to to control altitude by adjusting force/acceleration. If using simple P controller, you will adjust acceleration to reach target distance. On reaching the setpoint, acceleration is zero, but velocity is large (actually maximum). So target position is always overshoot, acceleration is set to negative value and velocity is eventually decreased to zero, but at maximum position overshoot. This results in sustained oscillation. And if there is some delay, oscillation amplitude will increase.
This system is quite equivalent to mass on spring system with controller output (P) = spring constant.

What you want is to control velocity. That way velocity is decreased to zero on reaching setpoint (and you have stable system). PID I term will remove steady state offset (you can't reach target when velocity is zero) and D term helps to stabilize system.

In altitude hold, velocity can't be measured, so it must be estimated.
The control system may be constructed using difference methodology, but IMO all will be quite similar in the end.

Also look how iNav handles it

bakwc, 2022-01-20:
Thank you, I will try to experiment with setting zero velocity. Actually GPS provide some velocity, and it's much more accurate than position. But still there could be some lags / glitches. Also we can estimate velocity from accelerometer (not sure if it's very accurate, but it should work if mix with other velocity source like GPS).

> Also look how iNav handles it

That's a good idea

ledvinap, 2022-01-20:
IMO you don't need accurate velocity estimate. PID shall correct the errors.
First use velicity from your model, to check that this approach works. Then try leaky integrator on acc data.

bakwc, 2022-01-20:
Checked velocity (in replace to acceleration PID):
1) When using a real velocity from simulation - everything works fine (better than using acceleration).
2) When using velocity based on acc data (with real simulated error) - everything is bad - fly in the sky / occilations and ground hit (actually it's expected - velocity estimation receives more and more error, and condition "velocity = 0" turns into "velocity = some random value based on acc error").
3) When mixing velocity estimation based on acc data and gps data - it's more-or-less stable, but amplitude of ocsilations are much bigger.
https://ibb.co/VCj49Dn  - acceleration based
https://ibb.co/6Fy8W5y  - velocity based

ledvinap, 2022-01-21:
@bakwc:
1. (this is good reference for tuning other variants)
2. should be simple to fix:
  - You can use leaky integrator (slightly reduce velocity toward zero in each step) 
  -  merge velocity measurement with height measurements (complementary filter). There are multiple ways to realize sensor fusion (mathematically quite similar). For example PI regulator with error between derivative of measured altitude and estimated velocity as input, adding output to ACC measurement.
3.  Add step input on altitude setpoint to test. It will better show regulator performance much better

ledvinap, 2022-01-21:
```
control algorithm:
   vel_setpoint = PI(D)(alt_setpoint, alt_estimate)
   throttle = P(I)D(vel_setpoint, vel_estimate)
estimation algorithm:
   # integrate acc->vel->pos
   vel_estimate += dt * acc_measurement
   alt_estimate += dt * vel_estimate
   # apply correction
  if (gps_sample): 
     pos_estimate, pos_error = fuse(pos_estimate, gps_estimate)
     vel_estimate += PID(0, pos_error) # KISS

bakwc, 2022-01-21:
We need another one PID for acceleration control. I checked ardupilot implementation - they use 3 PID filters: acceleration, velocity and altitude:
https://ardupilot.org/copter/docs/altholdmode.html
A nice scheme here:
https://ardupilot.org/copter/_images/Alt-Hold-PID-version-3.0.1.jpg

I will try this 3 pid approach later, currently porting my first approach with 2 pids (accel and altitude) to betaflight. If everything goes well - in a few days create PR and provide binaries to test.

ledvinap, 2022-01-21:
I'm quite sure that you won't get decent result without some form of velocity estimation. It is one of states of system you are trying to control.

Acceleration PID may improve performance, but it will probably work without it. And it will interact with gyro PIDs, but I have no idea if it will be improvement or fun.

bakwc, 2022-01-21:
Tested on 7″ and 5″ 5s, currently don't like how it works. Oscilations pretty high (+- 5 meters). I will try approach from ardupilot with 3 pids.

bakwc, 2022-01-22:
@ledvinap Tried to implement your approach more accurately, looks like it work pretty good in simulator
https://colab.research.google.com/drive/1fvWqWxYux3xSWPS1yDBx_jtSYRJo7OiA?usp=sharing
Leaky integrator gives some good velocity estimation alone (without mixing with gps / baro).
And acceleration PID is not required - everything works with two PIDs.
The only thing I added - smoothing for gps/baro based altitude measurements.
Now porting to betaflight. Seems like I'll have to use blackbox logging to debug all the stuff and find appropriate coefficients.

bakwc, 2022-01-24:
Time to start testing! Instruction and firmware in #11327
Feedback / suggestions appreciated.

ledvinap, 2022-01-24:
@bakwc: You should try step response for PID tuning .. 
```
for n in range(int(NUM_ITERS * Settings.ticksPerSecond)):
    if n > 1000:
        gState.target_alt = 100
    drawframe(n)
```

bakwc, 2022-01-24:
Added, https://colab.research.google.com/drive/1fvWqWxYux3xSWPS1yDBx_jtSYRJo7OiA?usp=sharing

Testing in simulator and in reality both, looks good.

bakwc, 2022-02-28:
It's not Inactive. It's waiting for review.

limonspb, 2022-04-16:
@haslinghuis  can we add some special tag so that the bot does not close it?

pkarc, 2022-04-19:
This sound interesting, now days using cinelifters and expensive cinema cameras, this would be pretty nice!

YKSyoutube, 2022-05-30:
Alt hold and position hold would be so great. All cinelifters pilots wants this. on failsafe it can save a 12 000$ setup. 
Better than rescue mode wich can make the setup hit a building or a tree during a shoot.

bakwc, 2022-07-01:
I am not continue of working on this mode because of the toxic behavior of some of the betaflight core team members.
So you can close it or someone else could implement it, I don't care.

cinedrone, 2022-10-29:
Just a thread bump for a useful feature. Cinematic FPV is now trending hard and many new people join the FPV world in order to fly cine lifters and smaller quads. Lets give them access to safety features. I already see the new Moneyshot mini v2 using iNav instead of BF.

YKSyoutube, 2022-10-30:
> 

I already put all my cinelifters on iNav, and to be honest, they fly as good as BF. that pos/alt hold + MC breaking is stunning to save a expensive rig in case you lose video. Ad the RTH is just perfect !

Alessiofpv, 2022-11-03:
> Just a thread bump for a useful feature. Cinematic FPV is now trending hard and many new people join the FPV world in order to fly cine lifters and smaller quads. Lets give them access to safety features. I already see the new Moneyshot mini v2 using iNav instead of BF.

I totally agree with you, but if i think adding a security feature to betaflight is like turning some people away from inav, now i could understand this behavior.

Anyway during my pain for lack of altitude hold, i take my Tango2, and desperately set a self made altitude hold using barometer altitude from crossfire telemetry, obviously telemetry is too slow and my experiment is very unstable but it works LOL.  So i think why not make altitude hold easier without a pid controller etc.  I am not a programmer I'm curious. @bakwc

charlesmarseille, 2023-03-10:
This is a sad end to an interesting project. I understand both sides of the medal, just sad that medal had to be burnt in the flames of mordor along with its creator. 
How are we standing as of today? Any alt hold going on?

notching, 2023-05-23:
Dang, this would really help my son who is just starting to get into the sport.  Can't quite believe betaflight does not have this option - it has almost everything else! Reminds me of : "640K ought to be enough for anybody" :-).

jawz101, 2023-05-26:
I'd like the ability to altitude hold or hover in place simply to scratch my nose or whatever while I'm flying.  Or if I am teaching someone to fly I can get it up in the air for them or they can pass it back to me if they are getting nervous.  Also, in a hypothetical future, what if flight times ever became > 30 minutes.  What if there was a doubles or relay race competition where flying was handled by teams of pilots.  They could pass the controller back & forth to keep up in the air.

RafaelCosman, 2023-11-28:
+1 for adding altitude hold. Would be a great feature and doesn't seem like it would take up much space!

iDevy, 2023-12-14:
+1. Found this post looking at options to add vl5310x to cinelog drone. For those cine*-drones its an e xremely valuable feature.

meminoner, 2024-01-27:
I can't believe that we don't have alt hold feature on BF! I bought a board with baro just for this 🤯

limonspb, 2024-01-28:
> I can't believe that we don't have alt hold feature on BF! I bought a board with baro just for this 🤯

you can test this PR tho.

mituritsyn, 2024-01-28:
> I can't believe that we don't have alt hold feature on BF! I bought a board with baro just for this 🤯

test this #13186

meminoner, 2024-01-28:
I did it yesterday. I'll write comment on that PR #13186

pavlokleymonov, 2024-03-10:
> Лично мне нравится betaflight и я хотел бы что бы удержание высоты было реализовано

это вам всем нужно чтобы легче сбросы делать?

dfaliush, 2024-07-20:
Does betaflight support the alt hold mode now?

limonspb, 2024-07-21:
> Does betaflight support the alt hold mode now?

@KarateBrot :)

closed 2024-09-04 by blckmn