betaflight/betaflight #11307: Wrong altitude when arming without enough GPS sats

bakwc, 2022-01-16:
### Describe the bug
I have a wrong altitude if I arming my quad to early (without enought GPS Sats). I have both - BARO and GPS enabled.

### To Reproduce
1) Make sure that you have GPS and BARO both.
2) Arm your quad with zero or few GPS sats
3) Start flying

### Expected behavior
Altitude is accurate

### Actual behavior
Altitude has a huge error (~100 meters)

### Proposal fix
I would like to implement the following fix:
1) Add option "Min number of GPS Sats to perform GPS altitude lock", set them to 6 by default
2) While having both GPS and BARO - delay GPS altitude locking untill required number of Sats reached. When using delayed lock - use BARO altitude as initial GPS lock altitude.
Does it make any sense?

SteveCEvans, 2022-01-16:
If you have both barometer and GPS I’d suggest using just the baro as the vertical resolution of GPS is inferior.

bakwc, 2022-01-16:
The issue with BARO only is that it's floating - when flying for a long time with only BARO enabled - the altitude could easily float up to 10-20 meters. GPS altitude is good enought - just need to avoid locking untill HDOP is good enough.

bakwc, 2022-01-16:
if (ARMING_FLAG(ARMED) && !altitudeOffsetSet) {
        baroAltOffset = baroAlt;
        gpsAltOffset = gpsAlt;
        altitudeOffsetSet = true;
    }

Here gps values are locked - no sanity checks at all. There could be even no fix at all! And values are still locked. Are you really sure that this is a good **default** behavior?

haslinghuis, 2022-01-16:
Did you shield your barometer with open cell foam?

bakwc, 2022-01-17:
> Did you shield your barometer with open cell foam?

How is shielding my barometer related to a wrong **GPS** locking values?

PR #11312 (2022-01-30): Fixed wrong altitude in DEFAULT (GPS + BARO) mode
Fixes #11307

Default altitude estimation behavior changed to following:

1) Drone arms
2) Drone lifts off and use BARO values untill gps is good enough (has at least N sats)
3) Quad start using both sensors

Motivation - using GPS altitude until it does not have enough sats leads to strong altitude estimation error (100 - 200 meters).
closed 2022-01-30 by haslinghuis