betaflight/betaflight #11309: SPRacing H7 Zero Betaflight 4.3 RC1/RC2 Bidirectional dshot issue - some motors not spinning

EdgaFPV, 2022-01-16:
### Describe the bug

2 out of 4 motors not spinning (Error: 100%) while Bidirectional Dshot is ON. After turning Bidirectional Dshot OFF all motors spin correctly.

### To Reproduce

1. Turn ON Betaflight Configurator (10.8.0 RC2).
2. Connect flight controller (SPRacing H7 Zero with Betaflight 4.3 RC2)
3. Go to Motors Tab
4. Connect battery to the Quad
5. Turn on "I understand the risks, ..."
6. Spin motors by pushing Master slider up
7. Motors 2 and 4 won't spin and have 100% error.

### Expected behavior

All 4 motors spinning

### Flight controller configuration

```
# version
# Betaflight / SPRACINGH7ZERO (SP7Z) 4.3.0 Dec 24 2021 / 18:27:08 (norevision) MSP API: 1.44

# start the command batch
batch start

board_name SPRACINGH7ZERO

# name: EDGA_FPV_SOURCE

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER
feature -AIRMODE

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 1 115200 57600 0 115200

# beeper
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# aux
aux 0 0 0 1000 1525 0 0
aux 1 28 0 1000 1250 0 0
aux 2 1 0 1000 1125 0 0
aux 3 1 0 1300 1375 0 0
aux 4 1 0 1800 1825 0 0
aux 5 13 0 1800 1825 0 0
aux 6 13 0 1925 1950 0 0
aux 7 35 0 1125 1150 0 0
aux 8 35 0 1200 1225 0 0
aux 9 35 0 1375 1400 0 0
aux 10 35 0 1450 1475 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -22,41,34,1
set mag_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 8
set serialrx_provider = CRSF
set blackbox_device = SPIFLASH
set dshot_idle_value = 450
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_pitch = 180
set vbat_min_cell_voltage = 340
set vbat_warning_cell_voltage = 365
set vbat_scale = 109
set ibata_scale = 160
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 10
set yaw_deadband = 25
set pid_process_denom = 2
set osd_warn_arming_disable = OFF
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_crash_flip = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_tim1 = 2561
set osd_vbat_pos = 395
set osd_rssi_pos = 2304
set osd_tim_1_pos = 2432
set osd_crosshairs_pos = 2318
set osd_current_pos = 416
set osd_mah_drawn_pos = 2464
set osd_altitude_pos = 2048
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2496
set osd_disarmed_pos = 2059
set osd_core_temp_pos = 408
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set name = EDGA_FPV_SOURCE

profile 0

# profile 0
set feedforward_smooth_factor = 50
set feedforward_jitter_factor = 10

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: FREE
A05: FREE
A06: FREE
A07: FREE
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: QSPI_CLK 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: FREE
B06: I2C_SCL 4
B07: I2C_SDA 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: QSPI_BK1CS 1
B11: FREE
B12: PREINIT 2
B13: FREE
B14: SERIAL_TX 1
B15: SERIAL_RX 1
C00: FREE
C01: ADC_CURR
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: ADC_BATT
C05: ADC_EXT
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SDIO_CMD
D03: SPI_SCK 2
D04: GYRO_EXTI
D05: FREE
D06: SPI_MOSI 3
D07: BEEPER
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: FREE
D11: QSPI_BK1IO0 1
D12: QSPI_BK1IO1 1
D13: QSPI_BK1IO3 1
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: QSPI_BK1IO2 1
E03: LED 1
E04: SYSTEM
E05: CAMERA_CONTROL
E06: FREE
E07: QSPI_BK2IO0 1
E08: QSPI_BK2IO1 1
E09: QSPI_BK2IO2 1
E10: QSPI_BK2IO3 1
E11: OSD_CS
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15:
    CH1 : CAMERA_CONTROL
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 1
DMA1 Stream 1: SPI_MOSI 3
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: SPI_MOSI 4
DMA1 Stream 4: SPI_MISO 4
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: SPRacing H7 Zero, bought from Dominic Clifton
 
 - Other components:
 - ESC: Holybro Tekko32 F3 45A with BlHeli32 V32.8 (tried on V32.7 - same issue)
 
 - How are the different components wired up: wired correctly, works withous issue with older Betaflight 4.2.11.

SteveCEvans, 2022-01-16:
Testing with current master (`ee740d1`) with a T-Motor F55A ESC there is no issue.

haslinghuis, 2022-01-17:
Tested SPRaving H750 Extreme on master with Turnigy Multistar BLHeli_32 ARM 51A Race Spec ESC 2-6S (Opto).
No issues with bidirectional Dshot.

EdgaFPV, 2022-01-17:
I have some some differences in resources:

```
Betaflight 4.3 RC2 

Currently   active Timers:
-------------------------
TIM5:   FREE
TIM8:
CH1   : DSHOT_BITBANG 1
 
Currently   active DMA:
--------------------
DMA1   Stream 0: DSHOT_BITBANG 1
DMA1   Stream 1: SPI_MOSI 3
DMA1   Stream 2: SPI_MISO 3
DMA1   Stream 3: SPI_MOSI 4
DMA1   Stream 4: SPI_MISO 4
```

```
Betaflight 4.2.11

Currently active Timers:
-------------------------
TIM5:
CH1: MOTOR 1
CH2: MOTOR 2
CH3: MOTOR 3
CH4: MOTOR 4
TIM8: FREE
 
Currently   active DMA:
--------------------
DMA1 Stream 0:   MOTOR 1
DMA1 Stream 1:   MOTOR 2
DMA1 Stream 2:   MOTOR 3
DMA1 Stream 3:   MOTOR 4
DMA1 Stream 4:   FREE
```
Maybe these differences have impact on BiDireectional Dshot not working properly on some motors. With BF 4.2.11 BiDirectional Dshot works flawlessly.

Xiang-FPV, 2022-01-25:
@SteveCEvans @haslinghuis I also encountered this problem. When UPDATING the mambaH7 FC with 4.3Rc2 and opening the Bidirectional Dshot, the motor error100% appeared. However, there was no problem when I updated 4.2.11. My other NEUTRONRCH7BT FC does not have this problem on 4.3RC2 and I suspect some of H7 FC will

haslinghuis, 2022-01-25:
No problems with SPRACINGH7EXTREME:

[image]

Xiang-FPV, 2022-01-26:
@haslinghuis Are SPRACINGH7CERO and SPRACINGH7EXTREME the same FC? I also found that the motor error rate was 100% when Bidirectional Dshot was turned on for only some types of flight controllers, such as MAMBA H743. I found that many of my friends had the same situation when they bought this FC

AlexRice13, 2022-01-26:
> @haslinghuis Are SPRACINGH7CERO and SPRACINGH7EXTREME the same FC? I also found that the motor error rate was 100% when Bidirectional Dshot was turned on for only some types of flight controllers, such as MAMBA H743. I found that many of my friends had the same situation when they bought this FC

>Experiencing the same problem on MAMBA H7 FC BMI270 version with MAMBA F65 128K，I managed force those motor work under 
bi-directional DShot by disabling dshot_bitbang in CLI
`set dshot_bitbang = OFF
save`
However It worked not fine, showing about 99.7 error, and two motors refreshed it's rpm in very slow rate.
这H7飞控反向升级了属于是。
[image]

haslinghuis, 2022-01-26:
@Xiang-FPV

[image]

Which ESC + firmware is being used (please try 32,8.3 with default settings).

howels, 2022-01-30:
I also have this problem with the JHEH743 - 2/4 motors are not working with bitbang enabled, when I disable bitbang I can see that motor4 appears to be the problem with only that motor failing to initialise. Tested with https://github.com/betaflight/betaflight/commit/074172b4c94206e33a42d3aa5d8e4da2bd1a3e6d

howels, 2022-01-30:
FYI - going back to 4.3.0 RC1 allows all four motors to spin (dshot_bitbang OFF). But RPM is only reported from 2/4 motors and error rate is very high
[image]

gchedzhemov, 2022-01-30:
> FYI - going back to 4.3.0 RC1 allows all four motors to spin (dshot_bitbang OFF). But RPM is only reported from 2/4 motors and error rate is very high 

I was also able to reproduce this behavior on my JHMECU GH743AIO (target `JHEH743`). So far I tried various versions of BF 4.3.0 (including RC1, RC2 and nightly), as well as various versions of 4.2.x

Experimentation with various combinations of `dshot_bitbang` and `dshot_bitbang_timer` was not very fruitful, as none of the combinations allow for more than two functional ESC motor outputs. Best I can do is to get motors 4 and 2 working with bidir dshot, while 1 and 3 are at 100% error. (`dshot_bitbang = ON` and `dshot_bitbang_timer = AUTO` or `TIM1`. `TIM8` breaks all motors)

Happy to try any and all suggestions for getting bidir dshot working on this H743 AIO.

howels, 2022-01-31:
I confirm that as per https://github.com/betaflight/betaflight/pull/10383#issuecomment-932937466, turning on `DEBUG=GDB` allows the board to work.

howels, 2022-02-01:
> I was also able to reproduce this behavior on my JHMECU GH743AIO (target `JHEH743`). So far I tried various versions of BF 4.3.0 (including RC1, RC2 and nightly), as well as various versions of [...]

Try a build with `DEBUG=GDB` set and leave dshot_bitbang=AUTO.  It should work then. Why it works is another question entirely...

sugaarK, 2022-02-10:
@howles  can you send me the hex you built for the jheh743 ?

haslinghuis, 2022-02-10:
@sugaarK choose the board in configurator (or use detect button) and then load this hex file locally:
[betaflight_4.3.0_STM32H743_3267f0417.zip](https://github.com/betaflight/betaflight/files/8044875/betaflight_4.3.0_STM32H743_3267f0417.zip)

howels, 2022-02-11:
> @howles can you send me the hex you built for the jheh743 ?

Waiting for a fix which gets bidirectional dshot working 100% of the time. I still get failure to bring motor3 online around 20% of the time.
I will add a unified target for JHEH743 once there is a fix for all H743 boards. For now there appears to be some discussion around H743 DMA configuration and the build is not reliable.

sugaarK, 2022-02-11:
> 

there is no target for this board

sugaarK, 2022-02-11:
> @sugaarK choose the board in configurator (or use detect button) and then load this hex file locally:
> [betaflight_4.3.0_STM32H743_3267f0417.zip](https://github.com/betaflight/betaflight/files/8044875/betaflight_4.3.0_STM32H743_3267f0417.zip)

there is no target for this board...

closed 2022-03-22 by github-actions[bot]

Bravehawk666666, 2022-05-19:
Same Problem here. T-Motor FC and ESC. 
"T_Motor_F55A_4in1_PRO_V2"
Betaflight 4.3 RC6. Updated ESCs to 32.9 from 32.8. No change.

The two not working ESCs become also invisible in BL-Heli Suite ("BLHeliSuite32_32.9.0.3") when BF4.3RC6 is installed.

Reinstalling version 4.2.11 (TMOTORF7V2, binary: betaflight.4.2.11_STM32F7X2.hex) fixes the problem. Also for BL-Heli Suite. All Motors working then normally.