betaflight/betaflight #11310: Helicopter Mode for BetaFlight

xuhao1, 2022-01-16:
### Is your feature request related to a problem? Please describe.
Currently, the BetaFlight has only a helicopter mixer, the logic related to the helicopter is lacking.
Helicopters are very different from quadrotors, a general F3C/3D RC helicopter uses collective to control the thrust and fixed rotor speed.
Also, the helicopter has very different dynamics compared to quadcopters, this is caused by flapping of the blade and the tip path plane (TPP) dynamics exist.

On the other hand, we control the angular rate of F3C/3D helicopter and direct pass the stick to the collective to generate linear thrust. These points are the same compared to FPV Racer. Some helicopter FC (VBar) also has the function of thrust boost (boost the collective). So there are common points from the helicopters to  FPV racers.

The best thing RC helicopters compared to FPV racer is it naturally has a so-called "3D" mode. The collective can be negative and generate negative thrust, this may also benefit FPV flight if you want have a FPV helicopter racer.

### Describe the solution you'd like
I choose to implement an acro helicopter mode in BetaFlight. I believe BetaFlight is easy to develop and more suitable to acro flight compared to PX4/ArduPilot/iNav

Currently, I have reworked the mixer in my own [branch](https://github.com/xuhao1/betaflight/tree/heli). 
Here is a demo video of my current progress. In this video, I use RealFlight 9 to simulate the helicopter, use [RealFlightBridge](https://github.com/xuhao1/RealFlightBridge) to bridge it to BetaFlight SITL (see this [issue](https://github.com/betaflight/betaflight/issues/11304) and [PR](https://github.com/betaflight/betaflight/pull/11303)).
[image]

I will continue the develop helicopter mode and test on a real RC helicopter with BetaFlight in the next few months.

Targets:
 - Rework PID loops for the helicopter. Better feed-forward for roll pitch control loop.
 - RPM managements, normal/IDLE\_UP1/IDEL\_UP2/IDEL\_UP3.
 - Swash and tail configurations.
 

### Describe alternatives you've considered
My possible targets (After months):
 - RPM telemetry and notch for the main rotor.
 - Support helicopters use a BLDC for yaw control.
 - Telemetry to transmitters.

KarateBrot, 2022-01-16:
Maybe you also want to have a look at https://github.com/heliflight3d/heliflight

xuhao1, 2022-01-16:
> Maybe you also want to have a look at https://github.com/heliflight3d/heliflight

@KarateBrot Thanks. Cool project but looks like the project is EOL.  Personally, I think submitting to BetaFlight is better than maintenance an own branch.

xuhao1, 2022-01-16:
I find a active helicopter fork of betaflight: rotorflight. I will moving my helicopter works to that project.

haslinghuis, 2022-01-16:
Rotorflight is not very active. I guess you should reconsider using Betaflight or iNav as these are the most active projects at the moment (besides ArduPilot / PX4)

SteveCEvans, 2022-01-16:
I’m impressed!

xuhao1, 2022-01-17:
> Rotorflight is not very active. I guess you should reconsider using Betaflight or iNav as these are the most active projects at the moment (besides ArduPilot / PX4)

xuhao1, 2022-01-17:
> Rotorflight is not very active. I guess you should reconsider using Betaflight or iNav as these are the most active projects at the moment (besides ArduPilot / PX4)

@haslinghuis Thanks for your advice!
I will try to submit my current work on  betaflight helicopter and reconsider continue develop on BF.

ghostkcleung, 2022-02-03:
I believe that the performance and the cost of F4, F7, H7 and the related gyro is better than VBar if the betaflight develop CCPM helicopter.

But I concern how to provide the power to the servos. Usually the flight controller we are using is 5V ADC. But the heli servo is 8V.

xsnoopy, 2022-04-19:
Rotorflight is what you need. First release candidate is coming soon. Join the discord server, very active community.

closed 2026-09-26 by blckmn