betaflight/betaflight #11314: After flash 5" quad with FW v4.3-RC2 w/default PIDs the quad is glitching with rapid stutter movements...

JoeSa3rd, 2022-01-18:
### Describe the bug
My 5” quad was flying just fine on Betaflight v4.2.10. I then flashed the quad with Betaflight fw v4.3-RC2 default PIDs, and on the first flight I’m getting what feels like glitching that is causing the quad to have rapid stutter like movements that seem to get worse with just slightly more stick movement. See link to video for a better example of the issue -> https://www.dropbox.com/s/xuqh82k0zgtti0h/Betaflight%20v4.3.0-RC2%20Issue%20Flight%20Video.mp4?dl=0

### To Reproduce
Flash to BF v4.3-RC2

### Expected behavior
Normal flight with no stutter like movements

### Flight controller configuration
iFlight SucceX-D Mini F7 TwinG FC – FW Betaflight v4.3-RC2 IFLIGHT_F722_TWING

```
# 

# dump

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan  9 2022 / 11:51:40 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F722_TWING, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC

# name: -

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
[... 1069 lines cut ...]
Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```
### Setup / Versions
Jumper T18 – OpenTX v2.5.0 (ADC Filter = Off)
TBS Crossfire TX - FW v6.13, Dynamic Pwr, 250mw
TBS Nano RX - FW v6.13
iFlight SucceX-D Mini F7 TwinG FC – FW Betaflight v4.3-RC2 IFLIGHT_F722_TWING

haslinghuis, 2022-01-18:
As we are still in development phase please retry with latest nightly build

JoeSa3rd, 2022-01-18:
I'll try the nightly build to see if that has better results. I'm running 4.3-R2 on two other quads without any notable issues so I'm a little puzzled about this setup...
'''
# Diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Jan  9 2022 / 11:51:40 (ee8560af4) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F722_TWING, version: c2b8c0d3, date: 2021-06-18T03:42:42Z

# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC

# feature
feature -SOFTSERIAL
feature GPS
feature TELEMETRY

# serial
serial 1 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 2 115200 9600 0 115200

# beacon
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 46 2 1300 2100 0 0
aux 2 13 1 1300 2100 0 0

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 28,-9,28,1
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set serialrx_provider = CRSF
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 85
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 38
set gps_rescue_initial_alt = 75
set gps_rescue_ground_speed = 3000
set gps_rescue_throttle_max = 1650
set gps_rescue_min_sats = 7
set gps_rescue_allow_arming_without_fix = ON
set osd_units = IMPERIAL
set osd_vbat_pos = 14340
set osd_rssi_pos = 14361
set osd_link_quality_pos = 14465
set osd_link_tx_power_pos = 14401
set osd_rssi_dbm_pos = 14433
set osd_current_pos = 14355
set osd_mah_drawn_pos = 14350
set osd_gps_speed_pos = 14387
set osd_gps_sats_pos = 14391
set osd_home_dir_pos = 14447
set osd_home_dist_pos = 14382
set osd_altitude_pos = 14374
set osd_warnings_pos = 14761
set osd_avg_cell_voltage_pos = 14346
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set gyro_2_align_yaw = 900

profile 0

rateprofile 0

# rateprofile 0
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 55
set pitch_srate = 55
set yaw_srate = 55

# end the command batch
batch end
'''

bakwc, 2022-01-18:
While testing another feature on master - I also experienced a little stutter. Not so big as @JoeSa3rd but noticeable. On 4.2 everything was smooth. Used the same PID values as in 4.2, also tried with default PID values. I will investigate later when have some time and provide more details.

haslinghuis, 2022-01-18:
@bakwc Please try last nightly as there have been relevant changes.

JoeSa3rd, 2022-01-18:
@haslinghuis I flash BF v4.3.0 nightly build (Jan 18th, 2022) and all is well now, absolutely no rapid stutter movements. BF v4.3 feels great, keep up the great work everyone.

haslinghuis, 2022-01-24:
Thanks keep in mind filters and pid defaults have changed. Check with blackbox for where resonances needs fixing.

closed 2022-01-24 by haslinghuis