betaflight/betaflight #11315: Crazybeef4fr spi Rx dropping connection

D3s0l4ti0n, 2022-01-18:
Crazybee f4 pro V3 on bf 4.3 rc2 spi frsky x2 binding with x9lite constantly loses connection with alot of stick inputs and if it just sits there with no inputs it will lose connection ever 10 to 25 seconds binding to a radiomaster tx16s there is no issue but binding to a x9 lite with access 2.0 firmware I can not fly my tiny whoop it will fall out of the air and then Recover and it gets worse the more you move the sticks. With the quad hooked up to beta flight watching the receiver tab and rotating the roll pitch stick with throttle at full the flight controller will glitch and lock up and drop Rx connection and the faster you move the stick the more it happens you can see the throttle got from full to none every few seconds while moving the roll pitch stick and not touching the throttle

SteveCEvans, 2022-01-18:
Please try current nightly as there have been relevant changes.

D3s0l4ti0n, 2022-01-19:
will do thank you .

phobos-, 2022-01-20:
@D3s0l4ti0n throttle up and roll either max left or max right is a stick command that is most likely causing the FC to write settings to eeprom, and that is causing these intermediate drop-outs. Could you try the same exercise but with throttle not set to full?

SteveCEvans, 2022-01-20:
If the current nightly doesn’t address the issue, please try the image from https://github.com/betaflight/betaflight/pull/11319

D3s0l4ti0n, 2022-01-21:
> @D3s0l4ti0n throttle up and roll either max left or max right is a stick command that is most likely causing the FC to write settings to eeprom, and that is causing these intermediate drop-outs. Could you try the same exercise but with throttle not set to full? I figured you didn't manually set these values?
> 
> ```
> set acc_trim_pitch = -46
> set acc_trim_roll = 4
> ```

it does the same with the throttle off or half i just had it at full to show what it was doing i had the arm switch flipped while doing this if i just let it sit there it will drop every 10 to 15 seconds but if i dont inputs even if its just one stick itll drop out about every 2 to 4 seconds im waiting on a motor and works been busy so i havent had time to try the nightly build im going to update to the nightly tomorrow and see what happens

Nikotttin, 2022-01-27:
Just wanted to let all know that I encountered the same issue. 
SPI FRSKY RX within the CrazybeeF4FR Pro AIO, RC smoothing set to 2. F411 overclocked by 25%.
The RC link "flip flops" between 0 and wherever your stick is. RSSI flip flops too between 0 and "normal" suggesting a series of link drops. I flashed the latest nightly and it still behaved as it.
I thus flashed back 4.2.11 and it works flawlessly, with steady RSSI...

Thanks for all the work you guys do !!

D3s0l4ti0n, 2022-01-29:
sorry i took a bit work has been crazy busy i flashed the latest dev build and the issue has gone away it no longer drops connection and all the rx issues are gone i haven't flown yet since im waiting on replacement motors but from what i can see it should fly great now  build number #2634 28-01-2022

D3s0l4ti0n, 2022-01-29:
never mind after a reboot its doing the same thing just not as frequently

D3s0l4ti0n, 2022-01-29:
before i setup my modes it was working fine then after setting up arm turtle air mode and angle horizon modes it started having the issue again

haslinghuis, 2022-01-30:
New builds are available for testing now.

D3s0l4ti0n, 2022-01-30:
still the same issue it drops connection randomly its like the whole fc is locking up

D3s0l4ti0n, 2022-01-30:
i have a crazybee f4 pro v2.0 and it doesn't have this issue but both v3 boards do there all frsky spi

D3s0l4ti0n, 2022-01-31:
Could my pid loop frequency be to high at 4khz with rpm filtering on motors

D3s0l4ti0n, 2022-01-31:
Gonna mess with some settings and see if it changes anything but the fact it works fine on a tx16s bound in frsky_X_V2 with no issue but binding to trannis x9 lite is unflyable

Professor77, 2022-02-01:
I also had this issue with my crazybeef4fr spi rx.
Almost posted myself but I checked to see if anyone else had mentioned it.
Went back to 4.2.11 and it's fine now. 
I know that the issue will be worked out eventually and I am super excited for 4.3!

Professor77, 2022-02-01:
RC3 seems to have fixed the issue. That was speedy work gentle-devs!

daleckystepan, 2022-02-07:
Fixed in RC3.

closed 2022-02-07 by daleckystepan

timemaster5, 2022-09-07:
Aaaand it’s broken again in 4.3.1 :( 4.3.0 from RC1 to 7 and also 4.3.0 stable works great.

Professor77, 2022-09-07:
I had to go back to 4.2.11 but 4.3 is great on some of my other quads with
external elrs receivers.