betaflight/betaflight #11317: BF 4.3.x fly-to-the-moon when arming Zeus10 (target:ZEUSF4EVO)

Mode-1, 2022-01-19:
### Describe the bug
When arming BF 4.2.x with Zeus10 (target:ZEUSF4EVO), motors do idle
When arming BF 4.3.x with Zeus10 (target:ZEUSF4EVO), motors do spin up rattling to approx 25-40% thrust, launch control automatically prevents a flytothemoon but quad lifts up to 1m above ground

### To Reproduce
just arm Zeus10 with BF 4.3.x flashed on it, leave props off !

### Expected behavior
all motors idle like Zeus10 with BF 4.2.x flashed on it

### Flight controller configuration

```
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.11 Nov  9 2021 / 20:28:23 (948ba6339) MSP API: 1.43
# config: manufacturer_id: HGLR, board_name: ZEUSF4EVO, version: 7310ccb4, date: 2021-10-10T02:07:48Z

# start the command batch
batch start

board_name ZEUSF4EVO
manufacturer_id HGLR

# name: TX-85HD

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature LED_STRIP
feature OSD

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 1 115200 57600 0 115200

# led
led 0 0,0::CW:9

# color
color 9 240,0,100

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 3 1700 2100 0 0
aux 3 35 2 1700 2100 0 0
aux 4 47 1 1300 1700 0 0

# adjrange
adjrange 0 0 1 900 1275 12 1 0 0
adjrange 1 0 1 1275 1700 12 1 0 0
adjrange 2 0 1 1700 2100 12 1 0 0

# rxfail
rxfail 3 h
rxfail 5 s 1000

# master
set acc_calibration = -14,1,50,1
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set fpv_mix_degrees = 12
set serialrx_provider = CRSF
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT300
set bat_capacity = 520
set vbat_max_cell_voltage = 450
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 4
set thrust_linear = 30
set osd_warn_batt_not_full = OFF
set osd_warn_batt_critical = OFF
set osd_warn_crash_flip = OFF
set osd_warn_core_temp = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 14580
set osd_rssi_pos = 100
set osd_link_quality_pos = 14469
set osd_crosshairs_pos = 14541
set osd_mah_drawn_pos = 14612
set debug_mode = GYRO_SCALED
set name = Zeus10

profile 0

rateprofile 0

# end the command batch
batch end
```

### Setup / Versions
 
 - Flight controller:  HGLRC Zeus10
 
 - Other components: caddx vista, crossfire tracer nano-receiver
 
 - How are the different components wired up:
Standard 85mm quad with a caddx vista on uart2 (msp) , serial crossfire tracer nano-receiver at uart 1

I compared the dump all of Bf 4.2.11 with the dump all of Bf 4.3.0 -RC2 but found (for me) no difference explaining this behavior.
Tried former versions of Bf 4.3.x - all happened to have this possible bug

haslinghuis, 2022-01-19:
Please try latest nightly build as some issues have been fixed.

Mode-1, 2022-01-19:
Tried build #2625 but behavior still persists.
# Betaflight / STM32F411 (S411) 4.3.0 Jan 18 2022 / 07:33:18 (9c3562a10) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: ZEUSF4EVO, version: 7310ccb4, date: 2021-10-10T02:07:48Z
Tried to disable the led-strip because i read of possible dma-conflicts of F411's, but behavior still persists.

klutvott123, 2022-01-19:
Have you tried tuning it? 4.3 defaults are not the same as 4.2 defaults. Both D and D min gains are higher for example. Gyro lpf cutoffs are a little higher too.

Mode-1, 2022-01-19:
@kluvott123 i did not see a necessarity to tune the quad since it's not able to hover and i got an equal quad (85er) with bf 4.3 rc2 and the zeus5 (different target HGLRCF411) which is working absolutely fine with the 4.3.rc2 standard-tune (all sliders at defaults, only thrust linear at 30%). The same applies to a 75er with zeus5, working fine on defaults and bf 4.3 rc2. Of course i cannot fully exclude that a PID-d is freaking for some unknown reason at the ZEUSF4EVO-target, but my bf 4.3 experience is that 85er / 75er quads are much more tolerant (less breakout) when you are touching an obstacle than in bf 4.2.  
I will try and activate the blackbox on the zeus10 and post the record here but i have to clamp the quad to the ground other it will be gone on arm :-)
Btw, all motors do run without hearable resonances including props when doing the motor tests and the directions are set correctly. BLheli-firmware is JESC 2.2 48kHz on all my 75 and 85ers with 0803 / 1103 motors and with BF 4.2.11 and BF4.2.11 defaults this quad flies like charm, so accelerometer and other FC-hardware should be ok.

Mode-1, 2022-01-19:
ok i found how to reproduce the bug: 
when i switch the blackbox on,  1kHz logging the idling motors are ok and the quad flies as exspected
when i switch the blackbox off,  "no logging" then the motors will freak out after arming

Mode-1, 2022-01-22:
... after some flights and blackbox-logs i found a nasty 106 Hz **frame resonance** causing this issue.
klutvott123 was right, BF 4.2 had more overall filtering a less D which made it work there.
What made it disappear with logging off was the tying to the ground so the frame could not resonate.
With a notch the quad is back to normal (i will have a stiffer frame soon) . I also set some min's to 80Hz:

set gyro_notch1_hz = 106
set gyro_notch1_cutoff = 100
set dyn_notch_q = 250
set dyn_notch_min_hz = 80
set gyro_lpf1_dyn_min_hz = 80
set gyro_lpf1_dyn_max_hz = 425

So my fault - sorry for the inconveniences !

closed 2022-01-22 by Mode-1