betaflight/betaflight #11318: intermittent failsafes while using expresslrs SPI

mard89, 2022-01-19:
### Describe the bug
I'm getting random failsafes when using the SPI expresslrs receiver on the 12a betafpv AIO.
Different builds of betaflight (with different changes made to the expresslrs code) behave differently. Not only in mean time between failsafes, but also in the time it takes for the link to reestablish.

I usually get a few late tasks if I run tasks a few times in the CLI. Maybe ones every 10 times a run tasks.

`# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       1       0    0.0%    0.0%         0      0     13       1
01 - (         SYSTEM)    931       2       0    0.1%    0.0%        11      0   1217       1
02 - (           GYRO)   8019      18      12   14.4%    9.6%      1948      0  10526       0
03 - (         FILTER)   4010      44      37   17.6%   14.8%      3111      0   5263       0
04 - (            PID)   4008     107     100   42.8%   40.0%      8127      0   5263       0
07 - (             RX)    500      33       3    1.6%    0.1%       602      2   2624      14
08 - (         SERIAL)     98    1557      34   15.2%    0.3%       982      0    128      34
09 - (       DISPATCH)    927       3       1    0.2%    0.0%        14      0   1220       1
10 - (BATTERY_VOLTAGE)    194       4       3    0.0%    0.0%         9      0    255       2
11 - (BATTERY_CURRENT)     50       1       1    0.0%    0.0%         0      0     65       0
12 - ( BATTERY_ALERTS)      6       2       1    0.0%    0.0%         0      0      7       1
13 - (         BEEPER)     98       2       1    0.0%    0.0%         2      0    128       1
18 - (            OSD)     12      21       6    0.0%    0.0%        42      0    256      15
20 - (            CMS)     20       5       4    0.0%    0.0%         0      0     26       4
21 - (        VTXCTRL)      6       2       1    0.0%    0.0%         0      0      7       2
22 - (        CAMCTRL)      6       1       3    0.0%    0.0%         0      0      7       3
24 - (    ADCINTERNAL)      3       2       2    0.0%    0.0%         0      0      1       2
26 - (SPEED_NEGOTIATION)     97       1       2    0.0%    0.0%         1      0    128       2
RX Check Function                 104      20                       915
Total (excluding SERIAL)                                64.5%`

### To Reproduce
The failsafes occur randomly. But I usually get one roughly when flying a pack or 4, or within 10 to 15 minutes on the bench. 

### Expected behavior
No more failsafes

### Flight controller configuration

```
# diff

# version
# Betaflight / BETAFPVF4SX1280 (BHER) 4.3.0 Jan 16 2022 / 03:45:22 (90cd5fb0c) MSP API: 1.44

# start the command batch
batch start

board_name BETAFPVF4SX1280
manufacturer_id BEFH

# timer
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature -LED_STRIP
feature -AIRMODE
feature MOTOR_STOP

# serial
serial 1 1 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1600 2100 0 0
aux 1 13 3 900 1225 0 0
aux 2 28 1 1350 1725 0 0
aux 3 35 1 900 1325 0 0

# vtxtable
vtxtable bands 2
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 850
set acc_hardware = NONE
set acc_trim_pitch = 4
set acc_trim_roll = -4
set acc_calibration = -59,29,96,1
set min_check = 1010
set max_check = 2000
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set serialrx_provider = CRSF
set blackbox_device = NONE
set min_throttle = 1010
set dshot_idle_value = 100
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 340
set yaw_motors_reversed = ON
set small_angle = 180
set report_cell_voltage = ON
set osd_vbat_pos = 343
set osd_rssi_pos = 228
set osd_link_quality_pos = 2497
set osd_rssi_dbm_pos = 2518
set osd_tim_2_pos = 2511
set osd_flymode_pos = 323
set osd_current_pos = 374
set osd_craft_name_pos = 362
set osd_avg_cell_voltage_pos = 2504
set osd_disarmed_pos = 2282
set osd_flip_arrow_pos = 2285
set osd_core_temp_pos = 182
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set osd_displayport_device = MSP
set osd_menu_background = GRAY
set debug_mode = RX_EXPRESSLRS_SPI
set cpu_overclock = 120MHZ
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 1
set displayport_msp_serial = 1
set displayport_max7456_inv = ON
set expresslrs_uid = 2,16,58,67,250,87

profile 0

# profile 0
set vbat_sag_compensation = 100
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 5
set throttle_boost = 20
set p_pitch = 78
set i_pitch = 139
set d_pitch = 94
set f_pitch = 290
set p_roll = 78
set i_roll = 140
set d_roll = 91
set f_roll = 294
set p_yaw = 150
set i_yaw = 150
set f_yaw = 250
set angle_level_strength = 120
set d_min_roll = 78
set d_min_pitch = 80
set thrust_linear = 25
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 4
set dyn_idle_min_rpm = 40
set simplified_pids_mode = RP
set simplified_master_multiplier = 175
set simplified_d_gain = 150
set simplified_dmax_gain = 50
set simplified_feedforward_gain = 140
set simplified_pitch_d_gain = 90
set simplified_pitch_pi_gain = 95

rateprofile 0

# rateprofile 0
set thr_mid = 100
set rates_type = BETAFLIGHT
set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 125
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set tpa_rate = 80
set tpa_breakpoint = 1450
set roll_level_expo = 40
set pitch_level_expo = 40

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 2
A01: ADC_BATT
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: BEEPER
A15: RX_SPI_CS
B00: ADC_CURR
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: GYRO_EXTI
B07: MOTOR 4
B08: MOTOR 1
B09: RX_SPI_EXPRESSLRS_RESET
B10: MOTOR 3
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: LED 1
C15: LED
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
    CH2 : DSHOT_BITBANG 1
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Setup / Versions
I run a 4k PID loop with bi-directional DSHOT and RPM filtering. I exclusively fly acro, so the accelerometer is not enabled. Same for telemetry, led strip and softserial. CPU usage in configurator is between 55-60%

I plan to try this board and setup with an external receiver, but it is delayed in the mail.

AndreaCCIE, 2022-02-10:
@mard89 I also had 2 "loss of connection" (the quad dropped from the sky in front of me) yesterday, but this happened only with the betafpv F4 12A AIO ELRS board, the happymodel F4 2G4 AIO SPI never disarmed instead.

mard89, 2022-03-01:
Fixed with #11380

closed 2022-03-01 by mard89