betaflight/betaflight #11324: BF 4.3 RC2 strange rc commands on H7 Board

wajoli, 2022-01-20:
**Describe the bug**
Intermediate pulsing of RC command **without** any stick input on transmitter

in this file you see pulse of throttle (no stick input at all) pulse from 885 to 988
[image]

in this file additional roll stick is moved to the right end
[image]

**To Reproduce**
just flash H7 board  with BF 4.3.x RC2 leave props of
- set esc protocol DSHOT 600
- set Rx UART port tab
- set RX to serial protocol
- set serial provider to CRFS

**Expected behavior**
only movements of rc-commands with stick inputs on transmitter

**Flight controller configuration**

[Diff_All_2022_01_18.txt]
(https://github.com/betaflight/betaflight/files/7907223/Diff_All_2022_01_18.txt)

[Tasks.txt](https://github.com/betaflight/betaflight/files/7907333/Tasks.txt)

**Setup / Versions**

Flight controller iLumenier LUX H7 HD

Transmitter TBS Mambo firmware 1.3.0 with Crossfire Micro TX modul with firmware 6.09
Crossfire nano diversity with firmware 6.09
- RF Profile 50Hz
- 8 CH
- no FLARM

I tried several adjustments in BF and Crossfire nothing solve the problem.

**Then I flashed BF 4.11 and problem is gone without any further change !!!!**

haslinghuis, 2022-01-20:
Please flash the latest nightly and report back.

SteveCEvans, 2022-01-20:
Tested with Crossfire 6.1.0 on with a nano diversity RX and the following BF. No issues observed at 50 or 150Hz.

```
# Betaflight / SPRACINGH7ZERO (SP7Z) 4.3.0 Jan 20 2022 / 18:13:19 (9c3562a) MSP API: 1.44
```

Config at default bar

```
set rc_smoothing_auto_factor = 40
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
```

danik-guitar, 2022-01-21:
обнови прошивку на передатчике. и всё

wajoli, 2022-01-21:
> Please flash the latest nightly and report back.

With Nightly from 21-01-2022 I had the same issue.

I flashed Crossfire to Beta 6.13 and problem solved !!!!!
Works with Nightly and RC2!!!

The Crossfire update process was not easy.
6.09 is marked as latest public.

Thank a lot for your quick help and reopening my comments

**It was not a BF bug!!!

Thanks all the developer for all the work to make this great Flight Controller Software!!!!!!**

wajoli, 2022-01-21:
Problem solved

closed 2022-01-21 by wajoli